clash-lang/clash-compiler #715: clash-ghc (0.99.3) failed on ghc-8.6.5 on win7

hreidmarr, 2019-08-15:
I tried to install clash-ghc on win7 (Used to work a year ago)

```
# stack build --copy-compiler-tool clash-ghc

Error: While constructing the build plan, the following exceptions were encountered:

In the dependencies for clash-ghc-0.99.3:
    clash-lib must match >=0.99.3 && <1.0, but the stack configuration has no specified version  (latest matching version is 0.99.3)
    clash-prelude must match >=0.99.3 && <1.0, but the stack configuration has no specified version  (latest matching version is 0.99.3)
    ghc-typelits-extra-0.3 from stack configuration does not match >=0.2.5 && <0.3  (latest matching version is 0.2.6)
    ghc-typelits-knownnat-0.6 from stack configuration does not match >=0.5 && <0.6  (latest matching version is 0.5.1)
    unbound-generics-0.4.0 from stack configuration does not match >=0.1 && <0.4  (latest matching version is 0.3.4)
needed since clash-ghc is a build target.

Some different approaches to resolving this:

  * Recommended action: try adding the following to your extra-deps in C:\sr\global-project\stack.yaml:

- clash-lib-0.99.3@sha256:1287ad21deea61d639dca3ad1dc52b01f309ec8d3e01d32b6f00d3747982635a,7604
- clash-prelude-0.99.3@sha256:3dd6d8abfc1fa3b694bea3cf0ba6889b2bdb459ed0af535c0214d440ac8ebf00,9146
- ghc-typelits-extra-0.2.6@sha256:064208c427f987f17c8eac168ee9fb9d944fc54eea146bd412f3db4258b891b6,4824
- ghc-typelits-knownnat-0.5.1@sha256:9d0ca221b404d574c16a7edbb62209e449815a56a0d1a7a46f6b67fd74cf9f6a,4634
- unbound-generics-0.3.4@sha256:103f82a251ff545ecd84cd014e70bfff37e6c684002ef30cb32413475922456b,5339

Plan construction failed.
```

I also followed the recommendation but it didn't work at all!

martijnbastiaan, 2019-08-16:
I don't think anything has changed on 0.99.3. The only thing I can think of that has changed is stack resolvers. Could you try a different one? Maybe:

```
stack build --resolver lts-12.14 clash-ghc
```

leonschoorl, 2019-08-16:
clash 0.99.3 doesn't build with ghc >= 8.6 on any platform.
Using `lts-12.xx` like Martijn suggested should work as this gets you ghc-8.4

hreidmarr, 2019-08-20:
Thank you everyone! Problem is now solved!

closed 2019-08-20 by martijnbastiaan