clash-lang/clash-compiler #730: Bad loop condition in generated verilog for Clash.Sized.Vector.fold

blaxill, 2019-08-21:
(verilog/systemverilog) Clash.Sized.Vector.fold produces a `for` loop which continues while `some genvar >= 0`, but ([following this ref guide, page 21](http://sutherland-hdl.com/pdfs/verilog_2001_ref_guide.pdf)) verilog 2001 genvar's are positive integers and so the condition is always true.

This currently triggers yosys to enter an infinite loop when reading the verilog file. I believe the main offender is Clash rather than Yosys, but I'd be easily convinced otherwise. I can get a minimal Clash->Yosys test case if needed. 

https://github.com/clash-lang/clash-compiler/blob/f283af782e76bdfd4155591b5353bde84dcfe087/clash-lib/prims/verilog/Clash_Sized_Vector.json#L287

christiaanb, 2019-08-21:
I think both yosys and clash are wrong: section 12.1.3.1 of the Verilog LRM (http://www-inst.eecs.berkeley.edu/~cs150/fa06/Labs/verilog-ieee.pdf) says that assigning a negative number to a genvar results in a exception. 

Clearly the for-loop generated by clash will always throw an exception at the end of the last iteration; however, yosys shouldn’t loop forever.

martijnbastiaan, 2019-08-21:
@christiaanb You can assign me if you don't have time to look at it.

commit 8edfa9f7 (2019-08-21): Don't use negative genvars. Fix #730

Verilog spec says we can't
commit 516ddebf (2019-08-21): Don't use negative genvars. Fix #730

Verilog spec says we can't
commit 00b56ebd (2019-08-22): Don't use negative genvars in 'fold'. Fix #730

Verilog spec says we can't
commit feb1a67c (2019-08-22): Don't use negative genvars in 'fold'. Fix #730

Verilog spec says we can't
closed 2019-08-22 by martijnbastiaan