clash-lang/clash-compiler #736: Add Hackage release logic to CI

martijnbastiaan, 2019-08-23:
It _should_ be simple enough, but I'm too unfamiliar with all the intricacies of documentation generation / package publishing to want to do it myself. @christiaanb Could you take a look? I've dug up an example from another project:

https://github.com/bennofs/th-lift-instances/blob/af6f6b573a5189a71fedc75a1742c6298e8ce866/travis/script.sh#L85-L110

If you need help taming GitLab CI I'd be happy to help :)

martijnbastiaan, 2019-08-27:
I remember there being some issue with dependencies having to be built separately with some documentation flag. Is that still the case for Cabal 3.0?

christiaanb, 2019-08-27:
Using `cabal-install-3.0` on `ghc-8.8.1` (to get the new `haddock`):

```
$ cabal sdist
```

To create the documentation:

```
$ cabal new-haddock --haddock-for-hackage --haddock-hyperlinked-source
```

Then:

```
$ cabal upload path/to/sdist.tar.gz
```

to upload a release candidate, and also:

```
$ cabal upload --documentation path/to/haddock.tar.gz
```

to upload the corresponding documentation. And then to move from release candidate to actual release you do:

```
$ cabal upload --publish path/to/sdist.tar.gz
$ cabal upload --documentation --publish path/to/haddock.tar.gz
```

martijnbastiaan, 2019-08-27:
Running the `new-haddock` command with `--enable-documentation` triggers lots of dependency rebuilds for me, so it looks like you're right and it has been fixed :)

leonschoorl, 2019-08-27:
Yes, if we want proper links to documentation of dependencies we must enable documentation generation for dependencies.

For example in the `Signed` module we export `type instance IxValue (Signed n)`.
In the haddock we like `IxValue` to be a link to its definition (in the lens package).

For this to work we need to the documentation for lens to be build before we generate the documentation 
When building documentation we should enable documentation globally, with something like:
cabal.project(.local):
```
package *
  documentation: True
```

There used to be a bug in cabal where enabling this didn't cause cabal to generate documentation for already installed packages. This is fixed in 3.0, possibly earlier.

martijnbastiaan, 2019-08-30:
Fixed in https://github.com/clash-lang/clash-compiler/pull/746.

closed 2019-08-30 by martijnbastiaan