clash-lang/clash-compiler #738: Allow active-low reset

ncihnegn, 2019-08-23:
Or the question is how to design with an active-low reset?

martijnbastiaan, 2019-08-24:
Hi @ncihnegn, welcome to Clash!

Active low resets are not supported in Clash 0.99.3, the current released version. We plan on releasing Clash 1.0 the 3rd of September, which does have support for it. If you want to run that version right now, please clone this repository and follow the build instructions in README.md.

As for Clash 1.0: reset polarity is part of the domain configuration. You can read about those [over here](https://clash-lang.org/blog/0005-synthesis-domain/). Please see the `VDomainConfiguration` data structure for all the options:

https://github.com/clash-lang/clash-compiler/blob/master/clash-prelude/src/Clash/Signal/Internal.hs#L480-L505

In your case, you would have to create a new domain with the reset polarity set to `ActiveLow`:

```haskell
createDomain vSystem{vTag="SystemLow", vPolarity=ActiveLow}
```

I do plan on making the record field names consistent with the reset of the code (`vTag` -> `vName`, `vPolarity` -> `vResetPolarity`), so please expect that to change before the release. You can now use that domain in your designs like:

```haskell
myCircuit
  :: Signal SystemLow Bool
  -> ...
```

martijnbastiaan, 2019-08-27:
@ncihnegn I've just merged the renames:

https://github.com/clash-lang/clash-compiler/pull/741

Do you feel like your question has been answered?

closed 2019-08-28 by martijnbastiaan