clash-lang/clash-compiler #748: Generalize `Undefined` to work with non-errors too

martijnbastiaan, 2019-08-29:
`deepErrorX` builds the structure/spine of a datatype and puts an error at its "core". This can be generalized such that it "lazyfies" a given type instead. Consider:

```haskell
Clash.Prelude> x = errorX "foo" :: Vec 3 Int
Clash.Prelude> map (const 3) x
<*** Exception: X: foo
```

We could build `lazify` (for the lack of a better name) to achieve:

```haskell
Clash.Prelude> x = lazify (errorX "foo") :: Vec 3 Int
Clash.Prelude> map (const 3) x
<3,3,3>
```

`deepErrorX` can now be defined as:

```haskell
deepErrorX = lazify . errorX
```

christiaanb, 2019-08-29:
It's somewhat specific to products of a single type though... how do you generalize this to e.g. tuples?

martijnbastiaan, 2019-08-29:
I'm not sure I understand what you're saying. The implementation for a tuple would be something like:

```haskell
instance (Undefined a, Undefined b) => Undefined (a, b) where
  -- lazify ~(a, b) = (lazify a, lazify b) ?
  lazify ab = 
    ( let (a, _) = ab in lazify a
    , let (_, b) = ab in lazify b
    )
```

christiaanb, 2019-08-29:
So what's the `class` supposed to look like?

martijnbastiaan, 2019-08-29:
```haskell
class Undefined a where
  lazify :: a -> a
```

? Could you just explain what issues you're seeing?

christiaanb, 2019-08-29:
I don't see what the benefit is of having:

```
lazify (1,2,errorX "foo") :: (Int,Int,(Int, Int))
```

over

```
(1,2,deepErrorX "Foo" :: (Int,Int))
```

martijnbastiaan, 2019-08-29:
Ah right. It allows your to implement `deepLazyV`, if you will.

martijnbastiaan, 2019-08-29:
To be more specific, `deepLazyV` would be a type specialized version of `lazify`.

martijnbastiaan, 2019-08-29:
Does that make sense @christiaanb?

christiaanb, 2019-08-29:
Yeah, that makes sense

jonfowler, 2019-08-30:
@christiaanb a potential use case is when you have `mux`:
```
mux (errorX "undefined") (pure (repeat 0)) someVecSig
```
will produce `errorX "undefined"`. With `lazify` you could write:
```
lazify $ mux ...
```
which will give you back the structure.

commit 0a969261 (2019-10-01): Implement 'NFDataX.ensureSpine'. Fix #748
commit 2c46c6c3 (2019-10-01): Implement 'NFDataX.ensureSpine'. Fix #748
closed 2019-10-01 by christiaanb