clash-lang/clash-compiler #778: `msb` and `lsb` succeed on void BitVectors

martijnbastiaan, 2019-09-05:
```haskell
Clash.Prelude> msb (0 :: BitVector 0)
0
Clash.Prelude> lsb (0 :: BitVector 0)
0
```

I'd expect an `error` (_not_ `errorX`) instead.

Thanks to @JvWesterveld for pointing this out.

alex-mckenna, 2019-11-28:
Update:

So type errors don't seem feasible here without fairly intrusive changes (due to `msb#` and `lsb#` being used in the instances for `SaturatingNum` and `Resize` on `BitVector`). I'd rather avoid that because adding extra constrains to their class functions seems like a bad idea :tm:

Using `error` throws up another issue though, which is that `reduceConstant` in `Clash.GHC.Evaluator` uses `lsb#`, and complains when `KnownNat` is added to it (which is done to check the width of the `BitVector` is not 0). I'm not entirely sure how to go about resolving this, as the offending line is:

```
src-ghc/Clash/GHC/Evaluator.hs:1541:26: error:
    • No instance for (KnownNat n0)
        arising from a use of ‘BitVector.lsb#’
    • In the expression: BitVector.lsb# (toBV i)
      In a pattern binding: Bit msk val = BitVector.lsb# (toBV i)
      In the expression:
        let
          resTy = getResultTy tcm ty tys
          Bit msk val = BitVector.lsb# (toBV i)
        in reduce (mkBitLit resTy msk val)
     |
1541 |            Bit msk val = BitVector.lsb# (toBV i)
     |                          ^^^^^^^^^^^^^^^^^^^^^^^
```

Thoughts @martijnbastiaan?

martijnbastiaan, 2019-11-28:
Hmm, yeah, I didn't consider the need to add a `KnownNat`. Also, I should have updated this issue a while ago. While I'm personally _for_ adding the `KnownNat` constraint and throwing an error, @christiaanb has basically said that we shouldn't consider it - it's documented behavior and it would change the API.

Sorry for wasting your time @alex-mckenna !

martijnbastiaan, 2019-11-28:
To directly answer your question btw, have a look at this: https://github.com/clash-lang/clash-compiler/blob/328b2334bdf1e8fc837b6fa3fe289f2ab3de25b6/clash-ghc/src-ghc/Clash/GHC/Evaluator.hs#L1434-L1442

martijnbastiaan, 2019-11-28:
@christiaanb Could you close this issue / add `wontfix` if you still feel the same?

christiaanb, 2019-11-29:
I'd rather avoid partial functions. If anything we should constrain the type with a `1 <= n` constraint.

closed 2019-11-29 by christiaanb