clash-lang/clash-compiler #779: Remove redundant generate mappings from verilog output

ncihnegn, 2019-09-05:
The verilog output of _refAdder32_ in https://github.com/ncihnegn/clash-with-stack/blob/master/src/PlayClash/Adder.hs contains two redundant generate mappings:

```
  // map begin
  genvar i_0;
  generate
  for (i_0=0; i_0 < 32; i_0 = i_0 + 1) begin : map_0
    wire  map_in_0;
    assign map_in_0 = b[i_0*1+:1];
    wire [0:0] map_out_0;
      assign map_out_0 = map_in_0;

    assign c$scrut_app_arg_0[i_0*1+:1] = map_out_0;
  end
  endgenerate
  // map end

  // map begin
  genvar i_1;
  generate
  for (i_1=0; i_1 < 32; i_1 = i_1 + 1) begin : map_1
    wire  map_in_1;
    assign map_in_1 = a[i_1*1+:1];
    wire [0:0] map_out_1;
      assign map_out_1 = map_in_1;

    assign c$scrut_app_arg_1[i_1*1+:1] = map_out_1;
  end
  endgenerate
  // map end
```

I'd like to take this task if anyone can point me to the right direction.

christiaanb, 2019-09-20:
@ncihnegn Apologies for the delay on this, it's actually less trivial than you might think. The reason being that those loops wouldn't be superfluant when you're using the VHDL backend.

The way forward is probably:

1. Extend `WorkInfo`, https://github.com/clash-lang/clash-compiler/blob/7ef39bd7f3e53aeb93f3d0afe37a3c90be4620a3/clash-lib/src/Clash/Core/Term.hs#L93-L103, with a `WordIdentity Int [Int]` which behaves just like `WorkNever`, with the additional constraint that it's the identity for the `Int`th argument give that the arguments in the `[Int]`th positions are also the identity.
2. Change the workinfo for `commonverilog` primitive definitions for `pack#`, https://github.com/clash-lang/clash-compiler/blob/7ef39bd7f3e53aeb93f3d0afe37a3c90be4620a3/clash-lib/prims/commonverilog/Clash_Sized_Internal_BitVector.json#L37 and `unpack#` https://github.com/clash-lang/clash-compiler/blob/7ef39bd7f3e53aeb93f3d0afe37a3c90be4620a3/clash-lib/prims/commonverilog/Clash_Sized_Internal_BitVector.json#L45 to `Identity 0 []`
3. Change the workinfo for the `verilog` primitive definition for `map` https://github.com/clash-lang/clash-compiler/blob/7ef39bd7f3e53aeb93f3d0afe37a3c90be4620a3/clash-lib/prims/verilog/Clash_Sized_Vector.json#L101 to `Identity 1 [0]`
4. Change the `isInteresting` function https://github.com/clash-lang/clash-compiler/blob/7ef39bd7f3e53aeb93f3d0afe37a3c90be4620a3/clash-lib/src/Clash/Normalize/Transformations.hs#L2203-L2234 to return a `Either (Id, Term) (Id, Term)` instead of a bool. Returning `Left` the original binding where we now return `False`, and `Right` the original binding where we now return `True`.
5. Extend the `isInteresting` function to do the "identity" analysis for primitives, and when this analysis shows that the primitive is the identity for one of its argument, instead of returning `Right` the original let-binding, you return `Right` "`unsafeCoerce` of the argument the primitive is the identity for`.

ncihnegn, 2019-09-21:
I am lost at the type of `isInteresting`. It has to return `bool` for `List.partition`, doesn't it?

christiaanb, 2019-09-21:
Ah, correct, you’ll have to exchange the `Left` and `Right` in my explanation; and use `Either.partitionEithers` of course.

leonschoorl, 2021-06-17:
As a simplified testcase you can use:
```haskell
topEntity :: BitVector 4 -> Vec 4 Bit
topEntity = bv2v
```

ZabelTech, 2021-08-03:
Hi, 
I came over from WorksHub and took a stab at that issue. 
Thanks @christiaanb for the detailed pointers and @leonschoorl for the minified example.

However, I'm not clear on two things. Lets start with the first one:

After identifying a term to be identical with one of its argument, 
it's suggested above that I `return Right "unsafeCoerce of the argument the primitive is the identity for`.
I'm tempted to just swap the term itself with it's identity here, and in any case I don't see how I could make use of `unsafeCoerce`.
Could anyone elaborate on this?

alex-mckenna, 2021-08-04:
Hi @ZabelTech,

The point of the unsafe coerce is to make sure the `Term` still has the correct type. With `pack# :: Bit -> BitVector 1`, there are different input and output types, so if we replace the primitive with it's argument we need to add an `unsafeCoerce#` to the term so the type is still correct, i.e.

```haskell
Prim pack# `App` bit ~~> Prim unsafeCoerce# bit
```

It becomes more meaningful if you imagine a primitive with many arguments. Then it becomes something more like

```haskell
-- where k <= n
Prim p `App` arg1 `App` ... `App` argN ~~> Prim unsafeCoerce# argK
```

I'm not sure we have a pre-constructed [`PrimInfo`](https://hackage.haskell.org/package/clash-lib-1.4.2/docs/Clash-Core-Term.html#t:PrimInfo) for `GHC.Prim.unsafeCoerce#` lying around, you may need to define it yourself. There are some examples of that [here](https://hackage.haskell.org/package/clash-lib-1.5.0/candidate/docs/Clash-Normalize-Primitives.html) if you look at the source. If you look at `undefinedTy` for the `removedArg` prim then you can see how to construct a type for `unsafeCoerce#`.

ZabelTech, 2021-08-05:
@alex-mckenna  nice, that helped alot.

To the other thing that I am not clear on:
This minified example does now not produce the loop anymore
```
topEntity :: BitVector 4 -> Vec 4 Bit
topEntity = bv2v
```
Which is desired.

However, the case that OP was complaining about still produces two loops (one less than before).
Now I'm not familiar with verilog or any hdl for that matter, so I can't tell what is expected here.

alex-mckenna, 2021-08-05:
> This minified example does now not produce the loop anymore

I'll take a look tomorrow morning and see if I can make a new minimised reproducer for you

> Now I'm not familiar with verilog or any hdl for that matter, so I can't tell what is expected here.

Looking at the current minimised example, the following can be reasoned:

  1. The function `bv2v` [is a synonym](https://hackage.haskell.org/package/clash-prelude-1.4.2/docs/src/Clash.Sized.Vector.html#bv2v) for `unpack` on `Vec`
  2. `unpack` on `Vec` [is defined as](https://hackage.haskell.org/package/clash-prelude-1.4.2/docs/src/Clash.Sized.Vector.html#line-2371)
 
       ```haskell
       unpack = map unpack . unconcatBitVector#
       ```

  3. `unconcatBitVector#` [has the (simplified) type](https://hackage.haskell.org/package/clash-prelude-1.4.2/docs/src/Clash.Sized.Vector.html#unconcatBitVector%23) `BitVector (n * m) -> Vec n (BitVector m)`
  4. In this example, `m = 1`, meaning the `unpack` in `map unpack` should be a no-op in verilog
  5. Which means the whole `map unpack` should be a no-op
  6. Since `unconcatBitVector#` renders as a no-op (see [the template](https://github.com/clash-lang/clash-compiler/blob/master/clash-lib/prims/verilog/Clash_Sized_Vector.primitives#L334) where it simply renders as in HDL as whatever the `BitVector` argument is) in verilog, as far as HDL generation is concerned, `topEntity = id`
  7. No `generate for` is rendered in verilog because every action is a no-op

I would assume (although I would have to double check) that the original example produced loops for similar reasons: the calls to `v2bv` and `bv2v` produced loops for these `map pack` / `map unpack` calls (giving 3 loops in total). Any of these calls which would involve mapping either `pack :: Bit -> BitVector 1` or `unpack :: BitVector 1 -> Bit` should be an identity in verilog and therefore produce no `generate for`

I hope this helps make it clearer if a `generate for` is unexpected or not. If not, please let me know :slightly_smiling_face:

ZabelTech, 2021-08-05:
thanks man, this is very detailed and you are spot on :+1:

`bv2v` collapses to a noop just fine, hence -1 loops in both examples. 
`v2bv` on the other hand does not.

Further experimentation reveals the call to `packXWith` in `pack` as the culprit.
This layer of indirection makes the identity analysis see a `Var` instead of the expected `pack#` - Prim as the first argument of `map` .

alex-mckenna, 2021-08-06:
> I'll take a look tomorrow morning and see if I can make a new minimised reproducer for you

Sounds like you've already figured this out yourself, but for the sake of completeness another reproducer would be

```haskell
topEntity :: Vec 4 Bit -> BitVector 4
topEntity = v2bv
```

Given we want both of these to work as intended, I would probably revise the reproducer to something like

```haskell
topEntity :: BitVector 4 -> BitVector 4
topEntity = v2bv . bv2v
```

which for me currently produces two `generate for` blocks but should produce none when this issue is fixed.

> Further experimentation reveals the call to packXWith in pack as the culprit.

Hmm, that seems strange to me. When we convert from GHC's core language to Clash's we remove calls to `packXWith` ([see here](https://github.com/clash-lang/clash-compiler/blob/master/clash-ghc/src-ghc/Clash/GHC/GHC2Core.hs#L375-L377)). So I would hope it doesn't appear in the core at all :thinking: Perhaps I'm missing something non-obvious about why it would be causing a problem (cc: @christiaanb)

christiaanb, 2021-08-06:
@ZabelTech could post (using your pastebin of choice) the output of a run with `-fclash-debug DebugFinal` (or even `-fclash-debug DebugApplied`) on your branch on the file where you say `packXWith` is the culprit? I'm not exactly sure by what you mean that the identify analysis is seeing a Var instead of `pack#`. Thanks!

ZabelTech, 2021-08-06:
`Further experimentation reveals the call to packXWith in pack as the culprit.`
This might have been misleading. All I wanted to say is, that it works without that call.

`This layer of indirection makes the identity analysis see a Var instead of the expected pack# - Prim as the first argument of map`
It seems that the (non-inlined) call to `packXWith` will make its arguments be put into `Var`s.
So when I later - during the identity analysis - check on the arguments for `map`, I will see:
`Var (Id {varName = Name {nameSort = User, nameOcc = "Clash.Class.BitPack.Internal.$fBitPackBit1",...`
instead of the usual 
`Prim PrimInfo{..}`.

Long story short:
To correctly check for identities I needed a way to resolve Vars to the original term they are pointing to.
I looked a bit around, but it's not obvious to me how to do that. 
Any pointers would be appreciated.

christiaanb, 2021-08-06:
Aha, it's still hiding in the `BitPack` class dictionary. 

Assuming you went ahead with my initial suggestion (https://github.com/clash-lang/clash-compiler/issues/779#issuecomment-533493784) of extending `isInteresting` which is part of the `inlineCleanup` transformation. Perhaps a way forward is to add:
```
apply "inlineCleanup" inlineCleanup >->
```
right after https://github.com/clash-lang/clash-compiler/blob/39a62558c2414699d98610e3808443b0a2da8471/clash-lib/src/Clash/Normalize.hs#L416
This will then ensure that `inlineCleanup` is called during the post-normalization hierarchy-flattening stage.

ZabelTech, 2021-08-06:
@christiaanb yes, I'm doing what you suggested initially above.

Second, nope, your recent proposed change does not do it - at least not on its own.
I then get a bunch of extra argument checks with `Lam` terms, which I don't currently know what to make of.
I'll have a look into that direction further.

Maybe this observation is also helpful:
Changing packXWith to `packXWith f x = f x` and removing the NOINLINE pragma makes it work right away.

alex-mckenna, 2021-08-06:
> Changing packXWith to `packXWith f x = f x` and removing the NOINLINE pragma makes it work right away.

I would guess this is because GHC is removing it in the simplifier before it gets to Clash (but that's just conjecture).

You may need to experiment more with where the new `inlineCleanup` Christiaan suggested should go, the rewrite system can be slightly fiddly in this way sometimes. If you have some useful debug output to share that might be helpful, i.e. the extra terms you're getting that you don't understand.

ZabelTech, 2021-08-06:
Alright, I got it. Now before I polish it up:
In principle, does something speak against resolving `Var` arguments via `lookupVarEnv i <$> Lens.use bindings` directly in the identityAnalysis i.e inside `inlineCleanup / isInteresting`?

alex-mckenna, 2021-08-06:
> In principle, does something speak against resolving `Var` arguments via `lookupVarEnv i <$> Lens.use bindings` directly in the identityAnalysis i.e inside `inlineCleanup / isInteresting`?

In theory no (at least not from me). If there are any improvements to be made I can explain them during the PR review because it's clearer when there's code to reference, and if you have a fix for the issue it's a good time to open a PR so the whole solution can be looked at

PR #1896 (2021-08-12): introduce `WorkIdentity Int [Int]` and corresponding `inlineIdentity`…
This patch introduces an noop analysis on Prims to prevent redundant HDL generation.
Closes #779

I deviated from the plan discussed in #779 by separating the identity analysis from `inlineCleanup`.

This is because inline cleanup as is would return the unchanged `Letrec`, if there was nothing found via `isInteresting`.
However there might have been 'uninteresting' terms, that were identified as noops and as such coerced to one of their arguments. In that case, those would have been not evaluated. 
To change that, the check [... cut]
closed 2021-08-12 by alex-mckenna