clash-lang/clash-compiler #810: Vec 1 Bit turned into single wire in Verilog

gergoerdi, 2019-09-20:
I have a Vivado constraints file that has a 16 wires wide `LED` bus. I can use a bottom slice easily from Clash without having to change the constraints file: I can connect, e.g., a `Vec 8 Bit` to it to drive only the lowest 8 bits.

However, this same method breaks down for `Vec 1 Bit`: At least when targeting Verilog, Clash 1.0 synthesizes that into a single `wire`, instead of a `wire [0:0]`, resulting in an unconnected output, since `LED` and `LED[0]` are not the same.

christiaanb, 2019-09-20:
Mmm... this behaviour was introduced by: https://github.com/clash-lang/clash-compiler/commit/7313dfd9271dcd4898834c912668384bdc1bfc46 but that was when `Bit` was still a type alias for `BitVector 1`; there days they're distinct types. So I think we can revert the rendering behavior introduced in that change. @thoughtpolice : thoughts?

christiaanb, 2019-09-30:
I think we should revert https://github.com/clash-lang/clash-compiler/commit/7313dfd9271dcd4898834c912668384bdc1bfc46

jhmcstanton, 2019-10-02:
Are there backend tests somewhere that need to be updated with this change? Didn't see anything related in `clash-lib`

christiaanb, 2019-10-02:
No, but we should probably **add** an additional "outputTest" (see e.g. https://github.com/clash-lang/clash-compiler/blob/c5a432776a32c92e3e78c47a9c6249d9ab816977/testsuite/Main.hs#L298 and https://github.com/clash-lang/clash-compiler/blob/master/tests/shouldwork/TopEntity/PortNames.hs)

That tests whether the verilog generated for:

```haskell
{-# ANN topEntity
  (Synthesize
    { t_name     = "Test_topEntity"
    , t_inputs   =
        [ PortName "inp0"
        , PortName "inp1"
        , PortName "inp2"
        , PortName "inp3"
        , PortName "inp4"
        , PortName "inp5"
        , PortName "inp6"
        ]
    , t_output = PortName "outp0"
    }) #-}
topEntity
  :: Bit
  -> BitVector 1
  -> Unsigned 1
  -> Vec 1 Bit
  -> Vec 1 (BitVector 1)
  -> Vec 1 (Unsigned 1)
  -> Vec 1 (Signed 1)
  -> Bool
topEntity _ _ _ _ _ _ _ = True
```

contains the following strings:

* `input  inp0`
* `input [0:0] inp1`
* `input [0:0] inp2`
* `input [0:0] inp3`
* `input [0:0] inp4`
* `input [0:0] inp5`
* `input [0:0] inp6`

jhmcstanton, 2019-10-02:
👍 looks like I was looking in the wrong place, seems easy enough. I'll take a look at this later today as well.

jhmcstanton, 2019-10-03:
Haven't forgotten about this, I've just had a hard time getting tests to build following the wiki instructions (`cabal new-run -fdisable-sv-tests -- clash-testsuite -j4` ===> "355 out of 1508 tests failed (3502.88s)"). Its been a combination of local dev environment issues (ghdl acting weird), and a lowish powered laptop taking forever to run.

christiaanb, 2019-10-04:
@jhmcstanton Which version of GHDL are you using? 0.31 as mentioned on https://github.com/clash-lang/clash-compiler/wiki/Getting-started#running-the-testsuite ? Because 0.33 doesn't work with the VHDL that Clash generates. Alternatively, you can build the very latest dev version of GHDL using the LLVM backend https://ghdl.readthedocs.io/en/latest/getting/LLVM.html; that's what I use on my local machine(s) these days.

martijnbastiaan, 2019-10-04:
@jhmcstanton I usually use this script: https://github.com/clash-lang/clash-compiler/issues/136#issuecomment-451936704

jhmcstanton, 2019-10-04:
Just rebuilt GHDL with the above script, and so far the test suite looks much healthier. Thanks @christiaanb and @martijnbastiaan !

PR #835 (2019-10-07): Verilog vec1bit output fix.
 Fixes #810. Change here is based on the discussion in the issue.
commit fe8397a3 (2019-10-07): Manual revert of 7313dfd9271dcd4898834c912668384bdc1bfc46 to fix #810

Properly rendering BitVector of size 1 as a singleton vector and not a single bit.
commit df941574 (2019-10-15): Manual revert of 7313dfd9271dcd4898834c912668384bdc1bfc46 to fix #810

Properly rendering BitVector of size 1 as a singleton vector and not a single bit.
commit ff9eb43e (2019-10-15): Manual revert of 7313dfd9271dcd4898834c912668384bdc1bfc46 to fix #810

Properly rendering BitVector of size 1 as a singleton vector and not a single bit.
closed 2019-10-07 by christiaanb