clash-lang/clash-compiler #813: Large IP templates

blaxill, 2019-09-22:
Some larger IP have a large number of input/output variables. E.g. Xilinx pcie_7x_0 module can have over 100 variables (input+output). My current approach has been to connect these modules together outside of Clash, but I'd like to avoid this. However I'm not entirely sure how to do this- 

I've naively experimented with a simple blackbox but in my case this resulted in a Haskell function with ~50 inputs as standard function arguments and ~70 outputs in a tuple. Now ignoring this is horrible to define/use, it doesn't actually work- GHC supports some 65 elements per tuple and will throw an error suggesting to nest them. Except I don't think I can do this in a straight forward way as the generated verilog will concatenate elements inside a tuple (am I wrong here?). It is possible to then extract the relevant parts out from the concatenation result but would be laborious.

Is there a simpler way to integrate large IP into Clash projects?

christiaanb, 2019-09-25:
But nesting the tuples works out much easier on the output side than on the input side as far as defining blackbox templates goes, especially for Verilog:

```
wire ~GENSYM[out1][0];
...
wire ~GENSYM[outN][N];

big_ip ~GENSYM[big_ip_~RESULT][M]
  ( .out1 (~SYM[0])
    ...
    .outN (~SYM[N])
  )

assign ~RESULT = {~SYM[0], ~SYM[1], ... ~SYM[N]};
```

christiaanb, 2019-09-25:
Also, for IPs with many parameters, I would create a primitive with individual inputs for every parameter. But then additionally create a wrapper that takes a record of all the parameters, e.g.

```haskell
myPrimitive#
  :: param1
  -> param2
  -> param3
  -> inp1
  -> inp2
  -> outp

data MyPrimSettings
  = MyPrimSettings
  { myPrimSetting1 :: param1
  , myPrimSetting2 :: param2
  , myPrimSetting3 :: param3
  }

myPrimDefault
  = MyPrimSettings
  { myPrimSetting1 = def1
  , myPrimSetting2 = def2
  , myPrimSetting3 = def3
  }

myPrimitive (MyPrimSettings param1 param2 param3) = myPrimitive# param1 param2 param3
``` 

then if you only want to change one parameter out of the 40 you just do:

```haskell
myPrimitive (myPrimDefault {myPrimSetting2 = xyz})
```

martijnbastiaan, 2019-10-03:
I think this question is answered now. Thanks for asking @blaxill !

closed 2019-10-03 by martijnbastiaan

blaxill, 2019-10-05:
I do think the story for many-I/O IP integration could be better here, as the current method is error prone. I have some thoughts on how to make this better for simple blackbox integration but I don't have time to work on it right now. Thanks for the solutions above though, they fix the immediate issue.