clash-lang/clash-compiler #817: Investigate whether we can use opaque constraint synonyms

martijnbastiaan, 2019-09-28:
We should investigate whether we can use it to make type signatures (and error messages?) of constructs using `Hidden*` more user friendly.

https://kcsongor.github.io/opaque-constraint-synonyms/

alex-mckenna, 2019-11-12:
@martijnbastiaan @christiaanb, current thinking is we can't have this without some changes. The constraints of the form `Hidden (HiddenXName dom) (X dom)` can't be used as constraints on a class (as IP cannot be a class constraint due to instance resolution not knowing which instance of an implicit parameter to use). 

I had a quick search through the project, and there don't seem to be any classes which have a `HiddenX` constraint attached to them, so hopefully I'm not missing anything obvious here.

alex-mckenna, 2019-11-12:
Update: I tried changing Hidden to be a class again with an associated constraint. This allows the classes to be defined, but still means functions using the class need to add the associated constraint to theirs (which just leads to a different list of constraints that gets expanded each time).

alex-mckenna, 2019-11-14:
Update: Having had another look at this, I think it could be possible in theory, but only by using the `relflection` package instead of the `-XImplicitParams` extension (as `Reifies s a` can be included in class constraints, but `IP s a` cannot).

This seems like a reasonably sized internal change, and would need further discussion before undertaking (@christiaanb, @martijnbastiaan)

martijnbastiaan, 2019-11-14:
@alex-mckenna We had a reflection based hidden clock implementation once, in parallel to current one. I don't remember why we decided on the implicit param version. I know @christiaanb wrote a report on this, but I can't find it right now.

christiaanb, 2019-11-18:
ImplicitParams version has/had better type inference; far fewer complaints about ambiguity. I'm saying "has/had" because that was the case before multiple implicit clocks, and I don't whether a reflection-based multiple implicit clocks implementation has the same "poor" type inference/ambiguity as the current ImplicitParams-based implementation.

martijnbastiaan, 2020-03-19:
Like @alex-mckenna said, we probably can't use opaque constraint synonyms. Future GHC versions should retain `HiddenClock` constraints instead of splitting it up all the way down to `IP`, so I think this is just a waiting game for now. Hence, I'm closing this issue.

closed 2020-03-19 by martijnbastiaan