clash-lang/clash-compiler #822: Unify blockram primitives

rowanG077, 2019-09-30:
Currently I think the BlockRam part of the Clash API could really use some kind of overhaul.

I see a few issues with the current situation:

1. Too many different functions to create a BlockRam sometimes with confusing names (`blockRAM1` for instance).
2. Clash is only able to create the most basic blockRAM. A blockRAM with a single read and write port which have the same data width and works with a single clock domain. Ideally I would like a user to be able to specify two clock domains and also that the input type and output type don't necessarily need to match. Clash would then need be able to pick the right primitive based on the types. This would allow the creation of asymmetric blockRAMS which can be used as a synchronization primitive.

I think we could move towards a situation where we have a single blockRAM function. And the actual implementation is inferred based on the types and parameters.

At least I know that Intel Quartus and the Cyclone V FPGA would support such a situation on the hardware level but I'm unfamiliar with other platform to make a call in that direction.

christiaanb, 2019-09-30:
Simulation will have to match the synthesized RAMs, so this "picking" based on types will have to be done at the Haskell-level.

christiaanb, 2019-10-01:
@martijnbastiaan That fifo sync uses asynchronous read ports, so might not synthesize to a blockRAM.

christiaanb, 2019-10-01:
@martijnbastiaan Although the address generator for the read port is based register-backed, which the synthesis tools might merge into the RAM; thus allowing synthesis to a blockRAM.

Only one way to find out: run it through a couple of synthesis tools

adamwalker, 2020-03-22:
Just adding a few more (possibly Xilinx centric) features I would like blockRams to have:

- Per-byte write enables
- Read enable
- True dual port mode
- Ultrarams (on Xilinx) 
- Write first, read first, no change modes

I'm happy to help out writing templates and testing, especially on Xilinx devices.