clash-lang/clash-compiler #823: "Duplicate name

leonschoorl, 2019-09-30:


christiaanb, 2019-09-30:
Line 165 on page 179 of the [VHDL LRM](https://edg.uchicago.edu/~tang/VHDLref.pdf) says: "Moreover, every extended identifier is distinct from any basic identifier."

So quartus is broken

martijnbastiaan, 2019-09-30:
If it's easy for us to do, I think we should not generate "duplicate" names. Even though it's not our fault, users might perceive Clash as broken due to this..

christiaanb, 2019-09-30:
I propose `-fclash-hdlsyn Quartus` to enable such work-arounds.

christiaanb, 2020-03-19:
@akilegaspi I'll be removing this issue from workshub, as we're going to fix this ourselves. That is, unless you have something that fixes this issue.

akilegaspi, 2020-03-19:
Please remove me from it, I haven't actually worked on it as I've expected;
thanks.

martijnbastiaan, 2020-08-06:
@christiaanb Have you removed this issue from WorksHub?

closed 2020-08-06 by workshub[bot]

christiaanb, 2020-08-06:
don't reopen