clash-lang/clash-compiler #825: Executables generated by snap-clash refuse to execute

martijnbastiaan, 2019-10-01:
Basically, when using a Clash from snapcraft, the following happens:

```
martijn@thinklap:~$ clash Main.hs -o main
[1 of 1] Compiling Main             ( Main.hs, Main.o )
Linking main ...
martijn@thinklap:~$ ./main 
bash: ./main: File or directory doesn't exist
```

Snapclash uses it's own (nix) linker (?):

```
martijn@thinklap:~$ file main
main: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /usr/nix, for GNU/Linux 2.6.32, not stripped
```

probably causing the error.

martijnbastiaan, 2019-10-01:
We can set the interpreter of the binary using `patchelf`:

```bash
patchelf --set-interpreter /lib64/ld-linux-x86-64.so.2 main
```

Then, with the right libraries installed, we get:

```bash
$ ./main 
Hello world!
```

(In my case I had to copy `libncursesw.so.6` and `libtinfo.so.6` into my pwd, and use `LD_LIBRARY_PATH=.`)

martijnbastiaan, 2019-10-01:
I've found another workaround.

1. Compile the binary using snapclash.
2. Execute `snap run --shell clash`. This will drop you to a shell with the same environment Clash was running in.
3. Execute `./main`.

christiaanb, 2019-10-01:
Another work-around for the interpreter:

```
clash Main.hs -optl -Wl,--dynamic-linker=/lib64/ld-linux-x86-64.so.2 -o main
```

this doesn't fix the ncursesw issue though

martijnbastiaan, 2019-10-24:
To actually solve this I plan on create a snap package not based on nix, but on Debian packages:

https://github.com/clash-lang/clash-compiler/pull/863

This should then use `/lib64/ld-linux-x86-64.so.2`, making it work on practically all Linux distributions.

martijnbastiaan, 2021-03-19:
Snaps are now based on Debian packages, mitigating this issue.

closed 2021-03-19 by martijnbastiaan