clash-lang/clash-compiler #832: Generated code contains modulo operator

adamwalker, 2019-10-04:
With the latest master (f0376), the following clash code:

```
import Clash.Prelude
topEntity :: Vec 8 Bool
topEntity = rotateRight (repeat False) 0
```
synthesizes to Verilog that contains a modulo operator as can be seen in the code fragment below. 
```
      assign c$divider = 64'sd8;

      // modInt# begin
      // remainder
      wire signed [63:0] rem_res;
      assign rem_res = x1 % c$divider;

      // modulo
      assign ww2 = (x1[64-1] == c$divider[64-1]) ?
                       rem_res :
                       ((x1 == 64'sd0) ? 64'sd0 : rem_res + c$divider);
      // modInt# end
```
This results in very long logic paths and bad timing with Vivado, even though it is modulo a power of 2. Ideally, Clash would not use the modulo operator in generated code.

christiaanb, 2019-10-04:
@adamwalker any suggestions on how to implement `rotateRight` without a modulo? The only other solution is to restrict the number of rotations to the number of elements in the Vector.

adamwalker, 2019-10-04:
Hmmm. I see the problem. I do think its preferable to restrict the rotation to the number of elements of the vector, purely because it avoids the modulo operator. 

I see why its technically required to support an arbitrary number of rotations, but I don't think the end user would expect a rotation to result in a modulo computation.

That's just my preference though. If the behaviour remains the same, it's probably a good idea to mention this in the docs.

christiaanb, 2019-10-04:
I could make it a run-time out-of-bounds error. I'll see what I can do.

christiaanb, 2019-10-04:
@adamwalker so I can implement it as:

```haskell
rotateRight :: forall n i a . (Enum i, KnownNat n)
            => Vec n a
            -> i
            -> Vec n a
rotateRight xs i = map (!! i) (reverse (rotations (reverse xs)))
 where
    rotateL ys   = tail ys :< head ys

    rotations :: Vec n a -> Vec n (Vec n a)
    rotations Nil          = Nil
    rotations ys@(Cons {}) = iterateI rotateL ys
{-# INLINE rotateRight #-}
```

then I get:

```
>>> let xs = 1 :> 2 :> 3 :> 4 :> Nil
>>> rotateRight xs 0
<1,2,3,4>
>>> rotateRight xs 1
<4,1,2,3>
>>> rotateRight xs 2
<3,4,1,2>
>>> rotateRight xs 3
<2,3,4,1>
```

however now you get:

```
>>> rotateRight xs 4
<*** Exception: Clash.Sized.Vector.(!!): index 4 is larger than maximum index 3
>>> rotateRight xs (-1)
<*** Exception: Clash.Sized.Vector.(!!): negative index
```

where these things used to work

christiaanb, 2019-10-04:
@adamwalker 

Where I can work-around the negative indices by:

```haskell
rotateRight :: forall n i a . (Enum i, KnownNat n)
            => Vec n a
            -> i
            -> Vec n a
rotateRight Nil _ = Nil
rotateRight xs@Cons {} i = map (!! i') (reverse (rotations (reverse xs)))
 where
    rotateL ys   = tail ys :< head ys

    rotations :: Vec n a -> Vec n (Vec n a)
    rotations Nil          = Nil
    rotations ys@(Cons {}) = iterateI rotateL ys

    i' :: Unsigned (CLog 2 n)
    i' = toEnum (fromEnum i)
```

but it doesn't help me with the out-of-bounds errors:

```
>>> let ys = 1 :> 2 :> 3 :> Nil
>>> rotateRight ys 3
<*** Exception: Clash.Sized.Vector.(!!): index 3 is larger than maximum index 2
```

martijnbastiaan, 2019-10-04:
@christiaanb It's going to break our API anyway, right? Maybe we should just make a separate function?

christiaanb, 2019-10-04:
@martijnbastiaan The CLog to unsigned does quite a bit already; we just need a LUT/ROM to do the final mapping.

christiaanb, 2019-10-04:
And unless you’re doing huge vectors, the LUT/ROM will be small

rowanG077, 2019-10-04:
Why not use `Index` type?

adamwalker, 2019-10-04:
@christiaanb I like both of those implementations. I will use the first one in my design.

adamwalker, 2019-10-04:
@rowanG077 I like the idea of using the Index type. It breaks the API though, so possibly a separate function is needed. 

Also, it might make sense to add a similar shift function to the Vector module for dynamic shifts.

martijnbastiaan, 2019-10-04:
The runtime error will break the API anyway, so if we're fine with that I think we should opt for @rowanG077 's suggestion instead. I'd rather have a compile error than a runtime error!

christiaanb, 2019-10-04:
I want to keep the current behavior (but not implementation) because that's what's likely to end up in ISAs, giving our users a super easy implementation when they want to implement that ISA in a CPU: just call `rotateRight`.

christiaanb, 2019-10-04:
It seems the VHDL standard library uses modulo as well: https://github.com/ghdl/ghdl/blob/657fcfde5fb93c1311fe5fd2d28146c89852614d/libraries/ieee/numeric_std-body.vhdl#L254