clash-lang/clash-compiler #833: Clash (V1.0) Bug Fix of Blinker

paddytheplaster, 2019-10-04:
Blinker module. Based on Clash 1.0 migration instructions, which gave me
compile-time errors. I fixed these errors but I may be doing something
wrong. I compiled with Clash, and synthesised with quartus, and executed
it on a De0nano.

Paddy

> {-# OPTIONS -Wall #-}

> module Blinker where

> import Clash.Prelude
> import Clash.Intel.ClockGen

> data LedMode = Rotate | Complement
>      deriving (Generic,NFDataX)

> complementLedMode :: LedMode -> LedMode
> complementLedMode Rotate = Complement
> complementLedMode _      = Rotate

> data BlinkerState = BlinkerState
>     { leds    :: BitVector 8
>     , mode    :: LedMode
>     , counter :: Index 16650001
>     }
>     deriving (Generic,NFDataX)

> initialState :: BlinkerState
> initialState = BlinkerState { leds = 1, mode = Rotate, counter = 0 }

> -- THIS NEEDS TO BE DEFINED BEFORE YOU CAN USE '@Input'
> -- 50 MHz ~ 1 / (50 * 10^6) sec
> createDomain vSystem { vName = "Input", vPeriod = 20000 }

> {-# ANN topEntity
>   (Synthesize
>     { t_name   = "blinker"
>     , t_inputs = [ PortName "CLOCK_50"
>                  , PortName "KEY0"
>                  , PortName "KEY1"
>                  ]
>     , t_output = PortName "LED"
>     }) #-}
> topEntity
>   :: Clock  Input
>   -> Signal Input Bool
>   -> Signal Input Bit
>   -> Signal Input (BitVector 8)
> topEntity clk20 resetButton modeButton
>     = exposeClockResetEnable
>           (mealy blinkerT initialState . isRising 1)
>           clock50
>           rst
>           en
>           modeButton
>   where
>     -- enable line for subcomponents: we'll keep it always running
>     en = enableGen
>     -- Start with first LED on, in rotate mode, with counter equal to 0
>     -- Signal coming for reset button is low when pressed, and high when
>     -- not pressed. We convert this signal to the polarity of our domain
>     -- with unsafeFromActiveLow
>     rst = unsafeFromLowPolarity resetButton
>     clock50 = fst $ altpll (SSymbol @ "altpll50") clk20 rst

> blinkerT :: BlinkerState -> Bool -> (BlinkerState,BitVector 8)
> blinkerT state0 key1R = (state1,leds0)
>   where
>     -- state extraction
>     counter0 = counter state0
>     mode0    = mode state0
>     leds0    = leds state0
>     -- intermediate state computation
>     counter1 | counter0 == maxBound = 0
>              | otherwise            = succ counter0
>     mode1 | key1R     = complementLedMode mode0
>           | otherwise = mode0
>     leds1 | counter0 == 0 = case mode0 of
>                                 Complement -> complement leds0
>                                 Rotate     -> rotateL leds0 1
>           | otherwise     = leds0
>     -- next state computation
>     state1 = BlinkerState { leds = leds1, mode = mode1, counter = counter1 }

paddytheplaster, 2019-10-08:
The type of the counter attribute should be 'Index 16650000'...

leonschoorl, 2019-10-08:
When posting code on github you can mark it as (haskell) code by wrapping it in tripple backticks.
And if you write the name of the language after the opening ` ``` ` it'll even do syntax highlighting.

    ```haskell
    module Blinker where
    topEntity = ...
    ```

leonschoorl, 2019-10-08:
I can find a couple of Blinker modules all of them seem different from your code above:

1. [Blinker.hs](https://github.com/clash-lang/clash-compiler/blob/master/examples/Blinker.hs) in the examples directory 
2. Blinker in the [Tutorial] after "For example, given the following specification:"
3. Blinker in the [Tutorial] after "Blinker circuit in Clash 0.99:"
4. Blinker in the [Tutorial] after "Blinker in Clash 1.0:"

(1) seems is ok to me
(2) is missing a couple of `@`'s
```diff
-    (pllOut,pllStable) = altpll Dom50 (SSymbol "altpll50") clk (unsafeFromLowPolarity rst)
+    (pllOut,pllStable) = altpll @Dom50 (SSymbol @"altpll50") clk (unsafeFromLowPolarity rst)
```
(4) Is trying to derive `Undefined`, but this was renamed to `NFDataX`
```diff
-   deriving (Generic, Undefined)
+   deriving (Generic, NFDataX)
```

[Tutorial]: http://hackage.haskell.org/package/clash-prelude-1.0.0/docs/Clash-Tutorial.html

paddytheplaster, 2019-10-09:
@leonschoorl Thanks for that. When I reported this this didn't work for the Clash 1.0 installation I downloaded. After reading your comment I compiled it with .nix.clash and it works fine after chaning 'Undefined' to 'NFDataX'.

closed 2019-10-09 by paddytheplaster