clash-lang/clash-compiler #834: Synthesis annotations on signals internal to a function don't appear in generated verilog

adamwalker, 2019-10-05:
When implementing stuff in legacy HDLs, I often add the attribute (*mark_debug = "true"*) to signals I want to observe in chipscope. I then collect all these signals and connect them up to an ILA in my top level TCL compilation script. 

I would like to achieve the same thing in Clash. I attempted to do this with the following code:

```
import Clash.Prelude
import Clash.Annotations.SynthesisAttributes

topEntity' 
    :: forall dom
    .  HiddenClockResetEnable dom
    => Signal dom (BitVector 16)
    -> Signal dom (BitVector 16)
topEntity' x = y + 1
    where
    y :: Signal dom (BitVector 16) `Annotate` 'StringAttr "mark_debug" "true"
    y = x + 1

topEntity = exposeClockResetEnable @System topEntity'
```

However, the generated verilog does not contain any signals with the "mark_debug" attribute. Am I doing something wrong? Note that I am adding annotations to local signals within a function. The documentation for annotations shows annotations only on Signals that are in a function's type signature. Is what I'm trying to do supported?

christiaanb, 2019-10-05:
It should work. So we can mark this as a bug

christiaanb, 2020-01-09:
It seems that GHC does the inlining already, because when you add

```haskell
{-# NOINLINE y #-}
``` 

inside the where clause, the annotation is preserved.