clash-lang/clash-compiler #836: Generate warning when `toInteger` blackbox drops MSBs

christiaanb, 2019-10-09:
Clash "erroneously" translates Haskell's `Integer` to 64-bit signed number; even when the context demands more. Properly translating `Integer` has proven non-trivial, so there's no easy path forward; outright banning `Integer` in Clash designs in intractable because of how ubiquitous it is in Haskell.

This issue mostly shows up when users are converting from one type to the other "through" `Integer`, e.g. using `fromIntegral`. This works when the source type is less than 64 bit, but drops MSBs when the source type is more than 64 bit. This conversion to `Integer` happens in the `toInteger` primitives. 

So Clash should issue a warning when these primitives are used in a "more-than-64-bit" context. The warning will show up late in the synthesis process, which is regrettable; but it's better than no warning at all!!

rowanG077, 2019-10-09:
An alternative is to generate 64 `safe` instances conversion instances for the affected types and a single instance which does the other ones which will issue the warning. Something like:

```
class SafeToInteger a where
  safeToInteger :: a -> Integer

instance SafeToInteger (Unsigned 1) where
  safeToInteger = toInteger#

instance SafeToInteger (Unsigned 2) where
  safeToInteger = toInteger#

instance SafeToInteger (Unsigned 3) where
  safeToInteger = toInteger#

instance {-# INCOHERENT #-} (KnownNat n, 4 <= n) => SafeToInteger (Unsigned n) where
  safeToInteger = unsafeToInteger

{-# WARNING unsafeToInteger "This is unsafe; I hope you know what you're doing" #-}
unsafeToInteger :: (KnownNat n) => Unsigned n -> Integer
unsafeToInteger = toInteger#
```

It's really ugly though but it means you do get the error message prior to synthesis.

paddytheplaster, 2019-10-10:
Thanks Christiaan.

I think it would be nice if  there was a special flag which gave the user the option to force a run-time error when the argument of `fromInteger` has a `BitSequence` representation which requres more than 64 bits. It may also be useful to give the user the option to cache such warning messages in a fie.

Just a thought.

Regards,

Paddy

commit ce5c78a2 (2020-01-17): toInteger function warn when dropping MSBs

Fixes #836
commit cc3241a8 (2020-01-17): toInteger function warn when dropping MSBs

Fixes #836
commit 0f60d417 (2020-01-17): toInteger function warn when dropping MSBs

Fixes #836
PR #1018 (2020-01-17): toInteger function warn when dropping MSBs
Fixes #836
closed 2020-01-17 by christiaanb