clash-lang/clash-compiler #841: Inconsistent language flags when using Haskell InlinePrimitive

blaxill, 2019-10-13:
This is as minor issue as language flags can be enabled in the usual manner.

Clash enables by default a number of language flags (e.g. DataKinds), however when using a haskell inlineprimitive, aren't enabled during the blackbox compilation pass.

e.g. `clash-compiler/tests/shouldwork/BlackBox/BlackBoxFunction.hs` currently requires ```{-# LANGUAGE DataKinds         #-}```

Without the language flags blackbox compilation will fail e.g. (different source)

~~~
GHC: Parsing and optimising modules took: 21.590s
GHC: Loading external modules from interface files took: 6.944s
GHC: Parsing annotations took: 0.054s

<no location info>: error:
    Clash error call:
    Encountered compilation errors while compiling blackbox template function "BBFN<Top.testTF>" for function "Top.testTF". Compilation reported:

src/Axi.hs:32:20: error:
    Illegal type: ‘"tvalid"’ Perhaps you intended to use DataKinds
~~~

christiaanb, 2019-11-11:
We probably need to add the additional language flags somewhere in: https://github.com/clash-lang/clash-compiler/blob/b359954da82e5a8265ea56d554f86ba3ab26f451/clash-lib/src/Clash/Driver.hs#L311-L358

These are the language flags that are normally enabled by default: https://github.com/clash-lang/clash-compiler/blob/655c0d748c118dd05f9480126ba60f5ecd19d02b/clash-ghc/src-ghc/Clash/GHC/LoadModules.hs#L620-L658

[DynFlags](http://hackage.haskell.org/package/ghc-8.6.5/docs/DynFlags.html#t:DynFlags) has an `extensions` field, with all the set and unset language flags, which we can probably pass to the call of `Clash.Driver.generateHDL` in https://github.com/clash-lang/clash-compiler/blob/655c0d748c118dd05f9480126ba60f5ecd19d02b/clash-ghc/src-bin-881/Clash/GHCi/UI.hs#L2143-L2154 (and similarly for `src-bin-861` and `src-bin-841`)

alex-mckenna, 2019-11-14:
@blaxill, can you explain what you do to get this error? I've tried running `clash-testsuite`, `clash` and `clashi` after removing the `{-# LANGUAGE DataKinds #-}` line from `BlackBoxFunction.hs` and I don't get any errors

blaxill, 2019-11-15:
Hmm, you're right that BlackBoxFunction.hs works- it seems DataKinds now is included. However I can reproduce the issue with other missing flags:

1. Save the following as `src/Blackboxes.hs`

~~~Haskell
{-# LANGUAGE BangPatterns        #-}
{-# LANGUAGE MagicHash           #-}
{-# LANGUAGE TypeApplications    #-}
{-# LANGUAGE RankNTypes    #-}

module Blackboxes where

import           Clash.Prelude
import           Language.Haskell.TH
import           Language.Haskell.TH.Syntax
import Clash.Annotations.Primitive
import Clash.Netlist.Types
import GHC.Generics

data XilinxDiffI (domain_aiVi :: Domain)
  = XilinxDiffI {_I :: (Clock domain_aiVi),
                 _IB :: (Clock domain_aiVi)}
  deriving GHC.Generics.Generic
data XilinxDiffO (domain_aiVi :: Domain)
  = XilinxDiffO {_O :: (Clock domain_aiVi)}
  deriving GHC.Generics.Generic
xilinxDiffI ::
  forall (domain_aiVi :: Domain).
  KnownDomain domain_aiVi =>
  Clock domain_aiVi
  -> Clock domain_aiVi
     -> XilinxDiffI domain_aiVi
xilinxDiffI clkArg0 clkArg1 = (XilinxDiffI clkArg0) clkArg1
xilinxDiff# ::
  forall (domain_aiVi :: Domain).
  KnownDomain domain_aiVi =>
  Clock domain_aiVi
  -> Clock domain_aiVi
     -> XilinxDiffO domain_aiVi
xilinxDiff# !_ !_ = XilinxDiffO clockGen
{-# NOINLINE xilinxDiff# #-}
xilinxDiff ::
  forall (domain_aiVi :: Domain).
  KnownDomain domain_aiVi =>
  XilinxDiffI domain_aiVi -> XilinxDiffO domain_aiVi
xilinxDiff (_arg) = (xilinxDiff# (_I _arg)) (_IB _arg)
{-# NOINLINE xilinxDiff #-}
xilinxDiffTF :: Clash.Netlist.Types.TemplateFunction
xilinxDiffTF = undefined

{-# ANN xilinxDiff# (Clash.Annotations.Primitive.InlinePrimitive
                                  Clash.Annotations.Primitive.Verilog
                                 " [{ \"BlackBox\" :\
  \          { \"name\" : \"Blackboxes.xilinxDiff#\"\
  \          , \"kind\": \"Declaration\"\
  \          , \"format\": \"Haskell\"\
  \          , \"templateFunction\" : \"Blackboxes.xilinxDiffTF\"\
  \          }\
  \      }]\
  \  ") #-}

topEntity :: Clock System
topEntity = _O $ xilinxDiff# (clockGen @System) (clockGen @System) 
~~~

1. `clashi -isrc src/Blackboxes.hs` loads without error and can interact with the types/functions as usual.

1.  `clash --verilog -isrc src/Blackboxes.hs`, generates errors during template compilation due to missing language flags:

~~~
GHC: Parsing and optimising modules took: 3.604s
GHC: Loading external modules from interface files took: 0.567s
GHC: Parsing annotations took: 0.001s

<no location info>: error:
    Clash error call:
    Encountered compilation errors while compiling blackbox template function "BBFN<Blackboxes.xilinxDiffTF>" for function "Blackboxes.xilinxDiff#". Compilation reported:

src/Blackboxes.hs:24:34: error:
    Illegal kind signature: ‘Domain’
      Perhaps you intended to use KindSignatures
    In the data type declaration for ‘XilinxDiffI’

src/Blackboxes.hs:28:34: error:
    Illegal kind signature: ‘Domain’
      Perhaps you intended to use KindSignatures
    In the data type declaration for ‘XilinxDiffO’

src/Blackboxes.hs:32:26: error:
    Illegal kind signature: ‘Domain’
      Perhaps you intended to use KindSignatures
    In the type signature for ‘xilinxDiffI’

src/Blackboxes.hs:39:26: error:
    Illegal kind signature: ‘Domain’
      Perhaps you intended to use KindSignatures
    In the type signature for ‘xilinxDiff#’

src/Blackboxes.hs:47:26: error:
    Illegal kind signature: ‘Domain’
      Perhaps you intended to use KindSignatures
    In the type signature for ‘xilinxDiff’

...
~~~

Note [KindSignatures](https://github.com/clash-lang/clash-compiler/blob/eca91a0dc4d64cba8f58c760bcd88e8baf431640/clash-ghc/src-ghc/Clash/GHC/LoadModules.hs#L637) is enabled by Clash.

alex-mckenna, 2019-11-18:
Hey @blaxill, I've got [a branch](/clash-lang/clash-compiler/tree/inline-primitive-flags) which should fix things for you. Do you mind checking whether it fixes things for you (it works for the example you sent me, but I'm aware you were using a different file initially).

alex-mckenna, 2019-11-18:
Update: Turns out @leonschoorl was working on this issue too, and his solution is technically better. So I'm going to delete the branch I just pushed in favour of GH-924

alex-mckenna, 2019-11-20:
Update: Fix in master from 9fe285916ded674d7ce2994c96cdbfd0d36621ac

closed 2019-11-20 by alex-mckenna