clash-lang/clash-compiler #867: View pattern definition of :> doesn't play nicely with typechecker

gergoerdi, 2019-10-23:
(This is all with Clash 1.0.0)

I think the following program should typecheck:

```
import Clash.Prelude

foo :: (1 <= n, KnownNat n) => Vec n a -> Vec n a
foo = id

bar :: (KnownNat n) => Vec n a -> Vec n a
bar Nil = Nil
bar xs@(_ :> _) = foo xs
```

but it doesn't, because:

```
/tmp/a.hs:16:10: error:
    • Couldn't match type ‘n’ with ‘n0 + 1’
      ‘n’ is a rigid type variable bound by
        the type signature for:
          bar :: forall (n :: Nat) a. KnownNat n => Vec n a -> Vec n a
        at /tmp/a.hs:14:1-42
      Expected type: Vec n a
        Actual type: Vec (n0 + 1) a
    • In the pattern: _ :> _
      In an equation for ‘bar2’: bar2 xs@(_ :> _) = foo xs

/tmp/a.hs:16:20: error:
    • Couldn't match type ‘1 <=? n’ with ‘'True’
        arising from a use of ‘foo’
    • In the expression: foo xs
      In an equation for ‘bar’: bar xs@(_ :> _) = foo xs

```

The same code works when using `Cons` directly:

```
bar1 :: (KnownNat n) => Vec n a -> Vec n a
bar1 Nil = Nil
bar1 xs@Cons{} = foo xs
```

and moreover, it works if I define a `(:>)`-like pattern synonym, but using `Cons` directly instead of `head` and `tail`:

```
pattern x ::> xs = Cons x xs

bar2 :: (KnownNat n) => Vec n a -> Vec n a
bar2 Nil = Nil
bar2 xs@(_ ::> _) = foo xs
```

gergoerdi, 2019-10-23:
Maybe this is all caused by something more sinister, in the patsyn typechecker, because I noticed that if I give the above definition of `(::>)` a type signature, then `bar2` suddenly breaks down:

```
pattern (::>) :: a -> Vec n a -> Vec (n + 1) a
pattern x ::> xs = Cons x xs

bar2 :: (KnownNat n) => Vec n a -> Vec n a
bar2 Nil = Nil
bar2 xs@(_ ::> _) = foo xs
```

```
/tmp/a.hs:28:10: error:
    • Couldn't match type ‘n’ with ‘n0 + 1’
      ‘n’ is a rigid type variable bound by
        the type signature for:
          bar2 :: forall (n :: Nat) a. KnownNat n => Vec n a -> Vec n a
        at /tmp/a.hs:26:1-42
      Expected type: Vec n a
        Actual type: Vec (n0 + 1) a
    • In the pattern: _ ::> _
      In an equation for ‘bar2’: bar2 xs@(_ ::> _) = foo xs

/tmp/a.hs:28:21: error:
    • Couldn't match type ‘1 <=? n’ with ‘'True’
        arising from a use of ‘foo’
    • In the expression: foo xs
      In an equation for ‘bar2’: bar2 xs@(_ ::> _) = foo xs
```

martijnbastiaan, 2019-10-23:
```
*Main> :t (::>)
(::>) :: b -> Vec n b -> Vec (n + 1) b
```

Weird..

christiaanb, 2019-10-23:
Use this:

```
pattern (::>) :: () => (m ~ (n + 1)) => a -> Vec n a -> Vec m a
```

blaxill, 2019-10-23:
To add to Christiaan’s reply, there is some explanation here https://stackoverflow.com/questions/51456200/unable-to-match-haskell-type <https://stackoverflow.com/questions/51456200/unable-to-match-haskell-type>

gergoerdi, 2019-10-23:
Oh wow, hoisted by my own petard... I guess it was only a matter of time before the crazy-complicated patsyn typesig syntax catches up with me :)

But shouldn't the "real" `(:>)` patsyn's type be `() => (n1 ~ (n + 1)) => a -> Vec n a -> Vec n1 a` as well, then?

christiaanb, 2019-10-23:
Yes, in hindsight, it should have been:

```
pattern (:>) :: forall n a . () => forall m . (n ~ (m + 1)) => a -> Vec m a -> Vec n a
```

christiaanb, 2019-10-23:
However, we can only give it that signature when we change the implementation to:

```haskell
pattern (:>) :: forall n a . () => forall m . (n ~ (m + 1)) => a -> Vec m a -> Vec n a
pattern (:>) x xs <- Cons x xs
  where
    (:>) x xs = Cons x xs
```

Which we can do (as in synthesize to HDL) these days because we have type-based case-alternative elimination for case-expressions matching on GADTs

christiaanb, 2019-10-23:
Actually, the old implementation would have worked with a slightly more constrained signature:

```
pattern (:>) :: forall n a . (1 <= n) => forall m . (n ~ (m + 1)) => a -> Vec m a -> Vec n a
pattern (:>) x xs <- ((\ys -> (head @(n-1) ys, tail @(n-1) ys)) -> (x,xs))
  where
    (:>) x xs = Cons x xs
```

But I suggest we change the implementation as suggested in my earlier comment:

```
pattern (:>) :: forall n a . () => forall m . (n ~ (m + 1)) => a -> Vec m a -> Vec n a
pattern (:>) x xs <- Cons x xs
  where
    (:>) x xs = Cons x xs
```

When we do that, we probably also want to change the signature and implementation for `(:<)`

```haskell
pattern (:<) :: forall n a . (1 <= n) => forall m . (n ~ (m + 1)) => Vec m a -> a -> Vec n a
pattern (:<) xs x <- ((\ys -> (init @(n-1) ys,last @(n-1) ys)) -> (xs,x))
  where
    (:<) xs x = xs ++ singleton x
```

gergoerdi, 2019-10-23:
At that point, why not just an implicitly bidirectional pattern synonym?

```
pattern (:>) :: forall n a . () => forall m . (n ~ (m + 1)) => a -> Vec m a -> Vec n a
pattern (:>) x xs = Cons x xs
```

christiaanb, 2019-10-23:
@gergoerdi yes, much simpler.

Also, for the snoc `(:<)`, this is the implementation that we want:

```haskell
pattern (:<) :: forall n a . () => forall m . (n ~ (m + 1)) => Vec m a -> a -> Vec n a
pattern (:<) xs x <- (snocPattern -> Cons x xs)
  where
    (:<) xs x = xs ++ singleton x

snocPattern :: Vec n a -> Vec n a
snocPattern Nil = Nil
snocPattern xs@(Cons _ _) = Cons (last xs) (init xs)
{-# INLINE snocPattern #-}
```

otherwise you can have run-time crashes on `Nil`. Also, I tested this with:

```haskell
foo :: Vec n a -> Bool
foo xs@(_ :< _) = True
foo Nil = False
```

and it does the right thing:

```
*Main> foo Nil
False
*Main> foo (1 :> Nil)
True
```

alex-mckenna, 2019-11-13:
Update: The pattern synonyms are changed in the [vec-patterns-fix](/clash-lang/clash-compiler/tree/vec-patterns-fix) branch, but the doctests for the patterns are currently failing due to the fact `Vec` is a GADT (and has untouchable type parameters). That means lines like

```haskell
f (x :> y :> _) = x + y       -- Example from doctest
g (Cons x (Cons y _)) = x + y -- Changed to not use pattern synonym
```
do not work without a type signature. For the functions in the vector module this isn't a problem, but so far `doctest` seems to be balking when I try to specify the type signature for `f` in the comment.

alex-mckenna, 2019-11-13:
Update: Things now seem to be working locally for me. If this next commit passes then the new patterns can be merged into master :tada:

alex-mckenna, 2019-11-19:
Reopening to address https://github.com/clash-lang/clash-compiler/pull/912#issuecomment-555273464 (and while I'm here fix the accidental change to the copyright notice from before @martijnbastiaan)

commit 0db91924 (2019-11-13): [Closes #867] Actually fix doctests

It seems that because the type of element in a Vec is existentially
bound, type inference cannot tell a function that returns vector
elements is the same type. The doctests now have very ugly let
bindings in the form

  let x :: T; x = ...

to allow toplevel functions to be defined in doctests without the
test crashing. This shouldn't be an issue in local bindings within
functions (as shown by the fact functions in the vector module have
been changed to use the patterns where appropriate), so provided
functions using vectors have type annotations things sho [... cut]
commit 060ce16b (2019-11-13): [Closes #867] Fix Vector pattern synonyms

The pattern synonyms for Vector now have more accurate types (as
pattern synonyms have two types of contraint) which allow them to
be used in cases where you would expect but were previously
not possible.
commit eca91a0d (2019-11-13): Merge pull request #912 from clash-lang/vec-patterns-fix

[Closes #867] Fix Vector pattern synonyms
commit 2060671b (2019-11-21): Merge pull request #926 from clash-lang/fix-vec-pattern-synonym-2

Fix type of (:<), add COMPLETE pragmas. Fixes #867.
closed 2019-11-21 by alex-mckenna