clash-lang/clash-compiler #901: mkPluginUsage: missing plugin library

rikstrijker, 2019-11-04:
Running Clashi snap 1.1.0 on Ubuntu 16.04 LTS can sometimes give a error when reloading a project . The error doesn't appear after restarting Clashi, can't reproduce it after a restart. See error below:    
```
clashi: panic! (the 'impossible' happened)
  (GHC version 8.6.4 for x86_64-unknown-linux):
	mkPluginUsage: missing plugin library, tried:
/usr/nix/store/dqn8v2rc66drcq4g889zvspdlc837fpf-ghc-typelits-natnormalise-0.6.2/lib/ghc-8.6.4/x86_64-linux-ghc-8.6.4/libHSghc-typelits-natnormalise-0.6.2-3a1zYTACyx9KnKzWCmE41L-ghc8.6.4.so
/usr/nix/store/q0mmcb09p75njdsskpy9492082gz1s43-ncurses-6.1-20190112/lib/libHSghc-typelits-natnormalise-0.6.2-3a1zYTACyx9KnKzWCmE41L-ghc8.6.4.so
/usr/nix/store/g3q0vlabp3zx0p2j8kjs3qy0mbyni8k5-libffi-3.2.1/lib/libHSghc-typelits-natnormalise-0.6.2-3a1zYTACyx9KnKzWCmE41L-ghc8.6.4.so
/usr/nix/store/50ln29n99svzipn4bxrq1bdkjm7rc0s9-gmp-6.1.2/lib/libHSghc-typelits-natnormalise-0.6.2-3a1zYTACyx9KnKzWCmE41L-ghc8.6.4.so
/usr/nix/store/ca2w6fda4li8jn8n8jb7l92d6k9n74jl-ghc-8.6.4/lib/libHSghc-typelits-natnormalise-0.6.2-3a1zYTACyx9KnKzWCmE41L-ghc8.6.4.so

  GHC.TypeLits.Normalise
  Call stack:
      CallStack (from HasCallStack):
        callStackDoc, called at compiler/utils/Outputable.hs:1160:37 in ghc:Outputable
        pprPanic, called at compiler/deSugar/DsUsage.hs:193:11 in ghc:DsUsage

Please report this as a GHC bug:  http://www.haskell.org/ghc/reportabug
```

leonschoorl, 2020-03-03:
We did some investigating on this when it happened again today.
Inside of the snap namespace the `/usr/nix/store` directory was completely missing.
`/usr/nix` should be a bind mount of `$SNAP/usr/nix`
It seems that something had unmounted this.

@rikstrijker next time this happens you could try to it remount with:
```
sudo nsenter -t `pidof clashi` -m   mount --bind /snap/clash/current/usr/nix /usr/nix
```
(this assumes you only have one instance of `clashi` running)

martijnbastiaan, 2020-03-04:
@rikstrijker is running the `--edge` version of clash that's updated every day. Snap auto updates, so maybe it has something to do with that?

martijnbastiaan, 2020-03-04:
We just checked and that's indeed what happens :tada: . While updating the clash snap, snap umounts layout binds. 

----------

While this is a nice hack:

```bash
sudo nsenter -t `pidof clashi` -m   mount --bind /snap/clash/current/usr/nix /usr/nix
```

I believe this will mount the version of the newly installed Clash!

martijnbastiaan, 2021-03-19:
Snap images produced from now on don't use layout binds anymore, solving this issue.

closed 2021-03-19 by martijnbastiaan