clash-lang/clash-compiler #903: Improve Test Suite in clash-cores

alex-mckenna, 2019-11-04:
The test suite in `clash-cores` currently runs a lot of similar tests in a manner which is

  * very repetitive, making it harder to read / write cases
  * potentially very specific, and could be more general using QuickCheck for instance

These tests should be refactored to address these problems now that clash-cores lives in the main repository.

martijnbastiaan, 2019-11-05:
I think @christiaanb was talking about this monster:

https://github.com/clash-lang/clash-compiler/blob/master/testsuite/Main.hs

christiaanb, 2019-11-05:
@martijnbastiaan No I was talking about the clash-cores one.

martijnbastiaan, 2019-11-05:
@christiaanb Too bad :P. I'll add another issue for the main testsuite.

commit 643e7fed (2019-11-06): Merge pull request #909 from clash-lang/clash-cores-test-refactor

Close #903 - Refactor Tests in clash-cores
closed 2019-11-06 by alex-mckenna