clash-lang/clash-compiler #905: Add documentation for using libraries with snapclash

martijnbastiaan, 2019-11-04:
A user on IRC asked:

```
davos> Hello, I am trying to use indexed monads with clash. I installed the package indexed from hackage (cabal install indexed), then I run clash.clashi. However, when I try to import Control.Monad.Indexed, I get the error Could not find module Contro.Monad.Indexed. I installed clash (1.0.1) via snap. I am running Ubuntu 18.04.3 and installed cabal via sudo apt install cabal (v 1.24). Is it possible to do this with clash
04:08 from snap? Any guidance much appreciated.
04:09 Also ghci has no problem importing Control.Monad.Indexed, so im sure its installed.
```

We should figure out what to do here.

dopamane, 2020-01-18:
Hello, I am davos on irc. My original question is no longer an issue for me since I switched to using cabal. With cabal I am now able to use the indexed package along with many others on hackage. In addition I have been begun packaging my own clash libraries using cabal. 

This works well, however the process was not smooth; most likely due to my inexperience with cabal. A major pain point was addressed here #677 (I needed to use those flags!) but I believe it should be documented elsewhere. Perhaps a wiki on making clash libraries with cabal and stack? Another smaller issue was solved by using the extensions NoStarIsType and TypeOperators; cabal complained about the `m*n` in my function:
`expandMask :: (KnownNat n, KnownNat m) => BitVector n -> BitVector (m*n)`. 
Using MagicHash was another gotcha moment with cabal for functions like `concatBitVector#`. Most every other issue cabal guided me.

That being said, unless there is some way to use packages with the snap binary, I am OK with closing this issue.

leonschoorl, 2020-01-21:
@dopamane We have a whole list of extensions that are automatically enabled/disabled inside clash(i) to make things work more smoothly.
If you use other external tools(ghc,haddock,...) you might have to specify some or all of these to get the same behaviour as in clash(i): 
https://github.com/clash-lang/clash-compiler/blob/398c4ad8abe135fca94625ff3581e8ac1d12adf2/clash-lib/src/Clash/Util.hs#L458-L494

alex-mckenna, 2020-01-21:
For those interested, these things will be documented publicly when #987 is merged.

leonschoorl, 2020-01-23:
@alex-mckenna In the case of snap it's not just a documentation thing.
I think we have to include a copy of cabal-install inside the snap image and then document how to use it.

alex-mckenna, 2020-01-23:
When I was talking documentation I was referring specifically to the language extensions Clash uses / expects by default. Not suggesting a fix for anything, just saying that it should soon be more transparent (compared to having to dig through the source to find the required extensions)

martijnbastiaan, 2020-11-09:
This has been solved by the [Clash starter project](https://github.com/clash-lang/clash-compiler/tree/master/clash-starters/clash-example-project).

closed 2020-11-09 by martijnbastiaan