clash-lang/clash-compiler #907: Improve testsuite

martijnbastiaan, 2019-11-05:
[Main.hs](https://github.com/clash-lang/clash-compiler/blob/master/testsuite/Main.hs) is "not as good as it could be". Ideally, I'd like to have it look something like: [this1](https://github.com/clash-lang/clash-compiler/blob/ce7119ce1bed1999e98f527f5aad2217c6745f55/testsuite/Test/Tasty/Clash.hs#L32-L43) [this2](https://github.com/clash-lang/clash-compiler/blob/ce7119ce1bed1999e98f527f5aad2217c6745f55/testsuite/Main.hs#L47).

martijnbastiaan, 2019-11-07:
Partly solved in https://github.com/clash-lang/clash-compiler/pull/910

alex-mckenna, 2021-08-04:
I'm going to close this because I think between the splitting out the testing programs into their own modules / using `IsTest`, and the recent testsuite refactoring that was merged, things look considerably better than they did when this issue was raised. If there are other testsuite issues, they should get their own more specific issues

closed 2021-08-04 by alex-mckenna