clash-lang/clash-compiler #908: Send Acknolwedge Signals in SPI

alex-mckenna, 2019-11-06:
Currently when SPI transmits data, it does not send an acknowledge signal to say that the underlying source of the data being sent is free to be changed again. This signal should be added, and the test suite extended to show that data can be changed only after the acknowledge signal has been sent.

alex-mckenna, 2019-11-07:
850e2b93 has added an acknowledgement signal which appears to intermittently set as expected, however it still needs to be tested for correctness. At a guess, I would assume tests should be able to show things like

  * every data transfer has exactly one acknowledgement at the start
  * data input does not change until after acknowledgement
  * multi-word messages send one acknowledgement for every word in the message

CC: @christiaanb

christiaanb, 2019-11-07:
@alex-mckenna yes, that test plan sound good

alex-mckenna, 2019-11-11:
The acknowledge signal is in, and the tests specific to that one signal are implemented to what should hopefully be an acceptable standard. To achieve full feature parity with the lattice SPI reference I've been looking at, an error signal will be added (to be set high for one cycle if the data is changed before an acknowledge signal). The prerequisites in the test suite to write tests for this already exist.

commit 21c465ad (2019-11-19): Merge pull request #916 from clash-lang/clash-cores-acknowledge-signal

Add acknowledge signal to SPI. Closes #908.
closed 2019-11-19 by alex-mckenna