clash-lang/clash-compiler #913: PortProduct prefix used as internal signal name

DigitalBrains1, 2019-11-14:
When using a `Synthesize` annotation to name input or output ports, writing `PortProduct "ext_df"` will not only prefix all port names in the product with `ext_df_`, but it will also use the name `ext_df` for the internal signal/wire to which the individual ports are aggregated.

This means the following:
```haskell
{-# ANN topEntity
   (Synthesize
      { t_name   = "f"
      , t_inputs = [ PortProduct "ext_df"
                                 [ PortName "o_rdy"
                                 , PortName "r_vld"
                                 , PortName "r" ]
                                 ]
      , t_output = PortProduct "ext_df"
                               [ PortName "r_rdy"
                               , PortName "o_vld"
                               , PortName "o"
                               ]
      }) #-}
topEntity :: ( Signal System Bool
             , Signal System Bool
             , Signal System (Unsigned 8)
             )
          -> ( Signal System Bool
             , Signal System Bool
             , Signal System (Unsigned 8)
             )
topEntity (oReady, rValid, r) = (pure True, pure False, pure undefined)
```

will create code where two separate signals/wires are called "ext_df", once for the product in the input and once for the product in the output. These signals are not part of the ports of the component, but purely for internal use. Synthesis subsequently fails because the signal names clash.

It doesn't seem to serve a purpose to be able to name an internal signal. I can see two fixes:

- Fully backward compatible: only change the name of the internal signals when they would clash otherwise, keep the specified name if not. This means existing, working code produces the exact same result. Existing code with name clashes would not have worked.
- Never use the "prefix" as the internal signal name, just always use an autogenerated one. Explicitly naming internal signals seems like a feature without a purpose. It will change code generation for existing designs, but is anyone really depending on the name of an internal signal?

If the name should only be changed when a naming clash is about to occur, this could also be implemented by beforehand inspecting all the PortProducts in the annotation and see if they lead to the same prefix being used in multiple places (but mind nested PortProducts).

leonschoorl, 2019-11-14:
I think the first option makes most sense, as it the resulting HDL nicer to read.

And also [`mkUniqueIdentifier`](https://github.com/clash-lang/clash-compiler/blob/eca91a0dc4d64cba8f58c760bcd88e8baf431640/clash-lib/src/Clash/Netlist/Util.hs#L890) already takes a name on which to base the unique identifier.

martijnbastiaan, 2020-02-18:
Fixed by https://github.com/clash-lang/clash-compiler/pull/1059, thanks for the report!

closed 2020-02-18 by martijnbastiaan