clash-lang/clash-compiler #918: X-Optimization in Clash Core

alex-mckenna, 2019-11-15:
Currently, Clash leaves all X-optimisation to the logic synthesis tools, including things like case-expressions where all but one branch evaluates to X. The compiler should, either as a separate pass, or as part of an existing pass, eliminate/transform such case-expressions into the non-X branch. Additionally, the WHNF (soon to be NF) evaluator should correctly implement the X-propagation rules for the existing operators.

alex-mckenna, 2019-11-15:
Update: Offline discussion with @leonschoorl was that there should be a test in `clash-testsuite` that looks at the generated netlist and counts the number of multiplexers that appear (as this is significantly better than just checking that a specific assignment statement appears).

martijnbastiaan, 2019-11-15:
@alex-mckenna The `Component` in the output of [`genNetlist`](https://github.com/clash-lang/clash-compiler/blob/4eae5faef2b23882f7669206c2ab6761aa8026d8/clash-lib/src/Clash/Netlist.hs#L78) should be what you're looking for. It will [contain `Declaration`s](https://github.com/clash-lang/clash-compiler/blob/4eae5faef2b23882f7669206c2ab6761aa8026d8/clash-lib/src/Clash/Netlist/Types.hs#L154), in turn housing [CondAssignment](https://github.com/clash-lang/clash-compiler/blob/4eae5faef2b23882f7669206c2ab6761aa8026d8/clash-lib/src/Clash/Netlist/Types.hs#L242-L248). 

I'm not entirely convinced this will be less fragile than just checking the textual output, but time will have to tell.

alex-mckenna, 2019-11-15:
Ja, it might not be _much_ stronger as a test, but at least it doesn't rely on details like internal name generation not changing

commit 9bdc84d9 (2019-11-26): Merge pull request #935 from clash-lang/feature-X-optimization

X Optimization. Closes #918
closed 2019-11-26 by martijnbastiaan