clash-lang/clash-compiler #919: Clash generating invalid Verilog after Vec operations

blaxill, 2019-11-15:
[This gist](https://gist.github.com/blaxill/9f5cb56dd35ee30c50ecaf103a5d7d6a) when compiled with `clash --verilog test.hs` creates [this output](https://gist.github.com/blaxill/277ce3b1d2834dc291f016f7549ff259). The output contains case statements starting line 162, which iverlog and vivado reject as erroneous (yosys seems to accept it) 

~~~verilog
    wire [3:0] c$case_alt_2;
        always @(*) begin
          case(ds4[33:32])
            2'b01 : acc_1_0 = c$case_alt_2;
            default : acc_1_0 = c$case_alt_2;
          endcase
        end

        assign c$case_alt_2 = acc_0_0[3:3] ? acc_0_0 : acc_0_1;
~~~

~~~
ben> iverilog -t null verilog/Search/topEntity.v
verilog/Search/topEntity.v:162: error: acc_1_0 is not a valid l-value in topEntity.
verilog/Search/topEntity.v:133:      : acc_1_0 is declared here as wire.
verilog/Search/topEntity.v:163: error: acc_1_0 is not a valid l-value in topEntity.
verilog/Search/topEntity.v:133:      : acc_1_0 is declared here as wire.
verilog/Search/topEntity.v:174: error: acc_1_1 is not a valid l-value in topEntity.
verilog/Search/topEntity.v:134:      : acc_1_1 is declared here as wire.
verilog/Search/topEntity.v:175: error: acc_1_1 is not a valid l-value in topEntity.
verilog/Search/topEntity.v:134:      : acc_1_1 is declared here as wire.
verilog/Search/topEntity.v:186: error: acc_1_2 is not a valid l-value in topEntity.
verilog/Search/topEntity.v:135:      : acc_1_2 is declared here as wire.
verilog/Search/topEntity.v:187: error: acc_1_2 is not a valid l-value in topEntity.
verilog/Search/topEntity.v:135:      : acc_1_2 is declared here as wire.
verilog/Search/topEntity.v:198: error: acc_1_3 is not a valid l-value in topEntity.
verilog/Search/topEntity.v:136:      : acc_1_3 is declared here as wire.
verilog/Search/topEntity.v:199: error: acc_1_3 is not a valid l-value in topEntity.
verilog/Search/topEntity.v:136:      : acc_1_3 is declared here as wire.
verilog/Search/topEntity.v:210: error: acc_2_0 is not a valid l-value in topEntity.
verilog/Search/topEntity.v:137:      : acc_2_0 is declared here as wire.
verilog/Search/topEntity.v:211: error: acc_2_0 is not a valid l-value in topEntity.
verilog/Search/topEntity.v:137:      : acc_2_0 is declared here as wire.
verilog/Search/topEntity.v:222: error: acc_2_1 is not a valid l-value in topEntity.
verilog/Search/topEntity.v:138:      : acc_2_1 is declared here as wire.
verilog/Search/topEntity.v:223: error: acc_2_1 is not a valid l-value in topEntity.
verilog/Search/topEntity.v:138:      : acc_2_1 is declared here as wire.
verilog/Search/topEntity.v:234: error: acc_3_0 is not a valid l-value in topEntity.
verilog/Search/topEntity.v:124:      : acc_3_0 is declared here as wire.
verilog/Search/topEntity.v:235: error: acc_3_0 is not a valid l-value in topEntity.
verilog/Search/topEntity.v:124:      : acc_3_0 is declared here as wire.
14 error(s) during elaboration.
~~~

The source contains the following, and it appears both the pattern match on S16/S32 and also the `reverse` call are required for Clash to generate the suspicious verilog.

~~~Haskell
searchT (S32 t, buff, offset, first) (Nothing, Just x) = ((S32 t,slice d63 d40 x,offset+8,False), m)
 where
  m = fmap collate
    $ fold (<|>)
    $ imap match
    $ reverse
    $ fmap pack
    $ windows1d d4 $ bitCoerce @_ @(Vec _ U8) $ x ++# buff
  match i v | t == v && not (first && i < 3) = Just i
  match _ _ = Nothing
  collate i = offset + resize (pack i)

searchT (S16 t, buff, offset, first) (Nothing, Just x) = ((S16 t,slice d63 d40 x,offset+8,False), m)
 where
  m = fmap collate
    $ fold (<|>)
    $ imap match
    $ reverse
    $ fmap pack
    $ windows1d d2 $ bitCoerce @_ @(Vec _ U8) $ x ++# slice d23 d16 buff
  match i v | t == v && not (first && i == 0) = Just i
  match _ _ = Nothing
  collate i = offset + resize (pack i)
~~~

leonschoorl, 2019-11-15:
With the two alternative of the case being the same assignment, it should just generate that assignment unconditionally.

leonschoorl, 2019-11-15:
It produces the same kinds of useless case statements in VHDL, only in VHDL they're not errors.

But it also produces a another unrelated error in VHDL:
```vhdl
      with (to_unsigned(i_1,max_0(1,integer(ceil(log2(real(8))))))) select
        \c$case_alt_9\ <= std_logic_vector'("0" & "---") when "000",
                          std_logic_vector'("1" & (std_logic_vector(to_unsigned(i_1,max_0(1,integer(ceil(log2(real(8))))))))) when others;
```
Here ghdl can't determine the type of `to_unsigned`:
```
$ ghdl -a search_types.vhdl topentity.vhdl 
topentity.vhdl:332:13:error: function call type is not locally static
topentity.vhdl:332:13:error: array type must be locally static
ghdl:error: compilation error
```

leonschoorl, 2019-11-15:
Somehow running clash with the `-g` flag works around the original issue.

leonschoorl, 2019-11-18:
I've managed to reduce the testcase somewhat to:

```haskell
{-# LANGUAGE TypeApplications #-}
{-# LANGUAGE ApplicativeDo       #-}
{-# LANGUAGE BlockArguments      #-}
{-# LANGUAGE GADTs               #-}
{-# LANGUAGE NumericUnderscores  #-}
{-# LANGUAGE RankNTypes          #-}

module Search where

import           Clash.Prelude           hiding ( filter )

type U8 = BitVector 8
type U16 = BitVector 16
type U32 = BitVector 32
type U64 = BitVector 64

data SearchTarget = None | 
                    S16 U16 | S32 U32
  deriving (Generic, Eq, Show, ShowX, NFDataX)
type SearchState = (SearchTarget, BitVector 24, BitVector 32, Bool)
type SearchResult = BitVector 32

searchT
  :: SearchState
  -> (Maybe U64)
  -> (Maybe SearchResult)

searchT (S32 t, buff, offset, first) (Just x) = m
 where
  m = fmap collate
    $ fold (<|>)
    $ imap match
    $ take d2
    $ id @(Vec 8 (Vec 4 U8))
    $ windows1d d4 $ bitCoerce @_ @(Vec _ U8)
    $ x ++# buff
  match i v = Just i
  collate i = resize (pack i)

searchT (S16 t, buff, offset, first) (Just x) = m
 where
  m = fmap collate
    $ fold (<|>)
    $ imap match
    $ take d2
    $ id @(Vec 8 (Vec 2 U8))
    $ windows1d d2 $ bitCoerce @_ @(Vec _ U8)
    $ x ++# slice d23 d16 buff
  match i v = Just i
  collate i = resize (pack i)

searchT _ _ = Nothing

topEntity = searchT
```

christiaanb, 2019-11-18:
Does reverting https://github.com/clash-lang/clash-compiler/commit/c4f4e56892ecd595a7af0d15b45727d88f625b09 make it go through? If so, then this line might be the cause: https://github.com/clash-lang/clash-compiler/blob/master/clash-lib/src/Clash/Primitives/Sized/Vector.hs#L178, because it's forcing all intermediate results to be `wire` https://github.com/clash-lang/clash-compiler/blob/eca91a0dc4d64cba8f58c760bcd88e8baf431640/clash-lib/src/Clash/Primitives/Sized/Vector.hs#L104, while sometimes they might need to be `reg` (see the logic used behind `~OUTPUTWIREREG[N]` as used in e.g. https://github.com/clash-lang/clash-compiler/blob/master/clash-lib/prims/verilog/Clash_Sized_Vector.json#L111)

commit 94900e14 (2020-01-09): Use `reg` instead of `wire` when appropiate in `foldBBF`

Fixes #919
PR #994 (2020-01-13): Use `reg` instead of `wire` when appropiate in `foldBBF`
Fixes #919
commit 88bfad7c (2020-01-13): Use `reg` instead of `wire` when appropiate in `foldBBF`

Fixes #919
closed 2020-01-13 by christiaanb