clash-lang/clash-compiler #929: Bug in Type Checker

paddytheplaster, 2019-11-20:
I'd like to report a bug in the typechecker: Clash, version 1.0.1 (using clash-lib, version: 1.0.1).
I am attaching the code. (I tried including the code in a previous attempt to report the bug but for some reason a sequence of 'x' followed by '*' followed by 'x' was turned into 'xx'.

Regards,

Paddy
[BUG.txt](https://github.com/clash-lang/clash-compiler/files/3869478/BUG.txt)

christiaanb, 2019-11-21:
Right... it seems to do with the order in which the plugins are called.

christiaanb, 2019-11-21:
And that's because the natnormalise-plugin makes a bad call, and calling the extra-plugin earlier ensures that the natnormalise-plugin cannot mess up.

christiaanb, 2019-11-21:
@paddytheplaster as a worked-around until the fixed versions are distributed you can add

```
{-# OPTIONS_GHC -fplugin GHC.TypeLits.Extra.Solver #-}
```

to the affected files. At least, that worked for your example, but I can't give any guarantees it's going to be a work-around in any of your normal code.

closed 2019-11-21 by christiaanb