clash-lang/clash-compiler #932: Installation instructions are out of date/hard to find

TheZoq2, 2019-11-21:
As someone who is not very used to the haskell ecosystem, I had a pretty hard time installing clash 1.0. The clearest instructions I was able to find are the instructions in the FAQ section under the wiki.

However, they seem to install an ancient clash version due to the old resolver. After some trial and error, I managed to install the newest version by compiling from source using `stack install` and then using `--resolver lts-14.4` when running stack exec. Is that the prefered way to do things? If so, it would be nice to have the instructions updated, and perhaps more visible (on the website or README perhaps)

TheZoq2, 2019-11-21:
Never mind, I must be blind since I didn't notice the `isntall clash` button on the main page :)

It would still be nice to have the FAQ here updated though

martijnbastiaan, 2019-11-21:
Did you manage to find:

* https://clash-lang.org/install/
* https://github.com/clash-lang/clash-compiler#using-clash-from-source

? If you did, how do you find it lacking?

martijnbastiaan, 2019-11-21:
> Never mind, I must be blind since I didn't notice the isntall clash button on the main page :)

We'll make it a pop-up ;)

> It would still be nice to have the FAQ here updated though

I see, the FAQ is indeed horribly outdated..

TheZoq2, 2019-11-21:
> https://clash-lang.org/install/

See my update message :)

> https://github.com/clash-lang/clash-compiler#using-clash-from-source

Huh, somehow I managed to miss that as well :sweat_smile: I did find some similar instructions, also in the wiki, but I guess those are also out of date.

christiaanb, 2019-11-21:
@martijnbastiaan I think our readme should also point/copy the "normal" snap-based installation instructions; in case people find Clash through github as opposed to the normal website.

alex-mckenna, 2019-11-26:
README changed in https://github.com/clash-lang/clash-compiler/pull/940

closed 2019-11-26 by alex-mckenna