clash-lang/clash-compiler #936: Possible Bug in delayed

paddytheplaster, 2019-11-22:
Hi again,

I suspect there is a bug in the implementation of `delayed`.

      trues :: DSignal System 0 Bool
      trues = dfromList $ Prelude.repeat True

      Paddy*> sampleN 4 $ toSignal $ delayed Nil trues
      [True,True,True,True] -- correct
      Paddy*> sampleN 4 $ toSignal $ delayed (False:>Nil) trues
      [False,False,True,True] -- wrong
      -- should have been [False,True,True,True] ??

All other calls with a positive length for the first argument of `delayed` have a delay which appears to be one clock cycle too long.

FWIW I find the name @delayed@ a bit poor because it's not used as a property and it's neither a verb nor a noun. Clearly `delay` would have been better but that's already taken. If it's still possible to rename the function, I'd suggest `defer`, `postpone`, or equivalent. Perhaps `extendDelayWith`, `addToDelay`, or `delayWith` are even better as they also take into account the values of the additional output signals.

Regards,

Paddy

martijnbastiaan, 2019-11-22:
What you're seeing is 

1. The power up value
2. The reset value

The tutorial describes it as:

> Now, let us get back to the functionality of the register function: it is a simple latch that only changes state at the tick of the global clock, and it has an initial value a which is its output at time 0. We can further examine the register function by taking a look at the first 4 samples of the register functions applied to a constant signal with the value 8:
>
```
>>> sampleN @System 4 (register 0 (pure (8 :: Signed 8)))
[0,0,8,8]
```

> Where we see that the initial value of the signal is the specified 0 value, followed by 8's. You might be surprised to see two zeros instead of just a single zero. What happens is that in Clash you get to see the output of the circuit before the clock becomes actives. In other words, in Clash you get to describe the powerup values of registers too. Whether this is a defined or unknown value depends on your hardware target, and can be configured by using a different synthesis Domain. The default synthesis domain, @System, assumes that registers do have a powerup value - as is true for most FPGA platforms in most contexts.

martijnbastiaan, 2019-11-22:
Note that `sampleN` asserts the reset for a single cycle.

paddytheplaster, 2019-11-22:
Thanks Martijn. I did read this a while ago but I forgot...

Regards,

Paddy

martijnbastiaan, 2019-11-22:
No problem, happy hacking!

closed 2019-11-22 by martijnbastiaan