clash-lang/clash-compiler #945: Rewrite rules for Vec Applicative Functor

christiaanb, 2019-11-27:
As reported in https://groups.google.com/d/msgid/clash-language/fa0cce6f-662f-4423-acf7-61d15c46cafa%40googlegroups.com?utm_medium=email&utm_source=footer

```haskell
f <$> as <*> bs
```
results in slow synthesis (both Clash and Quartus), compared to:
```haskell
zipWith f as bs
```
That's because the former translates to:
```
zipWith ($) (map f as) bs
```
where we get intermediate `Vec` of type `(b -> c)`, which Clash is forced to unroll.

Perhaps we can use GHC rewrite rules to transform:
```haskell
zipWith ($) (map f as) bs  ==> zipWith f as bs
```
The strictness of both expressions is the same, so no issues there.

basile-henry, 2019-11-29:
Will this fix the `sequenceA` vs `bundle` performance differences for `Vec`?

christiaanb, 2019-11-30:
@basile-henry you mean wrt transposing a `Vec . Signal` to `Signal . Vec`? Maybe if we’re allowed to match on types in rules then we could add a rule:

```
traverse# @a @(Signal dom) @_ @n id ==> vecBundle# @n @dom @a
```

commit 2cc8b826 (2020-01-09): Add rewrite rules to speed up synthesis of some Vec operations

Fixes #945
PR #993 (2020-01-13): Add rewrite rules to speed up synthesis of some Vec operations
Fixes #945
commit ce31044d (2020-01-13): Add rewrite rules to speed up synthesis of some Vec operations

Fixes #945
closed 2020-01-13 by christiaanb