clash-lang/clash-compiler #950: Partial evaluation

christiaanb, 2019-12-02:
Clash's existing compilation method of exhaustive rewriting, while significantly improved over the years, turns out to be a large performance bottleneck; especially when forced to unroll loops. Specifically, it's the successive traversal of the expression ADT to basically achieve compile-time evaluation in a very round-about way that is hurting us the most.

The idea is to repurpose Clash's existing WHNF-evaluator to an NF evaluator as a first pass of the compiler. Given that it was specifically designed for compile-time evaluation, it is much faster at this job than the exhaustive rewrite system. An early prototype of this approach did, however, show a lot of circuit duplication. So the final implementation should:
1. Make sure that the new NF-evaluator has techniques to stop duplication
2. Improve the existing Common-Subexpression-Elimination pass to recover more sharing.
3. Not only prevent duplication in the circuit, but also to prevent the NF-evaluator from re-evaluating already evaluated expression (e.g. using some sort of tag)
4. Attempt a parallel implementation, e.g. evaluate case-branches in parallel. This needs investigation with regard to sharing the evaluation of common subexpressions, and sharing/merging evaluator data-structures.

# Various thoughts
* [`Value`](https://github.com/clash-lang/clash-compiler/blob/9624b40f8c79e9b7bbec5f10973c33369c4c93f7/clash-lib/src/Clash/Core/Evaluator.hs#L100) should contain `Tick`ed and `Cast`ed values, this way the evaluator can preserve both. Ticks must be maintained because they contain register and instance names. Adding these values means we have to update the primitive evaluation rules (https://github.com/clash-lang/clash-compiler/blob/master/clash-ghc/src-ghc/Clash/GHC/Evaluator.hs), as they need to be able to look through ticks (ala `coreView`). Additionally, some casts may have to be translated to `Prim`s, such as casts from `Integer` to `Unsigned`

alex-mckenna, 2020-01-23:
Update: Another paper, [_Taming Code Explosion in Supercompilation_](https://www.researchgate.net/publication/220989852_Taming_code_explosion_in_supercompilation), seems to offer a reasonable approach to reducing the size of code after supercompilation. As their basic approach seems similar to that of Bolingbroke and Peyton Jones, this could be worthwhile to look at.

alex-mckenna, 2020-06-04:
Update: the latest PR for this issue (#1288) provides the common semantics for the evaluator, going all the way to beta-normal eta-long form (NF). _However_, it should be noted that this is not the full partial evaluator as intended, but simply a good place to stop and get feedback. As I see it, there are three problems to be solved before this is really a partial evaluator:

  * **Evaluating primitives**
    The new evaluator does not yet have any rules for evaluating primitives. A lot of this work was started in the [partial-evaluator-2](/clash-lang/clash-compiler/tree/partial-evaluator-2) branch and should be tidied up and moved over. In particular, we want to keep the sensible coercions between `Value` and types in Haskell.

  * **Unrolling recursion**
    We want to be able to unroll and evaluate recursive definitions in the partial evaluator. However, this is not possible in every case (Haskell is not a total language and some terminating Haskell programs may require non-strict evaluation to converge on a value). We could implement something more naive here, but I think a better way would be to write a termination analyser to be run before normalisation (as we want to be able to partially evaluate as far as possible). The _foetus_ termination checker of Andreas Abel looks to be the best bit for this.

  * **Removing transformations**
    As mentioned above, several transformation passes become redundant when the partial evaluator is implemented as intended. Removing these and folding their behaviour into the semantics of the evaluator is a fairly large PR in itself, and one which would benefit from being reviewed separately.