clash-lang/clash-compiler #952: Add name prefixes when functions are inlined

christiaanb, 2019-12-03:
We need to add name prefixes when functions are inlined. I.e. when we have

```
{-# ANN f Synthesize … #-}
f a b = ... g ... 

g p q = ... setName @”foo” h ... 

{-# ANN h Synthesize … #-}
h x y = ...
```
The instance of ‘h’ inside the ‘f’ module is currently called ‘foo’, note that there will be no ‘g’ module as it’s inlined because it doesn’t have a synthesize or noinline pragma. What we still need to do is that when ‘g’ is inlined, all the instance labels are prefixed with “g_”, so that the instance of ‘h’ in ‘f’ is called “g_foo”.

# implementation suggestion
* [`Tick`](https://github.com/clash-lang/clash-compiler/blob/db19778485bd3a1b72ffbaa25797255b898d3d54/clash-lib/src/Clash/Core/Term.hs#L65) the inlined expression with a `NameMod PrefixName (LitTy (SymTy <name_of_inlined_function>))`, make sure to remove all module name qualifiers from the name though.
* Locations where we're currently inlining: 
  * https://github.com/clash-lang/clash-compiler/blob/db19778485bd3a1b72ffbaa25797255b898d3d54/clash-lib/src/Clash/Normalize.hs#L318-L340
  * https://github.com/clash-lang/clash-compiler/blob/db19778485bd3a1b72ffbaa25797255b898d3d54/clash-lib/src/Clash/Normalize.hs#L347-L354
  * https://github.com/clash-lang/clash-compiler/blob/db19778485bd3a1b72ffbaa25797255b898d3d54/clash-lib/src/Clash/Normalize/Transformations.hs#L368
  * https://github.com/clash-lang/clash-compiler/blob/db19778485bd3a1b72ffbaa25797255b898d3d54/clash-lib/src/Clash/Normalize/Transformations.hs#L949
  * https://github.com/clash-lang/clash-compiler/blob/db19778485bd3a1b72ffbaa25797255b898d3d54/clash-lib/src/Clash/Normalize/Transformations.hs#L1011
  * https://github.com/clash-lang/clash-compiler/blob/db19778485bd3a1b72ffbaa25797255b898d3d54/clash-lib/src/Clash/Core/Evaluator.hs#L357-L360

alex-mckenna, 2019-12-05:
Accidentally closed this issue instead of the related one. Whoops.

alex-mckenna, 2019-12-05:
@christiaanb Where is the inlining being performed in `Clash.Core.Evaluator`? I can't quite make sense of what's being inlined in the snippet you linked.

christiaanb, 2019-12-05:
https://github.com/clash-lang/clash-compiler/blob/db19778485bd3a1b72ffbaa25797255b898d3d54/clash-lib/src/Clash/Core/Evaluator.hs#L357-L360

That `e` is the body of the global function `x'`; so we're basically inlining `x'`.

alex-mckenna, 2019-12-06:
So interestingly enough, the changes are in and the example in the issue now works as expected. However, the changes to the inlining functions in the `Transformations` module results in some tests failing for seemingly unrelated reasons:

  * Normalisation resulting in a recursive let binding (e.g. `Feedback/Fib.hs`)
  * Generated HDL has name clashes sometimes (e.g. `LambdaDrop`)

christiaanb, 2019-12-06:
`Feedback/Fib.hs` depends on `recToLetrec` firing correctly; but that transformation is fragile.

alex-mckenna, 2019-12-06:
That's good, gives me a place to start looking. With `LambdaDrop` it's generating a name clash between a port and a signal in the core module (prior to adding the `NamePrefix` tick it correctly changed the signal name from `x` to `x_0`).

@leonschoorl and @martijnbastiaan pointed me towards #476 which seems closely related

alex-mckenna, 2019-12-06:
Another failing test I didn't spot before: `CustomReprs.Indexed`

```
topentity.vhdl:16:11: character '$' can only be used in strings or comments
topentity.vhdl:16:12: ':' is expected instead of "case_alt"
topentity.vhdl:16:20: missing ";" at end of object declaration
topentity.vhdl:33:6: character '$' can only be used in strings or comments
topentity.vhdl:33:7: '<=' is expected instead of "case_alt"
topentity.vhdl:41:22: character '$' can only be used in strings or comments
topentity.vhdl:41:22: ';' expected at end of signal assignment
topentity.vhdl:41:22: (found: an identifier)
topentity.vhdl:41:32: '<=' is expected instead of 'when'
topentity.vhdl:41:32: unexpected token 'when' in a primary
```

I would suspect it's generating names with `$` in it which aren't being escaped (although I can't seem to get it to run with `cabal exec clash -- ...`)

christiaanb, 2019-12-06:
That's related to the `topLet` firing too early as well.

christiaanb, 2019-12-19:
Implemented in: https://github.com/clash-lang/clash-compiler/pull/958

commit 9a576431 (2019-12-05): Merge pull request #957 from clash-lang/setname-inlining

Respect setName for functions with Synthesize. Closes #952