clash-lang/clash-compiler #956: Clash not Respecting `setName`

alex-mckenna, 2019-12-05:
Given the following trivial program, when `clash` is called, the generated HDL refers to the inlined `g` in `f` as `h` instead of as `foo`. If `g` is annotated as `NOINLINE` then it refers to it as `g` in `f` instead of renaming to `foo` as expected.

```haskell
module Example where

import Clash.Prelude

f :: Int -> Int -> Int
f a b = g a b * g b a
{-# ANN f (Synthesize
    { t_name   = "f"
    , t_inputs = [PortName "f_x", PortName "f_y"]
    , t_output = PortName "result"
    }) #-}

g :: Int -> Int -> Int
g = setName @"foo" h

h :: Int -> Int -> Int
h x y = if signum (x - y) == 1 then x else y
{-# ANN h (Synthesize
    { t_name   = "h"
    , t_inputs = [PortName "x", PortName "y"]
    , t_output = PortName "diff"
    }) #-} 
```

The description of issue #952 makes me think this is the wrong behaviour, specifically where it says:

> The instance of ‘h’ inside the ‘f’ module is currently called ‘foo’, note that there will be no ‘g’ module as it’s inlined because it doesn’t have a synthesize or noinline pragma.

christiaanb, 2019-12-05:
@alex-mckenna Right, it seems to ignore it when `h` is given a `Synthesize` annotation, when merely giving it a `NOINLINE` pragma we do see the `f_h` component being named `foo` and `foo_0`.

The Synthesize case is handled over here: https://github.com/clash-lang/clash-compiler/blob/c860a3924646964ebdcc32d078c803ddaa7c3560/clash-lib/src/Clash/Netlist.hs#L533-L565

while the "normal" case is handled over here: https://github.com/clash-lang/clash-compiler/blob/c860a3924646964ebdcc32d078c803ddaa7c3560/clash-lib/src/Clash/Netlist.hs#L572-L599 
specifically, the instance label name is handled here: https://github.com/clash-lang/clash-compiler/blob/c860a3924646964ebdcc32d078c803ddaa7c3560/clash-lib/src/Clash/Netlist.hs#L593-L596

To fix, in https://github.com/clash-lang/clash-compiler/blob/c860a3924646964ebdcc32d078c803ddaa7c3560/clash-lib/src/Clash/Netlist/Util.hs#L1337

we need to add similar setName logic over here:

https://github.com/clash-lang/clash-compiler/blob/c860a3924646964ebdcc32d078c803ddaa7c3560/clash-lib/src/Clash/Netlist/Util.hs#L1370-L1371

alex-mckenna, 2019-12-05:
Fixed in #957

closed 2019-12-05 by alex-mckenna