clash-lang/clash-compiler #959: Index: Bug in even and odd

paddytheplaster, 2019-12-09:
Hi again,

I'd like to report some bugs in 'even' and 'odd'. (Clash, version 1.0.1 (using clash-lib, version: 1.0.1))

    -- even :: Integral a => a -> Bool

    bug1 = even (0 :: Index 1)
    bug2 = even (0 :: Index 2)
    bug3 = even (1 :: Index 2)

Now bug1 gives

*** Exception: X: Clash.Sized.Index: result 2 is out of bounds: [0..0]
CallStack (from HasCallStack):
  errorX, called at src/Clash/Sized/Internal/Index.hs:255:13 in clash-prelude-1.0.1-BorHjli5pMX3STUvZInnYc:Clash.Sized.Internal.Index
  fromInteger_INLINE, called at src/Clash/Sized/Internal/Index.hs:249:16 in clash-prelude-1.0.1-BorHjli5pMX3STUvZInnYc:Clash.Sized.Internal.Index

You get similar errors for 'odd'.

Here's a possible fix:

    odd', even' :: (KnownNat n) => Index (n+1) -> Bool                         
    odd' = bitToBool . lsb . pack                                     
    even' = not . odd'

  * These errors are __extremely__ difficult to find because the error messages thrown do not refer to sections in the user's source file. If the error is caused by a statement/function call in a deeply nested where clause, you can only find the error by making the deeply nested statements top-level functions and play with these functions until you've found the cause. It can take very long before you know which statement is the cause...
  * I get similar errors like these a lot. They're usually caused by errors on my behalf. Finding them can take very long.
  * Is there any way you can introduce a flag which forces error messages to pin-point the cause of the error to a location in the user's source file?

TIA for your time.

Regards,

Paddy

martijnbastiaan, 2019-12-09:
> Is there any way you can introduce a flag which forces error messages to pin-point the cause of the error to a location in the user's source file?

Yes, using profiling! For example, I've got:

```haskell
module Test where

import Clash.Prelude

triggered :: Bool
triggered = even (0 :: Index 1)

aboveTriggered :: Bool -> Bool
aboveTriggered b = b || triggered

main :: IO ()
main = putStrLn $ show $ aboveTriggered False
```

Using `clash Test.hs -main-is Test.main -o main && ./main` I get:

```
Loaded package environment from /home/martijn/code/clash-compiler/.ghc.environment.x86_64-linux-8.8.1
[1 of 1] Compiling Test             ( Test.hs, Test.o )
Linking main ...
main: X: Clash.Sized.Index: result 2 is out of bounds: [0..0]
CallStack (from HasCallStack):
  errorX, called at src/Clash/Sized/Internal/Index.hs:255:13 in clash-prelude-1.1.0-189457ad4525d4af545fff79572722beba436b21b84e33704551ecd778262bb1:Clash.Sized.Internal.Index
  fromInteger_INLINE, called at src/Clash/Sized/Internal/Index.hs:249:16 in clash-prelude-1.1.0-189457ad4525d4af545fff79572722beba436b21b84e33704551ecd778262bb1:Clash.Sized.Internal.Index
```

However, if I run it with profiling enabled, `clash Test.hs -prof -fprof-auto -main-is Test.main -o main && ./main +RTS -xc`, I get:

```
*** Exception (reporting due to +RTS -xc): (THUNK_2_0), stack trace: 
  Clash.XException.errorX,
  called from Clash.Sized.Internal.Index.fromInteger#,
  called from Test.triggered,
  called from Test.CAF
  --> evaluated by: Clash.Sized.Internal.Index.rem#,
  called from Test.triggered,
  called from Test.CAF
  --> evaluated by: Clash.Sized.Internal.Index.eq#,
  called from Test.triggered,
  called from Test.CAF
  --> evaluated by: Test.aboveTriggered,
  called from Test.main,
  called from Test.CAF
  --> evaluated by: Test.main,
  called from Test.CAF
  --> evaluated by: Test.main
main: X: Clash.Sized.Index: result 2 is out of bounds: [0..0]
CallStack (from HasCallStack):
  errorX, called at src/Clash/Sized/Internal/Index.hs:255:13 in clash-prelude-1.1.0-inplace:Clash.Sized.Internal.Index
  fromInteger_INLINE, called at src/Clash/Sized/Internal/Index.hs:249:16 in clash-prelude-1.1.0-inplace:Clash.Sized.Internal.Index

```

Note that your code will compile and run much slower with these flags enabled. If you use cabal to build/run clash, make sure to include `--enable-profiling`.

For technical reasons, we can't make `HasCallStack` point to user code.. :(

paddytheplaster, 2019-12-09:
Hi Martijn,

That's great. Had never thought of that.

It will save me a lot of time...

Regards,

Paddy

leonschoorl, 2019-12-10:
That profiling example is great, I think we should add it to the FAQ or something.

The bug itself caused the the assumption that you can always create the value `2` in [GHC.Real].
And this leads to even more weird behavior:
```
Clash.Prelude> even (0 :: Unsigned 1)
*** Exception: divide by zero
Clash.Prelude> even (0 :: Signed 1)
*** Exception: divide by zero
Clash.Prelude> even (0 :: Bit)
True
Clash.Prelude> even (1 :: Bit)
True
```

Your proposed solution would work for this, and it could even be simplified and generalized to:
```haskell
odd', even' :: BitPack a => a -> Bool                         
odd' = bitToBool . lsb
even' = not . odd'
```
We could export these from `Clash.Prelude` instead of the ones from base.
But they have their own downsides, because of the changed type:
- you can no longer call odd on Integers because their is no BitPack instance for Integer
- you can call odd on anything that has a BitPack instance even if doesn't really make sense, for example `Just ()` or `pi :: Float`
- because of undefined bits this can generate even weirder errors:
```
Clash.Prelude> odd' (Nothing :: Maybe Int)
*** Exception: X: Bool.unpack called with (partially) undefined arguments: .
CallStack (from HasCallStack):
  undefError, called at src/Clash/Sized/Internal/BitVector.hs:970:11 in clash-prelude-1.1.0-inplace:Clash.Sized.Internal.BitVector
```

[GHC.Real]: https://hackage.haskell.org/package/base-4.12.0.0/docs/src/GHC.Real.html#even

paddytheplaster, 2019-12-10:
i Leon,

I wouldn't change the signature of 'odd' and 'even' as you proposed. E.g. many algorithms require 'even' and 'odd' for 'Integer's.

I don't know how class extension of 'Integral' is implemented for 'Index'. IMHO it would be better to implement the change at this level.

Regards,

Paddy

leonschoorl, 2019-12-10:
I agreed that changing the type of odd and even is not a good idea.

The class instance is right here:
https://github.com/clash-lang/clash-compiler/blob/9a576431d6cf4d73312104e5438aaf86fbd95b1a/clash-prelude/src/Clash/Sized/Internal/Index.hs#L336-L353

But because odd/even aren't class methods we can't customize them for a specific type like Index.
The only thing I can think of right now to "solve" this odd/even issue is to add a `3 <= n` constraint on this instance.
But that feels totally arbitrary and weird.

rowanG077, 2019-12-10:
It's also possible to upconvert the index to `index (Max n 3)`. But that has some cost in hardware.

paddytheplaster, 2019-12-11:
Thanks Rowan.

Nice but it's not an ideal solution. Having 'n' equal to '1' or '2' should be allowed having 'n <= 3' would break a lot of code.

Regards,

Paddy

paddytheplaster, 2019-12-11:
Hi Leon,

Thanks. I incorrectly assumed 'even' and 'odd' were class functions but they're not. I don't think that adding the consraint '3 <= n' is a solution (as suggested before).

I see the following possibilities:

    * Escalate this as a ghc/Haskell bug because the definition of 'even' and 'odd' seems to assume that any type which extends 'Integral': should include '2'.
    * Not sure if this is a possible (you seem to suggest it's not). Clash already re-defines lots of functions comming from the 'Prelude'. Maybe you can change it at this level?

Regards,

Paddy

leonschoorl, 2019-12-11:
Re-defining odd and even in `Clash.Prelude` is possible.
But I'm not sure we can come up with a sensible implementation that works in all cases.

rowanG077, 2019-12-11:
Why not @leonschoorl? Only `Index {0, 1, 2}` and possible the zero width types of the other Clash numerical values. For index we could pattern match on the size and choose a different implementation when it's one of the 3 "weird" sizes. In the unit test we can ensure this is stripped out of the generated HDL.

rowanG077, 2019-12-12:
@leonschoorl Maybe I'm missing something but couldn't this be done:

```
{-# LANGUAGE BangPatterns          #-}

import Clash.Prelude hiding (odd, even)

class Parity a where
  even :: a -> Bool
  odd :: a -> Bool
  odd = not . odd

instance (KnownNat n) => Parity (Index n) where
  even !a =
    case snatToNum @Int (SNat @n) of
      1 -> True
      2 -> a == 0
      _ -> a `rem` 2 == 0

topEntity :: Signal System (Bool, Bool)
topEntity = pure $ (even (0 :: Index 1), even (1 :: Index 2))
```

After compilation the case statement is gone so it doesn't have an extra cost. Here is the resulting VHDL:

```
-- Automatically generated VHDL-93
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.NUMERIC_STD.ALL;
use IEEE.MATH_REAL.ALL;
use std.textio.all;
use work.all;
use work.main_types.all;

entity topentity is
  port(result_0 : out boolean;
       result_1 : out boolean);
end;

architecture structural of topentity is
  signal result : main_types.tup2;

begin
  result <= ( tup2_sel0_boolean_0 => true
            , tup2_sel1_boolean_1 => false );

  result_0 <= result.tup2_sel0_boolean_0;

  result_1 <= result.tup2_sel1_boolean_1;

end;
```

martijnbastiaan, 2019-12-12:
Creating a typeclass `Parity` seems like the way to go. I think we shouldn't go with overlapping typeclasses, as it would trigger the same error again in some corner cases. But as @rowanG077 said, it should be easy to reimplement. For example:

```haskell
import qualified Prelude as P

class Parity a where
  even :: a -> Bool
  default even :: Integral a => a -> Bool
  even = P.even

  odd :: a -> Bool
  default odd :: Integral a => a -> Bool
  odd = P.odd

instance Parity Int
instance Parity Integer
[..]
```

martijnbastiaan, 2019-12-12:
Maybe even:

```haskell
import qualified Prelude as P

class Parity a where
  even :: a -> Bool
  default even :: BitPack a => a -> Bool
  even = (==low) . lsb . pack

  odd :: a -> Bool
  default odd :: BitPack a => a -> Bool
  odd = (==high) . lsb . pack

instance Parity Int
instance Parity Integer
[..]
```

for easier-to-parse HDL?

paddytheplaster, 2019-12-12:
Hi Rowan,

Thanks for that. I was also thinking about an intermediate class but I couldn't find a right name. 'Parity' is perfect.

I am not sure that using 'sNatToInt' is the right test for identifying the small 'n' cases as this will cause rounding errors for larg 'n'. I'd use something along the following lines:

    even' :: forall (n :: Nat) . KnownNat n => Index n -> Bool
    even' a =
      case m of
        0 -> True
        1 -> a /= m
        _ -> a `rem` 2 == 0
      where m = maxBound :: Index n

Regards,

Paddy

paddytheplaster, 2019-12-12:
This is a related issue.

In Clash it is perfectly valid to write:

    f :: (KnownNat n) => Index n -> Bool
    f = (==) <INTEGER LITERAL>

But this function will cause a run-time error when 'n' is less than or equal to the '<INTEGER LITERAL>'. This is a source of many possibe errors, so IMO this should throw a compile-time error.

Any opinions?

Regards,

Paddy

martijnbastiaan, 2019-12-12:
@paddytheplaster You could use:

```haskell
-- | Initialize an "Index" value. As opposed to using literals, this version
-- will yield a compile-time error if a value outside of the Index's range is
-- instantiated. 
-- 
-- Example:
-- 
-- >>> iFromSNat (SNat @15) :: Index 16   
-- 15   
-- >>> iFromSNat (SNat @16) :: Index 16
-- <BLANKLINE>
-- <interactive>:21:1: error:
--    • Couldn't match type ‘'False’ with ‘'True’
--        arising from a use of ‘iFromSNat’
--    • In the expression: iFromSNat (SNat @15) :: Index 15
--      In an equation for ‘it’: it = iFromSNat (SNat @15) :: Index 15
iFromSNat 
  :: forall n m
   . ( KnownNat n
     , (m + 1) <= n   
     ) 
  => SNat m 
  -> Index n
iFromSNat = snatToNum
```

The error message is not as helpful as it could be :sweat: , but at least it fails at compile time!

leonschoorl, 2019-12-12:
@rowanG077 Yes, with a separate typeclass we can fix this.
I totally missed that because I was so focussed not changing the current type of `odd` and `even`.

leonschoorl, 2019-12-12:
And for the "I wanna write a Index constant and have it checked at compile time"-problem you could also do:

```haskell
liftIndex :: KnownNat n => Index n -> Q (TExp (Index n))
liftIndex i = case hasX i of
  Left err -> fail $ "liftIndex failed: " <> err
  Right _ -> TExp <$> lift i

test :: Index 3
test = $$(liftIndex 10)
```

```
testlit.hs:5:11: error:
    • liftIndex failed: X: Clash.Sized.Index: result 10 is out of bounds: [0..2]
CallStack (from HasCallStack):
  errorX, called at src/Clash/Sized/Internal/Index.hs:255:13 in clash-prelude-1.1.0-inplace:Clash.Sized.Internal.Index
  fromInteger_INLINE, called at src/Clash/Sized/Internal/Index.hs:249:16 in clash-prelude-1.1.0-inplace:Clash.Sized.Internal.Index
    • In the Template Haskell splice $$(liftIndex 10)
      In the expression: $$(liftIndex 10)
      In an equation for ‘test’: test = $$(liftIndex 10)
```

rowanG077, 2019-12-12:
@martijnbastiaan I didn't know about default signatures that's cool. But it has some problems for the index case unfortunately. The instance of `Integral` for `Index` doesn't have a constraint for `1 <= n` whereas the `BitPack` instance for `Index` has that constraint. This makes it not possible to create an instance for `Index` that complies to the current API. With a little code duplication we can work around that.

```
import Clash.Prelude hiding (odd, even)

evenParity :: forall n . (KnownNat n) => BitVector n -> Bool
evenParity a =
  case snatToNum @Integer (SNat @n) of
    0 -> True
    _ -> (==low) $ lsb a

oddParity :: forall n . (KnownNat n) => BitVector n -> Bool
oddParity a =
  case snatToNum @Integer (SNat @n) of
    0 -> False
    _ -> (==high) $ lsb a

class Parity a where
  even :: a -> Bool
  even = not . odd
  odd :: a -> Bool
  odd = not . even
  {-# MINIMAL odd | even #-}

instance Parity Integer
  where
    even a = a `rem` 2 == 0

instance Parity Int where
  even = evenParity . pack
  odd = oddParity . pack

instance Parity Word where
  even = evenParity . pack
  odd = oddParity . pack

instance Parity Bit where
  even = evenParity . pack
  odd = oddParity . pack

instance KnownNat n => Parity (BitVector n) where
  even = evenParity . pack
  odd = oddParity . pack

instance KnownNat n => Parity (Unsigned n) where
  even = evenParity . pack
  odd = oddParity . pack

instance KnownNat n => Parity (Signed n) where
  even = evenParity . pack
  odd = oddParity . pack

instance KnownNat n => Parity (Index n) where
  even a =
    case snatToNum @Integer (SNat @n) of
      1 -> True
      2 -> a == 0
      _ -> a `rem` 2 == 0

  odd a =
    case snatToNum @Integer (SNat @n) of
      1 -> False
      2 -> a == 1
      _ -> a `rem` 2 == 1
```

I still put the case statement to check for a `BitSize` of 1 since it will create nicer HDL.

@paddytheplaster  I guess you mean that when `n` because too large to fit in an `Int`? You are right that this could feasibly occur with `Index`. So I changed it to `Integer`.

martijnbastiaan, 2019-12-13:
@rowanG077 I don't think it's a bad idea to add the `1 <= n` constraint for `odd`/`even` on `Index`. There's no sensible answer for `n ~ 0` anyway. So I'd imagine the instance for `Index` would look something like (overriding the default method!):

```haskell
instance (KnownNat n, 1 <= n) => Parity (Index n) where
  even n = case SNat @n of { 1 -> False; _ -> ... }
  odd n = case SNat @n of { 1 -> True; _ -> ... }
```

Strictly seen it breaks the API, but IMO this is one of the cases where the API itself contains a bug..

paddytheplaster, 2019-12-13:
> @paddytheplaster You could use:
>
> ```haskell
> -- | Initialize an "Index" value. As opposed to using literals, this version
> -- will yield a compile-time error if a value outside of the Index's range is [...]

Hi Martijn,

Thanks for the suggestion.

Regards,

Paddy

paddytheplaster, 2019-12-13:
> And for the "I wanna write a Index constant and have it checked at compile time"-problem you could also do:
> 
> ```haskell
> module LiftIndex where
> import Clash.Prelude
> import Language.Haskell.TH.Syntax
> liftIndex :: KnownNat n => Index n -> Q (TExp (Index n))
> liftIndex i = case hasX i of
>   Left err -> fail $ "liftIndex failed: " <> err
>   Right _ -> TExp <$> lift i
> 
> -- put this in a different file:
> test :: Index 3
> test = $$(liftIndex 10)
> ```
> 
> ```
> testlit.hs:5:11: error:
>     • liftIndex failed: X: Clash.Sized.Index: result 10 is out of bounds: [0..2]
> CallStack (from HasCallStack):
>   errorX, called at src/Clash/Sized/Internal/Index.hs:255:13 in clash-prelude-1.1.0-inplace:Clash.Sized.Internal.Index
>   fromInteger_INLINE, called at src/Clash/Sized/Internal/Index.hs:249:16 in clash-prelude-1.1.0-inplace:Clash.Sized.Internal.Index
>     • In the Template Haskell splice $$(liftIndex 10)
>       In the expression: $$(liftIndex 10)
>       In an equation for ‘test’: test = $$(liftIndex 10)
> ```

Thanks for the suggestion Rowan. Requires a lot of work, though. 

Regards,

Paddy

rowanG077, 2019-12-15:
@martijnbastiaan The type error in your `iFromSnat` function can be improved considerably by using custom type errors.

```
{-# LANGUAGE UndecidableInstances #-}

import Clash.Prelude

type family ReportTooLarge (n :: Nat) (x :: Nat) tooLarge :: Bool where
  ReportTooLarge n x 'True =
    TypeError ( 'Text "Value " :<>: ShowType n :<>:
                'Text " is too large to fit in Index of size " :<>: ShowType x
              )
  ReportTooLarge n x 'False = 'True

type family FitsInIndex (n :: Nat) (x :: Nat) :: Bool where
  FitsInIndex n x = ReportTooLarge n x (n <=? x)

iFromSNat
  :: forall n m
   . ( KnownNat n
     , FitsInIndex n m ~ 'True
     )
  => SNat m
  -> Index n
iFromSNat = snatToNum
```

martijnbastiaan, 2019-12-15:
Ooh very nice!

commit 7092a9d7 (2019-12-14): Fixes #959, odd and even functions have been replaced by clash specific variants by
a Parity type class.
commit 866cea0c (2019-12-14): Fixes #959, odd and even functions have been replaced by clash specific variants by
a Parity type class.
commit 214edfa9 (2019-12-15): Fixes #959, odd and even functions have been replaced by clash specific variants by
a Parity type class.
commit 2ab8d6a0 (2019-12-15): Fixes #959, odd and even functions have been replaced by clash specific variants by
a Parity type class.
closed 2019-12-16 by martijnbastiaan