clash-lang/clash-compiler #967: Naming registers disconnects their output

leonschoorl, 2019-12-13:
```haskell
topEntity x = setName @"myRegister" $ register @System (0::Int) x
```

```vhdl
entity topentity is
  port(-- clock
       \$dIP\  : in main_types.clk_system;
       \$dIP1\ : in boolean;
       -- reset
       \$dIP2\ : in main_types.rst_system;
       eta     : in signed(63 downto 0);
       result  : out signed(63 downto 0));
end;

architecture structural of topentity is
begin
  -- register begin 
  myregister_register : process(\$dIP\,\$dIP2\)
  begin
    if \$dIP2\ =  '1'  then
      myRegister <= to_signed(0,64);
    elsif rising_edge(\$dIP\) then
      if \$dIP1\ then
        myRegister <= eta;
      end if;
    end if;
  end process;
  -- register end
end;
```
The register blackbox uses the name `myRegister`, but there's no signal declared with this name.
Nor is it connected to the `result` out port.

And `delay` has the same issue.

christiaanb, 2019-12-14:
Right... the special blackbox result Name handling code, used by register and delay, doesn’t seem to deal with the context where the result is the output of the entity.

leonschoorl, 2019-12-16:
Also the powerup value isn't rendered in HDL.
But that probably has the same cause.

christiaanb, 2019-12-16:
https://github.com/clash-lang/clash-compiler/pull/781 is the PR added the "primitive-can-control-result-naming" code

commit c6b43b93 (2020-01-09): Ensure the result gets a new binder if it's renamed

Fixes #967
commit 9c7bbe98 (2020-01-13): Ensure the result gets a new binder if it's renamed

Fixes #967
PR #989 (2020-01-13): Ensure the result gets a new binder if it's renamed
Fixes #967
commit 726e9876 (2020-01-13): Ensure the result gets a new binder if it's renamed

Fixes #967
closed 2020-01-13 by christiaanb