clash-lang/clash-compiler #978: Remove the need for `KnownDomain` constraints throughout Clash designs

martijnbastiaan, 2019-12-17:
Clash primitives currently require `KnownDomain` constraints in order to change their behavior depending on their [synthesis domain](https://clash-lang.org/blog/0005-synthesis-domain/). It's implemented using class constraints as this - apart from all the dormant bugs it exposed -  didn't require adding additional logic to the Clash compiler. An example of a primitive requiring a `KnownDomain` constraint is `register#`:

https://github.com/clash-lang/clash-compiler/blob/9d278c2a7749ce0a419791b000d4e502c73a0d5e/clash-prelude/src/Clash/Signal/Internal.hs#L1078-L1082

To remove the need for this constraint we would need to do two things:

1. On the simulation side of things, we'd need to remove all the `KnownDomain` constraints on Clash primitives. We subsequently need to add them to `clockGen`, `resetGen`, and `enableGen`. These primitives would use `knownDomain` to store a `SDomainConfiguration` inside the constructs they're building. For example, `Clock` might look like:

```haskell
data Clock (dom :: Domain) = 
  Clock 
    (SSymbol dom) 
    (SDomainConfiguration dom (KnownConf dom))
```

(If this doesn't work out, we could go for `VDomainConfiguration` too.) Simulation constructs can now simply extract the synthesis domains from their given clocks. Of course, deconstructing clocks, resets, and enables is still not synthesizable! Thus...

2. To get synthesis working we need domain-extracting primitives for clocks, resets, and enables. For example, for a `Clock` this would look like:

```haskell
clockDomain :: Clock dom -> SDomainConfiguration dom (KnownConf dom)
```

(We might want to wrap this in a typeclass so we can use the same function on clocks, resets, and enables all the same.) Clash would need to collect all instances of `KnownDomain` before synthesis and insert the result of `knownDomain` for every `clockDomain` it sees.

paddytheplaster, 2019-12-18:
Hi Martijn,

Brilliant idea.

Regards,

Paddy

martijnbastiaan, 2023-09-01:
Me and @leonschoorl talked about this yesterday. We concluded that the easiest way is probably to add GADT wrappers around `Clock`, `Reset`, `Enable`, and `Signal` that carry the `KnownDomain` dictionary. E.g.:

```haskell
data SClock (dom :: Domain) where
  SClock (Clock dom) :: KnownDomain dom => SClock dom 
```

A few observations:

- This probably doesn't need complicated changes to the compiler. Clash is already fine with wrappings like these.
- We do need to extensions to conjure up `KnownDomain` constraints from thin air. E.g., users should be able to use `SClock System -> ..` in their top entity. (Clash already keeps a list of known domains, so this shouldn't be too hard.)
- We probably also want these constraints in `Signal`, to prevent things like `unsafeToActiveHigh` to get a `KnownDomain` constraint.

rowanG077, 2023-09-01:
Will you be able to do something like `natToNum @(DomainPeriod dom)` without a `KnownDomain dom` constraint with this design? Or will you require to pattern match on `SClock`/`Signal`?

martijnbastiaan, 2023-09-01:
> edit: Ahh sorry I'm confused. KnownDomain doesn't even introduce KnownNat (DomainPeriod dom) anyway

It does since https://github.com/clash-lang/clash-compiler/pull/2541 :-)

You will need to pattern match, yeah.

> `domainPeriod :: forall proxy dom. Num a => proxy dom -> a`

The idea being that you can pass in any Clock/Signal/etc.? Yeah I like it.