clash-lang/clash-compiler #995: UART question

luzarox, 2020-01-09:
Hello

I would like to know if it is possible  to realize  a clash program, which allows to communicate a FPGA card in UART without using a black-box (for inputs and output ) and without using VHDL code?  

And if possible, can you give me an example ?

gergoerdi, 2020-01-10:
On Thu, 9 Jan 2020, luzarox wrote:

> I would like to know if it is possible to realize a clash program, which allows to
> communicate a FPGA card in UART without using a black-box (for inputs and output ) and
> without using VHDL code?
> 
> And if possible, can you give me an example ?

Hi,

In the Clash book I am working on, I am presenting the attached serial 
receiver and transmitter (for Clash 1.0). It is optimized for elegance / 
clarity.

Unfortunately, neither the code nor the book is in a state where I'd want 
to put it online, so I'm just including the current version as-is.

Hope that helps,
 	Gergo

luzarox, 2020-01-11:
Hi gergoerdi ,

Thanks for your feedback, could you please share your code  ?

Regards.

gergoerdi, 2020-01-11:
Hi,

I thought my attachment went through... If not, I'll upload the
current version somewhere and post the link here.

Bye,
Gergo

gergoerdi, 2020-01-12:
I've uploaded my library in its current state to Github; you can find the
UART transmitter and receiver at
https://github.com/gergoerdi/retroclash-lib/blob/master/src/RetroClash/SerialTx.hs
 and
https://github.com/gergoerdi/retroclash-lib/blob/master/src/RetroClash/SerialRx.hs

luzarox, 2020-01-27:
Thank you very much for your help, I will watch it carefully.

closed 2021-06-24 by alex-mckenna