clash-lang/clash-compiler #1005: topEntity clock lambda breaks transformation

DigitalBrains1, 2020-01-14:
The following code breaks compilation:
```Haskell
my_delay :: s -> s -> (s, s)
my_delay reg input = (input, reg)
my_delay_proc = mealy my_delay def

topEntity :: Clock System -> Signal System Bit -> Signal System Bit
topEntity = \c -> exposeClockResetEnable my_delay_proc c resetGen enableGen
```

But this is fine:
```Haskell
topEntity c = exposeClockResetEnable my_delay_proc c resetGen enableGen
```

The broken VHDL is:
```vhdl
-- Automatically generated VHDL-93
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.NUMERIC_STD.ALL;
use IEEE.MATH_REAL.ALL;
use std.textio.all;
use work.all;
use work.main_types.all;

entity topentity is
  port(-- clock
       c      : in main_types.clk_system;
       eta    : in std_logic;
       result : out std_logic);
end;

architecture structural of topentity is
  -- Short.hs:8:1-9
  signal \$d(%,,%)\ : main_types.rst_system;

begin
  -- register begin
  fbundletup22_result_register : process(c,\$d(%,,%)\)
  begin
    if \$d(%,,%)\ =  '1'  then
      $fBundle(,)2_result <= ('0');
    elsif rising_edge(c) then
      $fBundle(,)2_result <= eta;
    end if;
  end process;
  -- register end

  -- resetGen begin
  -- pragma translate_off
  \$d(%,,%)\ <=  '1' ,
            '0'  after (3001 ps + integer'(1 - 1) * 10000 ps);
  -- pragma translate_on
  -- resetGen end

end;
```

leonschoorl, 2020-01-14:
Christiaan just fixed another issue where the output of a register was left unconnected, maybe that will also help here: https://github.com/clash-lang/clash-compiler/issues/967

leonschoorl, 2020-01-14:
Yes, this is fixed in the current master, it will generate an extra line:
```vhdl
  result <= \$fBundle(,)2_result\;
```

christiaanb, 2020-01-14:
Fixed in `master`

closed 2020-01-14 by christiaanb