clash-lang/clash-compiler #1024: Errors Running clash-testsuite with SystemVerilog

alex-mckenna, 2020-01-21:
When running the testsuite on my machine from `master`, I currently get two test failures on _SystemVerilog_ only:

Reducer:
```
Program /opt/intelFPGA_pro/19.4/modelsim_ase/bin/vlog failed with error-code 2.
          
          Stderr was:
          
          
          Stdout was:
          Model Technology ModelSim - Intel FPGA Edition vlog 2019.2 Compiler 2019.04 Apr 17 2019
          Start time: 11:53:32 on Jan 21,2020
          vlog -sv -work work /home/axm/Documents/clash-compiler/.clash-test-tmp/clash-test--8df418675f1bb75d/./examples/Reducer/SystemVerilog/systemverilog/Reducer/Reducer_types.sv /home/axm/Documents/clash-compiler/.clash-test-tmp/clash-test--8df418675f1bb75d/./examples/Reducer/SystemVerilog/systemverilog/Reducer/Reducer_types.sv /home/axm/Documents/clash-compiler/.clash-test-tmp/clash-test--8df418675f1bb75d/./examples/Reducer/SystemVerilog/systemverilog/Reducer/topEntity.sv 
          -- Compiling package Reducer_types
          ** Warning: /home/axm/Documents/clash-compiler/.clash-test-tmp/clash-test--8df418675f1bb75d/./examples/Reducer/SystemVerilog/systemverilog/Reducer/Reducer_types.sv(1): (vlog-2275) 'Reducer_types' already exists and will be overwritten.
          -- Compiling package Reducer_types
          -- Compiling module topEntity
          -- Importing package Reducer_types
          ** Error: /home/axm/Documents/clash-compiler/.clash-test-tmp/clash-test--8df418675f1bb75d/./examples/Reducer/SystemVerilog/systemverilog/Reducer/topEntity.sv(297): (vlog-13172) Selected name after a part-select may only be a modport on an interface port.
          ** Error: /home/axm/Documents/clash-compiler/.clash-test-tmp/clash-test--8df418675f1bb75d/./examples/Reducer/SystemVerilog/systemverilog/Reducer/topEntity.sv(330): (vlog-13172) Selected name after a part-select may only be a modport on an interface port.
          ** Error: /home/axm/Documents/clash-compiler/.clash-test-tmp/clash-test--8df418675f1bb75d/./examples/Reducer/SystemVerilog/systemverilog/Reducer/topEntity.sv(1245): (vlog-13172) Selected name after a part-select may only be a modport on an interface port.
          ** Error: /home/axm/Documents/clash-compiler/.clash-test-tmp/clash-test--8df418675f1bb75d/./examples/Reducer/SystemVerilog/systemverilog/Reducer/topEntity.sv(1264): (vlog-13172) Selected name after a part-select may only be a modport on an interface port.
          ** Error: /home/axm/Documents/clash-compiler/.clash-test-tmp/clash-test--8df418675f1bb75d/./examples/Reducer/SystemVerilog/systemverilog/Reducer/topEntity.sv(1271): (vlog-13172) Selected name after a part-select may only be a modport on an interface port.
          ** Error: /home/axm/Documents/clash-compiler/.clash-test-tmp/clash-test--8df418675f1bb75d/./examples/Reducer/SystemVerilog/systemverilog/Reducer/topEntity.sv(1273): (vlog-13172) Selected name after a part-select may only be a modport on an interface port.
          ** Error: /home/axm/Documents/clash-compiler/.clash-test-tmp/clash-test--8df418675f1bb75d/./examples/Reducer/SystemVerilog/systemverilog/Reducer/topEntity.sv(1277): (vlog-13172) Selected name after a part-select may only be a modport on an interface port.
          ** Error: /home/axm/Documents/clash-compiler/.clash-test-tmp/clash-test--8df418675f1bb75d/./examples/Reducer/SystemVerilog/systemverilog/Reducer/topEntity.sv(1279): (vlog-13172) Selected name after a part-select may only be a modport on an interface port.
          ** Error: /home/axm/Documents/clash-compiler/.clash-test-tmp/clash-test--8df418675f1bb75d/./examples/Reducer/SystemVerilog/systemverilog/Reducer/topEntity.sv(1281): (vlog-13172) Selected name after a part-select may only be a modport on an interface port.
          End time: 11:53:32 on Jan 21,2020, Elapsed time: 0:00:00
          Errors: 9, Warnings: 1
```

TailOfTail:
```
Program /opt/intelFPGA_pro/19.4/modelsim_ase/bin/vlog failed with error-code 2.
              
              Stderr was:
              
              
              Stdout was:
              Model Technology ModelSim - Intel FPGA Edition vlog 2019.2 Compiler 2019.04 Apr 17 2019
              Start time: 11:59:18 on Jan 21,2020
              vlog -sv -work work /home/axm/Documents/clash-compiler/.clash-test-tmp/clash-test--8df418675f1bb75d/./tests/shouldwork/GADTs/TailOfTail/SystemVerilog/systemverilog/TailOfTail/TailOfTail_types.sv /home/axm/Documents/clash-compiler/.clash-test-tmp/clash-test--8df418675f1bb75d/./tests/shouldwork/GADTs/TailOfTail/SystemVerilog/systemverilog/TailOfTail/testBench/testBench_types.sv /home/axm/Documents/clash-compiler/.clash-test-tmp/clash-test--8df418675f1bb75d/./tests/shouldwork/GADTs/TailOfTail/SystemVerilog/systemverilog/TailOfTail/tailOfTail.sv /home/axm/Documents/clash-compiler/.clash-test-tmp/clash-test--8df418675f1bb75d/./tests/shouldwork/GADTs/TailOfTail/SystemVerilog/systemverilog/TailOfTail/TailOfTail_types.sv /home/axm/Documents/clash-compiler/.clash-test-tmp/clash-test--8df418675f1bb75d/./tests/shouldwork/GADTs/TailOfTail/SystemVerilog/systemverilog/TailOfTail/topEntity.sv /home/axm/Documents/clash-compiler/.clash-test-tmp/clash-test--8df418675f1bb75d/./tests/shouldwork/GADTs/TailOfTail/SystemVerilog/systemverilog/TailOfTail/testBench/testBench_types.sv /home/axm/Documents/clash-compiler/.clash-test-tmp/clash-test--8df418675f1bb75d/./tests/shouldwork/GADTs/TailOfTail/SystemVerilog/systemverilog/TailOfTail/testBench/testBench.sv 
              -- Compiling package TailOfTail_types
              -- Compiling package testBench_types
              -- Compiling module tailOfTail
              -- Importing package TailOfTail_types
              ** Warning: /home/axm/Documents/clash-compiler/.clash-test-tmp/clash-test--8df418675f1bb75d/./tests/shouldwork/GADTs/TailOfTail/SystemVerilog/systemverilog/TailOfTail/tailOfTail.sv(7): (vlog-13314) Defaulting port 'ds' kind to 'var' rather than 'wire' due to default compile option setting of -svinputport=relaxed.
              ** Error: (vlog-13069) /home/axm/Documents/clash-compiler/.clash-test-tmp/clash-test--8df418675f1bb75d/./tests/shouldwork/GADTs/TailOfTail/SystemVerilog/systemverilog/TailOfTail/tailOfTail.sv(15): near "[": syntax error, unexpected '[', expecting ';' or ','.
              ** Warning: /home/axm/Documents/clash-compiler/.clash-test-tmp/clash-test--8df418675f1bb75d/./tests/shouldwork/GADTs/TailOfTail/SystemVerilog/systemverilog/TailOfTail/TailOfTail_types.sv(1): (vlog-2275) 'TailOfTail_types' already exists and will be overwritten.
              -- Compiling package TailOfTail_types
              -- Compiling module topEntity
              ** Warning: /home/axm/Documents/clash-compiler/.clash-test-tmp/clash-test--8df418675f1bb75d/./tests/shouldwork/GADTs/TailOfTail/SystemVerilog/systemverilog/TailOfTail/testBench/testBench_types.sv(1): (vlog-2275) 'testBench_types' already exists and will be overwritten.
              -- Compiling package testBench_types
              -- Compiling module testBench
              -- Importing package testBench_types
              End time: 11:59:18 on Jan 21,2020, Elapsed time: 0:00:00
              Errors: 1, Warnings: 3
```

I also find it slightly odd that the command run by the testsuite goes to build the types module twice, although that is slightly less concerning than potentially invalid _SystemVerilog_ being produced.

leonschoorl, 2020-01-21:
Both are cases where clash tries to do too much in one expression: 
`wild[38:0].Tup2_8_sel0`
`ds[1:3][1:2]`

We need to either create a intermediate signal for the result of the first slice and do the next operation on that:
```systemverilog
logic [38:0] tmp;
assign tmp = wild[38:0];
... tmp.Tup2_8_sel0 ...
```

Or SystemVerilog has this magic `item` function:
```systemverilog
... item(wild[38:0]).Tup2_8_sel0 ...
```
Which is a lot shorter, but maybe less readable.

leonschoorl, 2020-02-12:
Similar problems with Nested modifiers can also happen with VHDL: https://github.com/clash-lang/clash-compiler/issues/630

PR #1128 (2020-02-28): Correctly handle nested modifiers in SV
Fixes #1024
commit 9446d745 (2020-02-28): Correctly handle nested modifiers in SV

Fixes #1024
closed 2020-02-28 by martijnbastiaan