clash-lang/clash-compiler #1025: `inlineCleanup` can create empty letrecs

martijnbastiaan, 2020-01-22:
Consider the following pseudo-code (due to GHC optimizations I can't write a small example that triggers this behavior):

```haskell
f a = prim (let b = a in (b && False))
```

Thanks to https://github.com/clash-lang/clash-compiler/pull/984 this will be turned into:

```haskell
f a = prim (let b = a in False)
```

`inlineCleanup` will then remove the binder `b`, because it is unused:

```haskell
f a = prim (let {} in False)
```

Leaving us with an `EmptyRec`. If this is forced in a blackbox, Clash will crash with `Forced to evaluate unexpected function argument`. For this particular case, we can patch `inlineCleanup` to omit a let if it has no bindings:

```diff
diff --git a/clash-lib/src/Clash/Normalize/Transformations.hs b/clash-lib/src/Clash/Normalize/Transformations.hs
index 2d24ab3b..5b6a2e3a 100644
--- a/clash-lib/src/Clash/Normalize/Transformations.hs
+++ b/clash-lib/src/Clash/Normalize/Transformations.hs
@@ -2297,8 +2297,10 @@ inlineCleanup (TransformContext is0 _) (Letrec binds body) = do
       bodyFVs       = Lens.foldMapOf freeLocalIds unitVarSet body
       (il,keep)     = List.partition (isInteresting allOccs prims bodyFVs) binds
       keep'         = inlineBndrs is1 keep il
-  if null il then return  (Letrec binds body)
-             else changed (Letrec keep' body)
+
+  if | null il -> return  (Letrec binds body)
+     | null binds -> changed body
+     | otherwise -> changed (Letrec keep' body)
```

In general, I'm not sure what to do. Should we assign the letrec to a var before passing it to a blackbox?

christiaanb, 2020-01-23:
Nah, https://github.com/clash-lang/clash-compiler/blob/398c4ad8abe135fca94625ff3581e8ac1d12adf2/clash-lib/src/Clash/Netlist/BlackBox.hs#L200 should just have an extra clause for let-expressions where it creates, declarations for the let-bindings, and recursively call itself on the body of the let expression

martijnbastiaan, 2020-01-23:
@christiaanb Is there a specific reason it wasn't accepting letrecs?

christiaanb, 2020-01-23:
@martijnbastiaan legacy reasons, we used to be stricter in our normal forms. You will have to call `mkUniqueNormalized` on the let-expression though, in order to ensure that the String parts of the name (which is used for the HDL name) is unique. See the other uses of `mkUniqueNormalized` in BlackBox.hs

commit d48d2e2a (2020-01-24): Don't produce empty letrecs in inlineCleanup. Fix #1025
commit 533a64e3 (2020-01-24): Don't produce empty letrecs in inlineCleanup. Fix #1025
commit baff941a (2020-01-25): Don't produce empty letrecs in inlineCleanup. Fix #1025
PR #1026 (2020-01-25): Don't produce empty letrecs in inlineCleanup
Fixes #1025.
closed 2020-01-25 by martijnbastiaan