clash-lang/clash-compiler #1028: Top entities with monomorphic constraints are potentially untranslatable

martijnbastiaan, 2020-01-23:
Consider

```haskell
module Test2 where

import Clash.Prelude

subFun
  :: forall dom
   . HiddenClockResetEnable dom
  => Signal dom Int
  -> Signal dom Int
subFun =
  case knownDomain @dom of
    SDomainConfiguration dom period edge reset init_ polarity ->
      case reset of
        SSynchronous -> fmap succ
        SAsynchronous -> fmap pred
{-# NOINLINE subFun #-}

topEntity
  :: ( HiddenClock System
     , HiddenReset System )
  => Signal System Int
  -> Signal System Int
topEntity = withEnable enableGen subFun
```

topEntity's core looks like:

```
λ($d(%,%)6989586621679074455 :: GHC.Classes.(%,%)7710162562058289156
                                  (GHC.Classes.IP3602879701896396848
                                     (GHC.TypeLits.AppendSymbol3674937295934325116
                                        "System"
                                        "_clk")
                                     (Clash.Signal.Internal.Clock8214565720323789554
                                        "System"))
                                  (Clash.Signal.Internal.KnownDomain8214565720323789576
                                     "System")) ->
λ($d(%,%)16989586621679074456 :: GHC.Classes.(%,%)7710162562058289156
                                   (GHC.Classes.IP3602879701896396848
                                      (GHC.TypeLits.AppendSymbol3674937295934325116
                                         "System"
                                         "_rst")
                                      (Clash.Signal.Internal.Reset8214565720323789578
                                         "System"))
                                   (Clash.Signal.Internal.KnownDomain8214565720323789576
                                      "System")) ->
λ(eta4755801206503243777 :: Clash.Signal.Internal.Signal8214565720323789599
                              "System"
                              GHC.Types.Int3674937295934324766) ->
letrec
  $dKnownDomain8286623314361781502 :: Clash.Signal.Internal.KnownDomain8214565720323789576
                                        "System"
  = GHC.Classes.$p2(%,%)8214565720323789678[GlobalId]
      @(GHC.Classes.IP3602879701896396848
          (GHC.TypeLits.AppendSymbol3674937295934325116
             "System"
             "_clk")
          (Clash.Signal.Internal.Clock8214565720323789554
             "System"))
      @(Clash.Signal.Internal.KnownDomain8214565720323789576
          "System")
      $d(%,%)6989586621679074455[LocalId]
in Test2.subFun8214565720323841755[GlobalId]
     @"System"
     (GHC.Classes.C:(%,,%)8214565720323846233
        @(GHC.Classes.(%,%)7710162562058289156
            (GHC.Classes.IP3602879701896396848
               "System_clk"
               (Clash.Signal.Internal.Clock8214565720323789554
                  "System"))
            (Clash.Signal.Internal.KnownDomain8214565720323789576
               "System"))
        @(GHC.Classes.(%,%)7710162562058289156
            (GHC.Classes.IP3602879701896396848
               "System_rst"
               (Clash.Signal.Internal.Reset8214565720323789578
                  "System"))
            (Clash.Signal.Internal.KnownDomain8214565720323789576
               "System"))
        @(GHC.Classes.(%,%)7710162562058289156
            (GHC.Classes.IP3602879701896396848
               "System_en"
               (Clash.Signal.Internal.Enable8214565720323789569
                  "System"))
            (Clash.Signal.Internal.KnownDomain8214565720323789576
               "System"))
        (GHC.Classes.C:(%,%)8214565720323846230
           @(GHC.Classes.IP3602879701896396848
               "System_clk"
               (Clash.Signal.Internal.Clock8214565720323789554
                  "System"))
           @(Clash.Signal.Internal.KnownDomain8214565720323789576
               "System")
           (GHC.Classes.$p1(%,%)8214565720323789680[GlobalId]
              @(GHC.Classes.IP3602879701896396848
                  (GHC.TypeLits.AppendSymbol3674937295934325116
                     "System"
                     "_clk")
                  (Clash.Signal.Internal.Clock8214565720323789554
                     "System"))
              @(Clash.Signal.Internal.KnownDomain8214565720323789576
                  "System")
              $d(%,%)6989586621679074455[LocalId])
           $dKnownDomain8286623314361781502[LocalId])
        (GHC.Classes.C:(%,%)8214565720323846230
           @(GHC.Classes.IP3602879701896396848
               "System_rst"
               (Clash.Signal.Internal.Reset8214565720323789578
                  "System"))
           @(Clash.Signal.Internal.KnownDomain8214565720323789576
               "System")
           (GHC.Classes.$p1(%,%)8214565720323789680[GlobalId]
              @(GHC.Classes.IP3602879701896396848
                  (GHC.TypeLits.AppendSymbol3674937295934325116
                     "System"
                     "_rst")
                  (Clash.Signal.Internal.Reset8214565720323789578
                     "System"))
              @(Clash.Signal.Internal.KnownDomain8214565720323789576
                  "System")
              $d(%,%)16989586621679074456[LocalId])
           $dKnownDomain8286623314361781502[LocalId])
        (GHC.Classes.C:(%,%)8214565720323846230
           @(GHC.Classes.IP3602879701896396848
               "System_en"
               (Clash.Signal.Internal.Enable8214565720323789569
                  "System"))
           @(Clash.Signal.Internal.KnownDomain8214565720323789576
               "System")
           (Clash.Signal.Internal.enableGen18214565720323789944[GlobalId]
              @"System")
           $dKnownDomain8286623314361781502[LocalId]))
     eta4755801206503243777[LocalId]
```

In this core the dictionary of `KnownDomain` is unavailable. In this specific case we get lucky, because Clash will still be able to synthesize this design thanks to `KnownConf` being a type family and `caseElemNonReachable` solving the case based on this. In general though, we should look up monomorphic constraints and inline their dictionary.

christiaanb, 2024-01-21:
Another option is to not make `KnownDomain dom` not part of the `HiddenClock/Reset/Enable` constraint synonym.

With regards to “just inlining the dictionaries” I take is you mean that as a preprocessing step for synthesis boundaries/entry points; not all binders that have a type-class type.

martijnbastiaan, 2024-01-22:
> With regards to “just inlining the dictionaries” I take is you mean that as a preprocessing step for synthesis boundaries/entry points; not all binders that have a type-class type.

Yeah, that was a poorly phrased way of saying we could conjure up the term (in fact, we should already have it somewhere in a `ClashEnv`y type).

> not all binders that have a type-class type

Not sure I understand, could you elaborate?

christiaanb, 2024-01-22:
We _could_ write a transformation that’s part of the normalization process that for every variable reference:

```
x :: TC
```

replaces `x` with a fresh dictionary. But I assumed that’s not what you meant.

DigitalBrains1, 2024-01-22:
> Another option is to not make `KnownDomain dom` not part of the `HiddenClock/Reset/Enable` constraint synonym.

Ah but I recently wrote [some documentation](https://hackage.haskell.org/package/clash-prelude-1.8.1/docs/Clash-Explicit-Signal.html#g:conveniencetypes) recommending people construct such tuples themselves...