clash-lang/clash-compiler #1029: Performance problem when loading some libraries

tilk, 2020-01-23:
I have a problem where loading `Language.Haskell.Meta` leads to extremely slow (over a minute) compilation times. The easiest way to see this problem is:
```
$ clash.clashi -XBangPatterns 
Clashi, version 1.0.1 (using clash-lib, version 1.0.1):
http://www.clash-lang.org/  :? for help
Clash.Prelude> :m Language.Haskell.Meta
Clash.Prelude Language.Haskell.Meta> let !z = toExp
```
Problem happens both with clashi and the clash compiler. The clash process runs at 100% processor usage during the delay. During that time, it does not respond to CTRL+C or even SIGTERM.
I have installed Clash using Snap.

martijnbastiaan, 2020-01-23:
This has been fixed in master (Clash 1.2) and in the beta release of 1.0. You can install the beta using:

```
snap install clash --beta
```

I guess this means we should release 1.0.2 soon :).

tilk, 2020-01-23:
This helped, thank you.

closed 2022-02-01 by martijnbastiaan