clash-lang/clash-compiler #1041: Nested product types are incorrectly decomposed into ports

gergoerdi, 2020-01-27:
(Clash 1.0.1)

I have the following (simplified) definitions:

```
type A = Unsigned 8

data VGASync dom = VGASync
    { vgaHSync :: Signal dom A
    , vgaVSync :: Signal dom A
    , vgaDE :: Signal dom A
    }

data VGAOut dom = VGAOut
    { vgaSync  :: VGASync dom
    , vgaR     :: Signal dom A
    , vgaG     :: Signal dom A
    , vgaB     :: Signal dom A
    }

vgaPort :: PortName
vgaPort = PortProduct "VGA"
    [ PortProduct ""
      [ PortName "HSYNC"
      , PortName "VSYNC"
      , PortName "DE"
      ]
    , PortName "RED"
    , PortName "GREEN"
    , PortName "BLUE"
    ]
```

I'd like to use it the following way:

```
{-# ANN topEntity
  (Synthesize
    { t_name   = "Pattern"
    , t_inputs =
          [ PortName "CLK_25MHZ"
          , PortName "RESET"
          ]
    , t_output =
            vgaPort
    }) #-}
topEntity
    :: Clock System
    -> Reset System
    -> VGAOut System
topEntity clk rst = VGAOut{..}
  where
    vgaSync = VGASync{..}
      where
        vgaHSync = pure 0
        vgaVSync = pure 1
        vgaDE = pure 2
    vgaR = pure 3
    vgaG = pure 4
    vgaB = pure 5
```

However, Clash seems to get confused about the nested records, since the generated Verilog looks like this:

```
module Pattern
    ( // Inputs
      input  CLK_25MHZ // clock
    , input  RESET // reset

      // Outputs
    , output wire [7:0] VGA_HSYNC
    , output wire [7:0] VGA_VSYNC
    , output wire [7:0] VGA_DE
    , output wire [7:0] VGA_RED
    , output wire [7:0] VGA_GREEN
    , output wire [7:0] VGA_BLUE
    );
  wire [47:0] VGA;
  wire [23:0] VGA;

  assign VGA = {{8'd0,8'd1,8'd2}
               ,8'd3
               ,8'd4
               ,8'd5};

  assign VGA = VGA[47:24];

  assign VGA_RED = VGA[23:16];

  assign VGA_GREEN = VGA[15:8];

  assign VGA_BLUE = VGA[7:0];

  assign VGA_HSYNC = VGA[23:16];

  assign VGA_VSYNC = VGA[15:8];

  assign VGA_DE = VGA[7:0];

endmodule
```

```

gergoerdi, 2020-01-27:
If I flatten the definition of `VGAOut`, then it seems OK:

```
data VGAOut dom = VGAOut
    { vgaHSync :: Signal dom A
    , vgaVSync :: Signal dom A
    , vgaDE    :: Signal dom A
    , vgaR     :: Signal dom A
    , vgaG     :: Signal dom A
    , vgaB     :: Signal dom A
    }

vgaPort :: PortName
vgaPort = PortProduct "VGA"
    [ PortName "HSYNC"
    , PortName "VSYNC"
    , PortName "DE"
    , PortName "RED"
    , PortName "GREEN"
    , PortName "BLUE"
    ]
```

results in

```
module Pattern
    ( // Inputs
      input  CLK_25MHZ // clock
    , input  RESET // reset

      // Outputs
    , output wire [7:0] VGA_HSYNC
    , output wire [7:0] VGA_VSYNC
    , output wire [7:0] VGA_DE
    , output wire [7:0] VGA_RED
    , output wire [7:0] VGA_GREEN
    , output wire [7:0] VGA_BLUE
    );
  wire [47:0] VGA;

  assign VGA = {8'd0,8'd1,8'd2,8'd3,8'd4,8'd5};

  assign VGA_HSYNC = VGA[47:40];

  assign VGA_VSYNC = VGA[39:32];

  assign VGA_DE = VGA[31:24];

  assign VGA_RED = VGA[23:16];

  assign VGA_GREEN = VGA[15:8];

  assign VGA_BLUE = VGA[7:0];

endmodule
```

christiaanb, 2020-01-27:
Right, it seems the naming gets confused by the `PortProduct ""`, and so we end up with two wires called "VGA". If you change it to `PortProduct "Q"` you'd see that it works correctly.

gergoerdi, 2020-01-27:
Hrm... So should I use `PortProduct ""` on the outside, and `PortProduct "VGA"` in the inside, and call the other signals `"VGA_RED"` etc.? I don't want to change the port names as far as the outside world is concerned...

christiaanb, 2020-01-27:
As a work-around, yes.

commit eb808497 (2020-02-03): Generate unique names for port wires. Fix #1041

When using names explicitly given in a topentity annotation,
the identifier for a PortProduct is now used to generate a unique
basic identifier (instead of using `uniquePortName`). This means
that if a nested product has an empty name, the names of wires
are not duplicated in the design.
commit 8bdc7fa4 (2020-02-03): Generate unique names for port wires. Fix #1041

When using names explicitly given in a topentity annotation,
the identifier for a PortProduct is now used to generate a unique
basic identifier (instead of using `uniquePortName`). This means
that if a nested product has an empty name, the names of wires
are not duplicated in the design.
commit ae3dd64a (2020-02-03): Merge pull request #1059 from clash-lang/nested-product-names

Generate unique names for port wires. Fix #1041
closed 2020-02-03 by alex-mckenna