clash-lang/clash-compiler #1046: `substTm` mixes up term and type vars

martijnbastiaan, 2020-01-29:
Consider two variables. One is a term variable:

```haskell
termVar :: Var Term
termVar = Id {
    varName = fakeName {nameUniq=unique, nameOcc="term"}
  , varUniq = unique
  , varType = ConstTy (TyCon fakeName)
  , idScope = LocalId
  }
```

and the other a type variable

```haskell
typeVar :: Var Type
typeVar = TyVar {
    varName = fakeName {nameUniq=unique, nameOcc="type"}
  , varUniq = unique
  , varType = ConstTy (TyCon fakeName)
  }
```

We'd expect that deshadowing `Var termVar` with an `inScopeSet` of just `typeVar` should be a no-op, as their uniques live in different name spaces. (In fact, it's not even possible to have both `termVar` and `typeVar` in an InScopeSet, as it only stores uniques!). `deShadowTerm` currently behaves differently though:

```haskell
>>> deShadowTerm (extendInScopeSet emptyInScopeSet typeVar) (Var termVar)
Var (TyVar {varName = Name {nameSort = User, nameOcc = "type", nameUniq = 20, nameLoc = UnhelpfulSpan "<no location info>"}, varUniq = 20, varType = ConstTy (TyCon (Name {nameSort = User, nameOcc = "fake", nameUniq = 0, nameLoc = UnhelpfulSpan "<no location info>"}))})
```

instead of:

```haskell
>>> deShadowTerm (extendInScopeSet emptyInScopeSet typeVar) (Var termVar)
Var (Id {varName = Name {nameSort = User, nameOcc = "term", nameUniq = 20, nameLoc = UnhelpfulSpan "<no location info>"}, varUniq = 20, varType = ConstTy (TyCon (Name {nameSort = User, nameOcc = "fake", nameUniq = 0, nameLoc = UnhelpfulSpan "<no location info>"})), idScope = LocalId})
```

The culprit seems to be anti space-leak measures in `lookupIdSubst`:

https://github.com/clash-lang/clash-compiler/blob/a009fe45a981031d91f58cecaf4af28e7601d416/clash-lib/src/Clash/Core/Subst.hs#L512

It should refuse to substitute `TyVar`s with `Id`s.

christiaanb, 2020-01-29:
The situation you describe though is really weird though. I have no idea how you end up in a situation where you have a free term variable with a type variable with the same unique in scope...

martijnbastiaan, 2020-01-29:
It's specific to `caseCon`, where we add the existential vars to the inscopeset and then do a deshadow. We can probably leave that out, but I'm much more comfortable with just fixing substitution instead.

christiaanb, 2020-01-29:
No, you should extend the inScopeSet. However, now I get what's going on, from `substTm`s point-of-view we have built up an "illegal" inScopeSet because we originally only used it to pick new variable names that were not in-scope. However, `substTm` expects that any shadowing variable exists in the substitution environment, but we don't rename bound variables as we traverse down the expression during normalization and only keep an inscopeset around.

commit 8b98bd6b (2020-01-29): Don't confuse term/type namespaces in 'lookupIdSubst'. Fix #1046
commit 040fd6d6 (2020-01-29): Don't confuse term/type namespaces in 'lookupIdSubst'. Fix #1046
commit a3eac01b (2020-01-29): Don't confuse term/type namespaces in 'lookupIdSubst'. Fix #1046
commit d66e047e (2020-01-29): Don't confuse term/type namespaces in 'lookupIdSubst'. Fix #1046
closed 2020-01-30 by martijnbastiaan