clash-lang/clash-compiler #1049: ghdl v0.36 causes error

IchiroKawashima, 2020-01-30:
I tried running tests of the compiler with `ghdl-v0.36` which is compiled from the source in centos 7 then the tests gave following error:

```
/path/to/ghdl: unknown option '-o' for command '-m'
```

When testing, I compiled [this commit](https://github.com/clash-lang/clash-compiler/commit/7a6491ee5ec179d2014bcc396a4596fba7a1467b) for the test and the version of `ghdl` is like this:
```
GHDL 0.36 (tarball) [Dunoon edition]
 Compiled with GNAT Version: 4.8.5 20150623 (Red Hat 4.8.5-28
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2019 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

The test should work because other one could use the version of `ghdl` in #136 .
Or am I using a wrong one?

alex-mckenna, 2020-01-31:
`ghdl` has three separate backends (`mcode`, `llvm` and `gcc`). It appears the packaged version for your OS is the `mcode` backend. This is the only backend which doesn't output an executable binary on building (which Clash requires).

I would recommend using the `llvm` version, which is _considerably_ easier to build and install than the `gcc` version. See more in [the `ghdl` documentation](https://ghdl.readthedocs.io/en/latest/getting/index.html).

christiaanb, 2020-01-31:
I guess we could try to detect the backend of ghdl and change how the clash testsuite calls ghdl accordingly.

alex-mckenna, 2020-01-31:
Or at the very least have the test suite initially check that 

  * programs are installed for the tests to be run
  * installed programs are compatible versions

and react accordingly. This would also help make users aware if, for example, their version of `ghdl` was too old for Clash (like recently when we found `ghdl-0.31` was too old).

IchiroKawashima, 2020-01-31:
I could run the test with `ghcl-llvm` as you said @alex-mckenna 😃, the installation of `llvm-5.0` was little a bit tricky for centos 7 tho. The version of the simulator is as follows:

```
GHDL 0.36 (tarball) [Dunoon edition]
 Compiled with GNAT Version: 4.8.5 20150623 (Red Hat 4.8.5-28
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2019 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

IchiroKawashima, 2020-01-31:
But, I still got errors related to a SystemVerilog compiler. The error message is like this

<details><summary>Error message</summary>
<p>

```
              Program /opt/Mentor/questasim/bin/vlog failed with error-code 2.

              Stderr was:

              Stdout was:
              QuestaSim vlog 2019.1 Compiler 2019.01 Jan  1 2019
              Start time: 21:20:58 on Jan 31,2020
              vlog -sv -work work /home/ichiro/Projects/clash-compiler/.clash-test-tmp/clash-test--faee4f7b8833d3f4/./tests/shouldwork/GADTs/TailOfTail/SystemVerilog/systemverilog/TailOfTail/TailOfTail_types.sv /home/ichiro/Projects/clash-compiler/.clash-test-tmp/clash-test--faee4f7b8833d3f4/./tests/shouldwork/GADTs/TailOfTail/SystemVerilog/systemverilog/TailOfTail/testBench/testBench_types.sv /home/ichiro/Projects/clash-compiler/.clash-test-tmp/clash-test--faee4f7b8833d3f4/./tests/shouldwork/GADTs/TailOfTail/SystemVerilog/systemverilog/TailOfTail/topEntity.sv /home/ichiro/Projects/clash-compiler/.clash-test-tmp/clash-test--faee4f7b8833d3f4/./tests/shouldwork/GADTs/TailOfTail/SystemVerilog/systemverilog/TailOfTail/TailOfTail_types.sv /home/ichiro/Projects/clash-compiler/.clash-test-tmp/clash-test--faee4f7b8833d3f4/./tests/shouldwork/GADTs/TailOfTail/SystemVerilog/systemverilog/TailOfTail/tailOfTail.sv /home/ichiro/Projects/clash-compiler/.clash-test-tmp/clash-test--faee4f7b8833d3f4/./tests/shouldwork/GADTs/TailOfTail/SystemVerilog/systemverilog/TailOfTail/testBench/testBench_types.sv /home/ichiro/Projects/clash-compiler/.clash-test-tmp/clash-test--faee4f7b8833d3f4/./tests/shouldwork/GADTs/TailOfTail/SystemVerilog/systemverilog/TailOfTail/testBench/testBench.sv
              -- Compiling package TailOfTail_types
              -- Compiling package testBench_types
              -- Compiling module topEntity
              -- Importing package TailOfTail_types
              ** Warning: /home/ichiro/Projects/clash-compiler/.clash-test-tmp/clash-test--faee4f7b8833d3f4/./tests/shouldwork/GADTs/TailOfTail/SystemVerilog/systemverilog/TailOfTail/TailOfTail_types.sv(1): (vlog-2275) 'TailOfTail_types' already exists and will be overwritten.
              -- Compiling package TailOfTail_types
              -- Compiling module tailOfTail
              ** Warning: /home/ichiro/Projects/clash-compiler/.clash-test-tmp/clash-test--faee4f7b8833d3f4/./tests/shouldwork/GADTs/TailOfTail/SystemVerilog/systemverilog/TailOfTail/tailOfTail.sv(7): (vlog-13314) Defaulting port 'ds' kind to 'var' rather than 'wire' due to default compile option setting of -svinputport=relaxed.
              ** Error: (vlog-13069) /home/ichiro/Projects/clash-compiler/.clash-test-tmp/clash-test--faee4f7b8833d3f4/./tests/shouldwork/GADTs/TailOfTail/SystemVerilog/systemverilog/TailOfTail/tailOfTail.sv(15): near "[": syntax error, unexpected '[', expecting ';' or ','.
              ** Warning: /home/ichiro/Projects/clash-compiler/.clash-test-tmp/clash-test--faee4f7b8833d3f4/./tests/shouldwork/GADTs/TailOfTail/SystemVerilog/systemverilog/TailOfTail/testBench/testBench_types.sv(1): (vlog-2275) 'testBench_types' already exists and will be overwritten.
              -- Compiling package testBench_types
              -- Compiling module testBench
              -- Importing package testBench_types
              End time: 21:20:59 on Jan 31,2020, Elapsed time: 0:00:01
              Errors: 1, Warnings: 3
```

</p>
</details>

I'm using QuestaSim instead of ModelSim and the version is:

```
Questa Sim  vsim 2019.1 Simulator 2019.01 Jan  1 2019
```

I thought QuestaSim is compatible with MedelSim. Is the problem on the simulator?
Should I have made new issue?

alex-mckenna, 2020-01-31:
This is part of [a different issue](#1024) I think, unless you see this with all tests run with SystemVerilog?

IchiroKawashima, 2020-01-31:
I checked thrugh the test result. I think the problem is the same as #1024.

alex-mckenna, 2020-01-31:
Closing in favour of the new related issue, as the other remaining problem is part of #1024.

closed 2020-01-31 by alex-mckenna