clash-lang/clash-compiler #1055: Simulation performance of `Unsigned n` for special values of `n`

gergoerdi, 2020-02-02:
Could `Unsigned 8` be represented as a `Word8`, with a no-op `bitCoerce` both ways? Similarly for `Signed 16` and `Int16` etc.? I am currently writing some Clash code that I'd like to simulate interactively as well, writing RGB color values into an SDL texture. If I use `Unsigned 8` in my code and convert to `Word8` on the edges, it is much slower than using `Word8` through and through internally.

christiaanb, 2020-02-03:
Myeah... I don't know how. Sadly the following isn't allowed:

```haskell
module Test where

import Clash.Prelude
import Data.Word

class UnsignedX (n :: Nat) where
  data UnsignedN n

instance {-# OVERLAPPING #-} UnsignedX 8 where
  data UnsignedN 8 = U8 Word8

instance {-# OVERLAPPING #-} UnsignedX 16 where
  data UnsignedN 16 = U16 Word16

instance {-# OVERLAPPING #-} UnsignedX 32 where
  data UnsignedN 32 = U32 Word32

instance {-# OVERLAPPING #-} UnsignedX 64 where
  data UnsignedN 64 = U64 Word64

instance {-# OVERLAPPABLE #-} UnsignedX n where
  data UnsignedN n = UN (Unsigned n)

instance Num (UnsignedN 8) where
  fromInteger = U8 . fromInteger

instance Num (UnsignedN 16) where
  fromInteger = U16 . fromInteger

instance Num (UnsignedN 32) where
  fromInteger = U32 . fromInteger

instance Num (UnsignedN 64) where
  fromInteger = U64 . fromInteger
```

```
[1 of 1] Compiling Test             ( Test.hs, interpreted )

Test.hs:10:8: error:
    Conflicting family instance declarations:
      UnsignedN 8 -- Defined at Test.hs:10:8
      UnsignedN n -- Defined at Test.hs:22:8
   |
10 |   data UnsignedN 8 = U8 Word8
   |        ^^^^^^^^^

Test.hs:13:8: error:
    Conflicting family instance declarations:
      UnsignedN 16 -- Defined at Test.hs:13:8
      UnsignedN n -- Defined at Test.hs:22:8
   |
13 |   data UnsignedN 16 = U16 Word16
   |        ^^^^^^^^^

Test.hs:16:8: error:
    Conflicting family instance declarations:
      UnsignedN 32 -- Defined at Test.hs:16:8
      UnsignedN n -- Defined at Test.hs:22:8
   |
16 |   data UnsignedN 32 = U32 Word32
   |        ^^^^^^^^^

Test.hs:19:8: error:
    Conflicting family instance declarations:
      UnsignedN 64 -- Defined at Test.hs:19:8
      UnsignedN n -- Defined at Test.hs:22:8
   |
19 |   data UnsignedN 64 = U64 Word64
   |        ^^^^^^^^^
Failed, no modules loaded.
```

martijnbastiaan, 2020-02-03:
Could backpack be used for this?

alex-mckenna, 2020-02-03:
Not without breaking compatibility with `stack`, and thus Windows IIRC

gergoerdi, 2020-02-04:
Just to show my work and to save others the frustration, here's how far I got:

```
data Unsigned (n :: Nat) where
    U8 :: Word8 -> Unsigned 8
    U16 :: Word16 -> Unsigned 16
    U :: (IsSpecialSize n ~ False) => Integer -> Unsigned n

-- So far so good
instance (KnownNat n) => Eq (Unsigned n) where
    U8 x == U8 y = x == y
    U16 x == U16 y = x == y
    U x == U y = x == y

-- But things get tricky for e.g. Num#fromInteger:

type family IsSpecialSize (n :: Nat) :: Bool where
    IsSpecialSize 8 = True
    IsSpecialSize 16 = True
    IsSpecialSize _ = False

data SSpecialSize (n :: Nat) where
    Special8 :: SSpecialSize 8
    Special16 :: SSpecialSize 16
    NonSpecial :: (IsSpecialSize n ~ False) => SSpecialSize n

class (KnownNat n) => SpecialSize (n :: Nat) where
    specialSize :: proxy n -> SSpecialSize n

instance {-# OVERLAPPING #-} SpecialSize 8 where
    specialSize _ = Special8

instance {-# OVERLAPPING #-} SpecialSize 16 where
    specialSize _ = Special16

instance {-# OVERLAPPABLE #-} (KnownNat n, IsSpecialSize n ~ False) => SpecialSize n where
    specialSize _ = NonSpecial

instance forall n. (KnownNat n) => Num (Unsigned n) where
    fromInteger = case specialSize (Proxy @n) of
        Special8 -> U8 . fromInteger
        Special16 -> U16 . fromInteger
        NonSpecial -> U . fromInteger
```

This fails because `SpecialSize n`'s head matches the `specialSize (Proxy @n)` call, resulting in

>   • Couldn't match type `IsSpecialSize n` with `'False` arising from a use of `specialSize`
>   • In the expression: `specialSize (Proxy @n)`
>

gergoerdi, 2020-02-04:
https://stackoverflow.com/q/60057227/477476

leonschoorl, 2020-02-04:
It would be cool if we could get something like this to work.
But isn't the much simpler option to just use the `Word*` or `Int*` types directly?
And if you need to write any generic functions for them, use a constraint like `(Num n, BitPack n) => ...`

gergoerdi, 2020-02-04:
Maybe?
But in some situations, I'd like to enforce unsignedness at least. For example, for a VGA output with `r`, `g` and `b` bits of red/green/blue DAC depth, it'd feel weird to allow `Signed r` as the red channel.

gergoerdi, 2020-02-05:
Based on https://stackoverflow.com/a/60059575/477476 I think we can do this with quite nice ergonomics actually.

The individual instances can be lifted as follows:

```
instance (KnownNat n) => Show (Unsigned n) where
    show = unary show show show show show

instance (KnownNat n) => Eq (Unsigned n) where
    (==) = bin (==) (==) (==) (==) (==)
    (/=) = bin (/=) (/=) (/=) (/=) (/=)

instance (KnownNat n) => Ord (Unsigned n) where
    compare = bin compare compare compare compare compare

instance (KnownNat n) => Num (Unsigned n) where
    fromInteger = con fromInteger fromInteger fromInteger fromInteger fromInteger

    (+) = binOp (+) (+) (+) (+) (+)
    (-) = binOp (-) (-) (-) (-) (-)
    (*) = binOp (*) (*) (*) (*) (*)
    abs = unOp abs abs abs abs abs
    signum = unOp signum signum signum signum signum
```

And turns out the margin of this Github issue is *not* too small to show the full infrastructure needed!

```
{-# LANGUAGE DataKinds, TypeFamilies, GADTs, RankNTypes #-}
{-# LANGUAGE TypeOperators, TypeApplications, ScopedTypeVariables #-}
{-# LANGUAGE RecordWildCards #-}
import Data.Word
import GHC.TypeLits (KnownNat, Nat)
import Data.Proxy
import Data.Type.Equality
import GHC.TypeNats (sameNat)
import Data.Singletons.Prelude.Bool
import Unsafe.Coerce

eqNat :: forall n m. (KnownNat n, KnownNat m) => Either ((n == m) :~: 'False) (n :~: m)
eqNat =
  case sameNat @n @m Proxy Proxy of
    Just r -> Right r
    Nothing -> Left (unsafeCoerce (Refl :: 'False :~: 'False))

type IsSpecialSize (n :: Nat) = (n == 8) || (n == 16) || (n == 32) || (n == 64)

data SSize (n :: Nat) where
    SSize8 :: SSize 8
    SSize16 :: SSize 16
    SSize32 :: SSize 32
    SSize64 :: SSize 64
    SSizeOther :: (IsSpecialSize n ~ False) => SSize n

sSize :: forall n. (KnownNat n) => SSize n
sSize =
  case eqNat @n @8 of
    Right Refl -> SSize8
    Left Refl -> case eqNat @n @16 of
        Right Refl -> SSize16
        Left Refl -> case eqNat @n @32 of
            Right Refl -> SSize32
            Left Refl -> case eqNat @n @64 of
                Right Refl -> SSize64
                Left Refl -> SSizeOther

data Unsigned (n :: Nat) where
    U8 :: Word8 -> Unsigned 8
    U16 :: Word16 -> Unsigned 16
    U32 :: Word32 -> Unsigned 32
    U64 :: Word64 -> Unsigned 64
    U :: (IsSpecialSize n ~ False) => Integer -> Unsigned n

type Kit n a8 a16 a32 a64 a0 a =
    ((n ~ 8) => a8) ->
    ((n ~ 16) => a16) ->
    ((n ~ 32) => a32) ->
    ((n ~ 64) => a64) ->
    ((IsSpecialSize n ~ False) => a0) ->
    a

con :: forall n a. (KnownNat n) => Kit n (a -> Word8) (a -> Word16) (a -> Word32) (a -> Word64) (a -> Integer) (a -> Unsigned n)
con f8 f16 f32 f64 f0 = case sSize @n of
    SSize8 -> U8 . f8
    SSize16 -> U16 . f16
    SSize32 -> U32 . f32
    SSize64 -> U64 . f64
    SSizeOther -> U . f0

unary :: forall n a. (KnownNat n) => Kit n (Word8 -> a) (Word16 -> a) (Word32 -> a) (Word64 -> a) (Integer -> a) (Unsigned n -> a)
unary f8 f16 f32 f64 f0 = case sSize @n of
    SSize8 -> \(U8 x) -> f8 x
    SSize16 -> \(U16 x) -> f16 x
    SSize32 -> \(U32 x) -> f32 x
    SSize64 -> \(U64 x) -> f64 x
    SSizeOther -> \(U x) -> f0 x

type UnOp a = a -> a

unOp :: forall n. (KnownNat n) => Kit n (UnOp Word8) (UnOp Word16) (UnOp Word32) (UnOp Word64) (UnOp Integer) (UnOp (Unsigned n))
unOp f8 f16 f32 f64 f0 = unary (U8 . f8) (U16 . f16) (U32 . f32) (U64 . f64) (U . f0)

type Bin a r = a -> a -> r

bin :: forall n a. (KnownNat n) => Kit n (Bin Word8 a) (Bin Word16 a) (Bin Word32 a) (Bin Word64 a) (Bin Integer a) (Bin (Unsigned n) a)
bin f8 f16 f32 f64 f0 = case sSize @n of
    SSize8     -> \ (U8 x)  (U8 y)  -> f8 x y
    SSize16    -> \ (U16 x) (U16 y) -> f16 x y
    SSizeOther -> \ (U x)   (U y)   -> f0 x y

type BinOp a = a -> a -> a

(...) :: (r -> r') -> (a -> b -> r) -> (a -> b -> r')
f ... g = \x y -> f (g x y)

binOp :: forall n. (KnownNat n) => Kit n (BinOp Word8) (BinOp Word16) (BinOp Word32) (BinOp Word64) (BinOp Integer) (BinOp (Unsigned n))
binOp f8 f16 f32 f64 f0 = bin (U8 ... f8) (U16 ... f16) (U32 ... f32) (U64 ... f64) (U ... f0)
```

gergoerdi, 2020-02-05:
I strongly suspect that `con`, `unary` and `bin` can be unified somehow, but that's now an internal implementation detail.

gergoerdi, 2020-02-05:
In fact, we might even use `Word8` for `Unsigned n` if `n <= 8`, if only we do sufficient truncation on results before converting to something else. Just a thought; this could get ugly fast.

christiaanb, 2020-02-06:
But is it actually faster? Since it seems to suggest that a lot of constant folding and specialization needs to happen at compile-time; otherwise we're doing those `eqNat` things at runtime, and also still pattern-match on the constructor of this new `Unsigned` instead of the constructor of `Integer` which is used by the current version of `Unsigned`.

It would be interesting to see the results of the clash-prelude benchmark (https://github.com/clash-lang/clash-compiler/blob/master/clash-prelude/clash-prelude.cabal#L370) for this new representation of `Unsigned`

gergoerdi, 2020-02-09:
Running the full benchmark would require putting in the full work, before we even know if it is worth it. However, I have now run parts of the benchmark that didn't require me to do any thinking :) I'd say the numbers are very promising for arithmetic performance. All tests are with `WORD_SIZE_IN_BITS` set to 64.

With `Integer` as the underlying representation (i.e., before my changes):

```
benchmarking Signed/fromInteger# WORD_SIZE_IN_BITS
time                 75.14 ns   (74.89 ns .. 75.38 ns)
                     1.000 R²   (1.000 R² .. 1.000 R²)
mean                 75.35 ns   (75.13 ns .. 75.67 ns)
std dev              849.4 ps   (635.4 ps .. 1.220 ns)
variance introduced by outliers: 11% (moderately inflated)
                           
benchmarking Signed/+# WORD_SIZE_IN_BITS
time                 64.08 ns   (63.86 ns .. 64.35 ns)
                     1.000 R²   (1.000 R² .. 1.000 R²)
mean                 64.21 ns   (64.05 ns .. 64.40 ns)
std dev              614.2 ps   (513.8 ps .. 746.6 ps)
                           
benchmarking Signed/negate# WORD_SIZE_IN_BITS
time                 50.00 ns   (49.67 ns .. 50.57 ns)
                     0.997 R²   (0.993 R² .. 0.999 R²)
mean                 51.53 ns   (50.40 ns .. 53.67 ns)
std dev              4.981 ns   (2.986 ns .. 6.960 ns)
variance introduced by outliers: 91% (severely inflated)
                           
benchmarking Signed/abs# WORD_SIZE_IN_BITS
time                 50.04 ns   (49.82 ns .. 50.22 ns)
                     1.000 R²   (1.000 R² .. 1.000 R²)
mean                 49.97 ns   (49.85 ns .. 50.13 ns)
std dev              465.5 ps   (375.0 ps .. 638.3 ps)
                           
benchmarking Signed/-# WORD_SIZE_IN_BITS
time                 65.74 ns   (65.52 ns .. 66.01 ns)
                     1.000 R²   (1.000 R² .. 1.000 R²)
mean                 65.75 ns   (65.56 ns .. 66.03 ns)
std dev              772.2 ps   (586.0 ps .. 1.054 ns)
variance introduced by outliers: 12% (moderately inflated)
                           
benchmarking Signed/*# WORD_SIZE_IN_BITS
time                 101.9 ns   (101.4 ns .. 102.4 ns)
                     0.999 R²   (0.996 R² .. 1.000 R²)
mean                 104.4 ns   (102.0 ns .. 113.9 ns)
std dev              15.31 ns   (1.095 ns .. 32.56 ns)
variance introduced by outliers: 95% (severely inflated)
                           
benchmarking Signed/pack# WORD_SIZE_IN_BITS
time                 12.11 ns   (12.07 ns .. 12.15 ns)
                     1.000 R²   (1.000 R² .. 1.000 R²)
mean                 12.08 ns   (12.05 ns .. 12.13 ns)
std dev              145.4 ps   (114.3 ps .. 217.8 ps)
variance introduced by outliers: 14% (moderately inflated)
```

With `Int64` representation (i.e. with my changes):

```
benchmarking Signed/fromInteger# WORD_SIZE_IN_BITS
time                 49.32 ns   (49.01 ns .. 49.69 ns)
                     1.000 R²   (0.999 R² .. 1.000 R²)
mean                 49.68 ns   (49.37 ns .. 50.11 ns)
std dev              1.233 ns   (1.013 ns .. 1.734 ns)
variance introduced by outliers: 38% (moderately inflated)
                           
benchmarking Signed/+# WORD_SIZE_IN_BITS
time                 30.84 ns   (30.49 ns .. 31.31 ns)
                     0.999 R²   (0.998 R² .. 1.000 R²)
mean                 31.11 ns   (30.77 ns .. 31.50 ns)
std dev              1.157 ns   (934.6 ps .. 1.511 ns)
variance introduced by outliers: 59% (severely inflated)
                           
benchmarking Signed/negate# WORD_SIZE_IN_BITS
time                 18.44 ns   (18.39 ns .. 18.50 ns)
                     1.000 R²   (1.000 R² .. 1.000 R²)
mean                 18.54 ns   (18.48 ns .. 18.62 ns)
std dev              233.0 ps   (178.4 ps .. 299.8 ps)
variance introduced by outliers: 14% (moderately inflated)
                           
benchmarking Signed/abs# WORD_SIZE_IN_BITS
time                 18.93 ns   (18.85 ns .. 19.00 ns)
                     1.000 R²   (1.000 R² .. 1.000 R²)
mean                 18.97 ns   (18.89 ns .. 19.06 ns)
std dev              296.8 ps   (240.9 ps .. 376.3 ps)
variance introduced by outliers: 20% (moderately inflated)
                           
benchmarking Signed/-# WORD_SIZE_IN_BITS
time                 29.68 ns   (29.51 ns .. 29.86 ns)
                     0.999 R²   (0.997 R² .. 1.000 R²)
mean                 30.37 ns   (29.78 ns .. 31.88 ns)
std dev              2.766 ns   (513.8 ps .. 4.893 ns)
variance introduced by outliers: 90% (severely inflated)
                           
benchmarking Signed/*# WORD_SIZE_IN_BITS
time                 29.11 ns   (29.00 ns .. 29.25 ns)
                     1.000 R²   (1.000 R² .. 1.000 R²)
mean                 29.32 ns   (29.20 ns .. 29.46 ns)
std dev              437.1 ps   (367.7 ps .. 530.3 ps)
variance introduced by outliers: 19% (moderately inflated)
                           
benchmarking Signed/pack# WORD_SIZE_IN_BITS
time                 13.55 ns   (13.49 ns .. 13.60 ns)
                     1.000 R²   (1.000 R² .. 1.000 R²)
mean                 13.50 ns   (13.43 ns .. 13.55 ns)
std dev              199.8 ps   (163.7 ps .. 235.6 ps)
variance introduced by outliers: 19% (moderately inflated)
```                        

Code is at https://github.com/gergoerdi/clash-compiler/tree/signed-int

christiaanb, 2020-02-09:
Those number are indeed very encouraging! Especially since I see that those functions are still marked `NOINLINE` i.e. stopping GHC dead in its tracks to do any kind of specialization. It would be interesting to extend the benchmark with explicitly polymorphic with NOINLINEs (to simulate "library" code where the `Num (Signed n)` dictionary still takes a `KnownNat n` argument:

```haskell
multBenchPoly :: Benchmark
multBenchPoly = env setup $ \m ->
  bench "*# WORD_SIZE_IN_BITS" $ nf (uncurry (polyMultSigned)) m
  where
    setup = return (smallValue_pos1,smallValue_pos2)

polyMultSigned :: KnownNat n => Signed n -> Signed n -> Signed n
polyMultSigned = (*#)
{-# NOINLINE #-}
```

as this would also stop GHC from doing specialization on the `Num (Signed n)` dictionary and eliminating those `eqNat`s.

gergoerdi, 2020-02-14:
It doesn't seem to change anything:

```
benchmarking Signed/Mono *# WORD_SIZE_IN_BITS
time                 29.79 ns   (29.61 ns .. 29.98 ns)
                     1.000 R²   (1.000 R² .. 1.000 R²)
mean                 29.66 ns   (29.53 ns .. 29.81 ns)
std dev              466.3 ps   (362.1 ps .. 619.3 ps)
variance introduced by outliers: 20% (moderately inflated)
```

vs. 

```
benchmarking Signed/Poly *# WORD_SIZE_IN_BITS
time                 29.40 ns   (29.28 ns .. 29.53 ns)
                     1.000 R²   (1.000 R² .. 1.000 R²)
mean                 29.44 ns   (29.29 ns .. 29.58 ns)
std dev              472.3 ps   (400.0 ps .. 553.3 ps)
variance introduced by outliers: 21% (moderately inflated)
```

christiaanb, 2020-02-14:
Perhaps we can take some of the code from https://github.com/Bodigrim/mod, at least for `Unsigned`.

gergoerdi, 2020-02-14:
I'm not giving up on this idea. I think a good next step would be to finish the implementation for `Signed` so that I can run the test suites (you guys have test suites, right?).

I'm publishing my work as I go to https://github.com/gergoerdi/clash-compiler/tree/signed-int in case anyone wants to help out.

gergoerdi, 2020-02-14:
The functions that are slower (MUCH slower) are the extending ones, where there is no easy way to exploit the special cases... E.g. for `plus#`:

Before:
```
benchmarking Signed/plus# WORD_SIZE_IN_BITS
time                 15.84 ns   (15.75 ns .. 16.00 ns)
                     0.998 R²   (0.997 R² .. 1.000 R²)
mean                 16.23 ns   (16.00 ns .. 16.74 ns)
std dev              1.123 ns   (596.4 ps .. 1.863 ns)
variance introduced by outliers: 84% (severely inflated)
```

With a naive implementation via `Integer`:

```
benchmarking Signed/plus# WORD_SIZE_IN_BITS
time                 76.22 ns   (75.69 ns .. 76.83 ns)
                     0.999 R²   (0.999 R² .. 1.000 R²)
mean                 75.60 ns   (75.05 ns .. 76.31 ns)
std dev              2.061 ns   (1.562 ns .. 2.779 ns)
variance introduced by outliers: 42% (moderately inflated)
```

by naive implementation, I mean the following:

```
plus# a b = fromInteger_INLINE $ toInteger_INLINE a + toInteger_INLINE b
```

gergoerdi, 2020-02-20:
I'm withdrawing from this one because it seems like a red herring. I have manually changed my Bounce example to use `Int`s everywhere, and there is not much improvement to the simulation time for a single frame.

I think I initially misunderstood the timing figures I saw. It is true that using `Word8` instead of `Unsigned 8` for the VGA color channels improved end-to-end simulation performance tremendously, but that is because ultimately I am shoving everything into an SDL texture which is a `Ptr Word8`; so the real overhead is in the `bitCoerce :: Unsigned 8 -> Word8` conversion before writing to the texture.

Of course, if we had `Unsigned 8` represented as `Word8`, it could still be that this could be compiled to a zero-cost `coerce` instead...

christiaanb, 2020-03-03:
@gergoerdi Although it doesn't help with the `bitCoerce`, I have been able to speed up the numeric operators (without changing the underlying data representation too much: Natural instead of Integer) at: https://github.com/clash-lang/clash-compiler/pull/1132

christiaanb, 2020-03-03:
Also by adding:

```haskell
unsigned8toWord8 :: Unsigned 8 -> Word8
unsigned8toWord8 (U (NatS# u#)) = W8# (narrow8Word# u#)
unsigned8toWord8 (U (NatJ# u#)) = W8# (narrow8Word# (bigNatToWord u#))
{-# NOINLINE unsigned8toWord8 #-}

{-# RULES
"bitCoerce/Unsigned 8 -> Word8" bitCoerce = unsigned8toWord8
 #-}
```

I can make bitCoerce for those types go from:

```
time                 22.08 ns   (22.05 ns .. 22.11 ns)
                     1.000 R²   (1.000 R² .. 1.000 R²)
mean                 22.10 ns   (22.07 ns .. 22.13 ns)
std dev              96.59 ps   (78.29 ps .. 129.6 ps
```
to:
```
time                 4.572 ns   (4.566 ns .. 4.585 ns)
                     1.000 R²   (0.999 R² .. 1.000 R²)
mean                 4.594 ns   (4.570 ns .. 4.667 ns)
std dev              150.1 ps   (11.44 ps .. 289.1 ps)
```

Of course I'd have to add a primitive for the HDL backends; But that's probably worth it.

christiaanb, 2020-03-03:
Fast conversions added at: https://github.com/clash-lang/clash-compiler/pull/1144