clash-lang/clash-compiler #1063: Port naming error detection logic not always working

leonschoorl, 2020-02-07:
```haskell
module TestNaming where
import Clash.Prelude

type T = Bool -- error as expected
-- type T = Int  -- no error :(

{-# ANN topEntity (defSyn "top")
  {t_inputs = [PortProduct "wrong" [PortName "one", PortName "two"]]}
  #-}
topEntity :: T -> T
topEntity = id
```
Returns an error as expected:
```
*** Exception: Clash.Netlist.Util(635): Ports were annotated as product, but type wasn't one: 

   Filtered was: FilteredHWType Bool []

   Ports was: PortProduct "wrong" [PortName "one",PortName "two"]
```

But if I change `T` to be `Int` clash will happily name the resulting in port `wrong`.

leonschoorl, 2020-02-07:
I guess it's technically true that in haskell Int is a one element product type: 
```haskell
data Int = GHC.Types.I# GHC.Prim.Int#
```
But that's not how clash handles it.
It names the port just `wrong`, not `wrong_one`.

christiaanb, 2020-03-04:
@leonschoorl Clash should not recognize `Int` as a one-product. https://github.com/clash-lang/clash-compiler/blob/ee45c470acafa522db754c7327ca1ea58b71cdc1/clash-lib/src/Clash/Netlist/Util.hs#L430 should prevent that. If it's not, then something is wrong with https://github.com/clash-lang/clash-compiler/blob/ee45c470acafa522db754c7327ca1ea58b71cdc1/clash-lib/src/Clash/Netlist/Util.hs#L406

commit 762c2bbe (2020-03-05): Single-dc, single-field types cannot have PortProduct annotation

Fixes #1063
commit 18abbbc5 (2020-03-05): Single-dc, single-field types cannot have PortProduct annotation

Fixes #1063
PR #1149 (2020-03-05): Single-dc, single field types cannot have a PortProduct annotation
@christiaanb: Changes the logic in 'filterVoidPorts' to stop recognizing these "wrapper" types as one-products

Fixes #1063
commit 883674a0 (2020-03-05): Single-dc, single-field types cannot have PortProduct annotation

Fixes #1063
closed 2020-03-05 by martijnbastiaan