clash-lang/clash-compiler #1074: 'splitTopAnn' fails for certain top entities

martijnbastiaan, 2020-02-12:
Example:

```haskell
{-# ANN topEntity
  (Synthesize
    { t_name   = "top"
    , t_inputs =
        [ PortProduct "" [PortName "theClock", PortName "theReset"]
        , PortName "theInput"
        ]
    , t_output = PortName "theResult"
    }
  )#-}
topEntity
  :: forall dom1
   . dom1 ~ System
  => forall dom2
   . dom2 ~ System
  => (Clock dom1, Reset dom2)
  -> Signal dom1 Int
  -> Signal dom2 Int
topEntity (clk, rst) = id
```

leonschoorl, 2020-02-12:
Is it because of the two foralls?

martijnbastiaan, 2020-02-12:
Exactly. It goes into an "eq removal" loop followed by the actual logic after seeing a number of foralls. But it should keep removing foralls/eqs until it sees something that's neither a forall nor an eq.

leonschoorl, 2020-02-12:
@martijnbastiaan Did you maybe test before you merged https://github.com/clash-lang/clash-compiler/pull/1072?
Because this seems to just work now.

leonschoorl, 2020-02-13:
If I add an extra (non-equality) constraint in between the first and second forall then it does break:
```haskell
topEntity
  :: forall dom1
   . dom1 ~ System
  => HasCallStack
  => forall dom2
   . dom2 ~ System
  => (Clock dom1, Reset dom2)
  -> Signal dom1 Int
  -> Signal dom2 Int
```

martijnbastiaan, 2020-02-18:
It also breaks for things on `Vec`.

commit 8262384f (2020-02-18): splitTopAnn: skip over ForAllTy's

Fixes #1074
commit bcea31e5 (2020-02-21): splitTopAnn: skip over ForAllTy's

Fixes #1074
PR #1065 (2020-02-21): Report polymorphic topEntities as errors and expand type equality constraint support on them
This PR does a couple of related thing:

- introduce and error when clash is asked to normalize polymorphic functions instead of just going ahead and running into `Clash.Netlist.Util(...): Not in normal form: ...`
- expand the handling of type equality constraints on topentities introduced in #1003, with many situations which weren't handled before: https://github.com/clash-lang/clash-compiler/blob/3a3e121387923d435074dcc70488ab51885e6520/tests/shouldwork/Basic/TyEqConstraints.hs#L16-L47
- fix #1074 

 [... cut]
closed 2020-02-21 by martijnbastiaan