clash-lang/clash-compiler #1082: Clash.Annotations.SynthesisAttributes do not work as expected in VHDL

JvWesterveld, 2020-02-14:
As mentioned in its documentation, the synthesis attributes are an experimental feature, and I have found several cases where it does not work as I would expect with Clash from last week's git master when compiling to VHDL. Probably related to #834. Here are a couple of examples:

Directly on the `topEntity` only prints the attribute for the result:
```
topEntity
  :: Bool `Annotate` 'StringAttr "chip_pin" "C4"
  -> Bool `Annotate` 'StringAttr "chip_pin" "C5"
topEntity = not
```
```
 grep -nirI "attribute"
A/topentity.vhdl:14:  attribute chip_pin : string;
A/topentity.vhdl:15:  attribute chip_pin of result : signal is "C5";
```

When using an additional function `a`, Clash does print the attributes, but it does so in `topentity.vhdl` and not in `a.vhd`.
```
{-# NOINLINE a #-}
a
  :: Bool `Annotate` 'StringAttr "chip_pin" "C4"
  -> Bool `Annotate` 'StringAttr "chip_pin" "C5"
a = not

--topEntity
--  :: Bool-- `Annotate` 'StringAttr "chip_pin" "C4"
--  -> Bool-- `Annotate` 'StringAttr "chip_pin" "C5"
topEntity = a
```
```
 grep -nirI "attribute"
A/topentity.vhdl:14:  attribute chip_pin : string;
A/topentity.vhdl:15:  attribute chip_pin of result : signal is "C5";
A/topentity.vhdl:16:  attribute chip_pin of \c$arg\ : signal is "C4";
```

But when adding the type signature for `topEntity`, it prints it only for the function argument, not for the result:
```
{-# NOINLINE a #-}
a
  :: Bool `Annotate` 'StringAttr "chip_pin" "C4"
  -> Bool `Annotate` 'StringAttr "chip_pin" "C5"
a = not

topEntity
  :: Bool
  -> Bool
topEntity = a
```
```
 grep -nirI "attribute"
A/topentity.vhdl:14:  attribute chip_pin : string;
A/topentity.vhdl:15:  attribute chip_pin of \c$arg\ : signal is "C4";
```

Finally, this prints the attributes twice, in `topentity.vhdl` and `b.vhdl`, but not in `a.vhdl`:
```
{-# NOINLINE a #-}
a
  :: Bool `Annotate` 'StringAttr "chip_pin" "C4"
  -> Bool `Annotate` 'StringAttr "chip_pin" "C5"
a = not

{-# NOINLINE b #-}
b = a

topEntity = b
```
```
 grep -nirI "attribute"
A/topentity.vhdl:14:  attribute chip_pin : string;
A/topentity.vhdl:15:  attribute chip_pin of result : signal is "C5";
A/topentity.vhdl:16:  attribute chip_pin of \c$arg\ : signal is "C4";
A/b.vhdl:14:  attribute chip_pin : string;
A/b.vhdl:15:  attribute chip_pin of result : signal is "C5";
A/b.vhdl:16:  attribute chip_pin of \c$arg\ : signal is "C4";
```

martijnbastiaan, 2020-02-14:
@JvWesterveld Thanks for all the examples!

JvWesterveld, 2020-02-14:
Now that I think of it, the `chip_pin` attributes I used in the examples are of course not very useful for anything other than the topentity ports, but I wanted to use this feature for passing some other attributes, like `keep`, to Vivado.